vhelin/wla-dx #137: Power operator only supports integer exponents

maxim-zhao, 2017-05-08:
The docs say everything is doubles underneath but the implementation of the "^" operator does integer powers of doubles. 

I needed the precision so I trivially changed it to use pow() from <math.h> - is there a reason that is avoided?

vhelin, 2017-05-08:
> The docs say everything is doubles underneath but the implementation of
> the "^" operator does integer powers of doubles.
>
> I needed the precision so I trivially changed it to use pow() from
> <math.h> - is there a reason that is avoided?

No reason that I'd remember any more. :)

vhelin, 2017-05-14:
Hmm, I think WLA already calculated the pow in doubles, at least q and v are both doubles, so your fix just makes WLA to use math-lib's pow() function. ???

vhelin, 2017-05-14:
Oh man, my bad! One cannot do x^y where y is a real number, the loop makes it an integer. I'll add your code to master.

closed 2017-05-14 by vhelin

maxim-zhao, 2017-05-15:
Confusing git workflow meant I managed to submit a PR to my own fork with the bugfix - but your change is identical.