vhelin/wla-dx #140: Error report line numbers are sometimes wrong

maxim-zhao, 2017-05-24:
I have a project where errors are reporting line numbers quite far away from the actual line containing the error. I will try to nail down what causes it...

nicklausw, 2017-05-24:
Which program is reporting the inaccurate errors, the assembler or linker?

nicklausw, 2017-05-24:
Wait, are you using C-style comments (\/*...*/\)? Some other programs tend to throw away newlines when they come across that and it spans multiple lines.

maxim-zhao, 2017-05-24:
A little experimenting suggests I "lose" one line every time I nest a struct (!).

```
.struct Word
Lo db
Hi db
.endst
ld a, 1000 ; error on line 5 (correct)
.struct SoundChannel
  Attenuation db
  Tone instanceof Word
.endst
ld a, 1000 ; error on line 9 (off by one)
```

maxim-zhao, 2017-05-24:
Seems to be caused by the (optional) array size integer following the struct name - it looks like the line number doesn't get incremented when the number is at the end of the line. Adding a number makes it work.

I think the code at pass_1.c:2238 needs to call next_line() if input_number() returned ``INPUT_NUMBER_EOL``.

vhelin, 2017-06-04:
Thanks for the fix!

PR #142 (2017-06-04): Bugfix for #140
Fixes #140
closed 2017-06-04 by vhelin