vhelin/wla-dx #141: Line continuation

maxim-zhao, 2017-05-24:
I can't seem to find any way to split lines, a bit like the trailing '\' for C macros. This would be quite useful for macros with a lot of args or for .dstruct.

```
.dstruct enemy1 \
  instanceof enemystruct 
  0, 12 \ ; comments still allowed though...
  behaviour1, \
  pattern1, \
```

vs.

```
.dstruct enemy1 instanceof enemystruct data 0, 12, behaviour1, pattern1
```

nicklausw, 2017-05-24:
I think this could just be a case of the parser scanning for `\` and grabbing the next line if it's there, over and over until there's no `\`. Increment the line number, too. Sounds easy enough to implement but it might be error-prone, so care should be taken to make sure that no crashes happen at the end of file or anything like that.

nicklausw, 2017-05-24:
Question is, if there's something after the `\` then is it discarded or added before the next line?

maxim-zhao, 2017-05-24:
I think the rule is a '\' followed only by whitespace and comments joins to the line after it (maybe easier to reason about only line comments).

maxim-zhao, 2017-08-04:
Reading around a bit:

https://stackoverflow.com/questions/41585136/masm-8086-assembly-language-line-continuation

> in MASM 6.1, a trailing comma in initializers implies that the line continues onto the next line. Therefore, the following declarations are legal:
>
>```
>EvenNumbers DB  02h, 04h, 06h, 08h,
>                 0Ah, 0Ch, 0Eh, 0Fh
>```
>```
>OutputStr DB  "Hello ",
>               "world!"
>```
Sounds like a good idea and arguably nicer to read, if a bit more spooky, than explicit continuation.

vhelin, 2021-03-15:
Haha, I've pretty well neglected these old issues so far, but I'll try to go through them all in the near future and fix/implement what's doable in a relatively short time.

Kroc, 2021-03-15:
Great addition! How does this work with strings? For example, I want to wrap long error messages. In CA65 I use `.concat` to join the strings across multiple lines.

vhelin, 2021-03-15:
Currently it doesn't support string splitting / concat, '\\' only works between elements, but if you are doing

.db "A very long string..."

you can now do

.db "A very long " \
  "string..."

(which in fact is the same as '.db "A very long ", "string..."')

:) Or how would you like it to work?

Kroc, 2021-03-15:
I'm just thinking of ".FAIL" and the like, which I believe, don't support concatenating strings

vhelin, 2021-03-15:
I don't think .FAIL supports any strings at all. :D Which is actually a nice feature idea...

If you have a list of directives that could use string concatenating, please list them here. I'm just thinking that is it a bigger feature to implement to add multistring support (like .INCLUDE has) to those or to make something like "hello " + "world" -> "hello world" to work in the stack calculator...

Hmm...

vhelin, 2021-07-21:
Ok, with the latest sources a string can be split to multiple lines using a '\' at the end of a line. All it really took was a small modification to the parser that took care of strings inside " ... "

vhelin, 2021-07-23:
As the issue reported here is solved I'll close this issue. If the solution is not working, please reopen this issue.

maxim-zhao, 2021-07-23:
Great, thanks! I'll try to use it soon...