vhelin/wla-dx #143: Define fails when using labels as parameters

KungFuFurby, 2017-05-27:
Using the absolute latest version of WLA-DX (commit ID 716f717020e5bdcf9ba22fa9f8f180bc801358d5), and this ASM file: [testDefinesUsingLabels.asm.zip](https://github.com/vhelin/wla-dx/files/1033614/testDefinesUsingLabels.asm.zip)

I was attempting to use defines that used labels as a parameter (I did not use the sizeof feature as I was unaware of it). The define was mis-identified as a label (as I found out when I looked at the .obj file, which listed these variables as labels... and they do not show up on the debug output when I activate it) and gave me an error as following (using testDefinesUsingLabels.obj):
```
testDefinesUsingLabels.obj:testDefinesUsingLabels.asm:52: FIX_REFERENCES: Reference to an unknown label "TestItem1ID".
```

This is the debug output from the assembler...
```
LINE    22: (STACK) CALCULATION ID = 0 (c0): S(TestItem1PTR), S(TestArray1PTR), -, V(3.000000), /
LINE    28: (STACK) CALCULATION ID = 1 (c1): S(TestItem2PTR), S(TestArray1PTR), -, V(3.000000), /
LINE    34: (STACK) CALCULATION ID = 2 (c2): S(TestItem3PTR), S(TestArray1PTR), -, V(3.000000), /
LINE    40: (STACK) CALCULATION ID = 3 (c3): S(TestItem4PTR), S(TestArray1PTR), -, V(3.000000), /
LINE    46: (STACK) CALCULATION ID = 4 (c4): S(TestItem5PTR), S(TestArray1PTR), -, V(3.000000), /
```

...and the linker...
```
**********************************************************************
**********************************************************************
**********************************************************************
*** LOADED LOADED LOADED LOADED LOADED LOADED LOADED LOADED LOADED ***
**********************************************************************
**********************************************************************
**********************************************************************

----------------------------------------------------------------------
---                         LABELS                                 ---
----------------------------------------------------------------------

label: "TestArray1PTR" file: testDefinesUsingLabels.obj status: 0 section: 1 bank: 0 slot: 0 base: 0 address: 0/$0
label: "TestItem1PTR" file: testDefinesUsingLabels.obj status: 0 section: 1 bank: 0 slot: 0 base: 0 address: 0/$0
label: "TestItem2PTR" file: testDefinesUsingLabels.obj status: 0 section: 1 bank: 0 slot: 0 base: 0 address: 3/$3
label: "TestItem3PTR" file: testDefinesUsingLabels.obj status: 0 section: 1 bank: 0 slot: 0 base: 0 address: 6/$6
label: "TestItem4PTR" file: testDefinesUsingLabels.obj status: 0 section: 1 bank: 0 slot: 0 base: 0 address: 9/$9
label: "TestItem5PTR" file: testDefinesUsingLabels.obj status: 0 section: 1 bank: 0 slot: 0 base: 0 address: 12/$c

----------------------------------------------------------------------
---                    (STACK) CALCULATIONS                        ---
----------------------------------------------------------------------

----------------------------------------------------------------------
stackitem: label              : TestItem1PTR
stackitem: label              : TestArray1PTR
stackitem: operator           : -
stackitem: value              : 3.000000/$3
stackitem: operator           : /
id: 0 file: testDefinesUsingLabels.obj line: 22 type: 3 bank: -123456 position: 0
----------------------------------------------------------------------
stackitem: label              : TestItem2PTR
stackitem: label              : TestArray1PTR
stackitem: operator           : -
stackitem: value              : 3.000000/$3
stackitem: operator           : /
id: 1 file: testDefinesUsingLabels.obj line: 28 type: 3 bank: -123456 position: 0
----------------------------------------------------------------------
stackitem: label              : TestItem3PTR
stackitem: label              : TestArray1PTR
stackitem: operator           : -
stackitem: value              : 3.000000/$3
stackitem: operator           : /
id: 2 file: testDefinesUsingLabels.obj line: 34 type: 3 bank: -123456 position: 0
----------------------------------------------------------------------
stackitem: label              : TestItem4PTR
stackitem: label              : TestArray1PTR
stackitem: operator           : -
stackitem: value              : 3.000000/$3
stackitem: operator           : /
id: 3 file: testDefinesUsingLabels.obj line: 40 type: 3 bank: -123456 position: 0
----------------------------------------------------------------------
[... 71 lines cut ...]
id: 2 file: testDefinesUsingLabels.obj line: 34 type: 3 bank: -123456 position: 0 result: 2/$2
----------------------------------------------------------------------
stackitem: value              : 9.000000/$9
stackitem: value              : 0.000000/$0
stackitem: operator           : -
stackitem: value              : 3.000000/$3
stackitem: operator           : /
id: 3 file: testDefinesUsingLabels.obj line: 40 type: 3 bank: -123456 position: 0 result: 3/$3
----------------------------------------------------------------------
stackitem: value              : 12.000000/$c
stackitem: value              : 0.000000/$0
stackitem: operator           : -
stackitem: value              : 3.000000/$3
stackitem: operator           : /
id: 4 file: testDefinesUsingLabels.obj line: 46 type: 3 bank: -123456 position: 0 result: 4/$4
----------------------------------------------------------------------

----------------------------------------------------------------------
---                          REFERENCES                            ---
----------------------------------------------------------------------

testDefinesUsingLabels.obj:testDefinesUsingLabels.asm:52: FIX_REFERENCES: Reference to an unknown label "TestItem1ID".
```

Note that the defines do not show up in the debug output.

vhelin, 2017-06-04:
Currently WLA expects you to .export those definitions, otherwise WLALINK doesn't get them. Why WLA doesn't replace TestItem1ID and the others with their contents, I'm not sure, don't remember any more how that part of WLA works.

KungFuFurby, 2017-06-04:
Thanks! That actually solved my problem... any definitions that use labels need to be exported (explaining their presence in the .obj file, and not anywhere else), even if it's a one-file case (this means the readme seems to be off on how .export works... either that, or I didn't quite understand how it worked the first go round).

Here's my working ASM file (simply by adding .exports)...
[testDefinesUsingLabelsBF.asm.zip](https://github.com/vhelin/wla-dx/files/1050365/testDefinesUsingLabelsBF.asm.zip)

Feel free to compare the two. Unusually, I have never had to use this with anything else (and exporting them actually gives them another useful feature for me: adding them to the definitions section of the .sym table), even in multi-file cases when I use .include and put some definitions in a separate file to use for multiple cases (then again, I'm also not referencing labels). This may be partially because I don't actually use the library feature (or multiple object files).

vhelin, 2017-06-04:
The question remains: why WLA doesn't replace the definition name reference with calculation index, and the user needs to .export the calculation definition? With value definitions replacing works...

But good that .export's work as a workaround for you. :)

vhelin, 2017-06-05:
Now this should work without an .export.

input_number() was made to return the calculation definition name instead of its index, thus an .export was needed, but this is the case no longer. .DB, and many other directives, already could process calculation definition indices. Why this was left half done - I have no idea.

I hope the fix works all around the code as there might be some places that don't handle calculation definition indices, only their names...

closed 2017-06-05 by vhelin