vhelin/wla-dx #146: Suggestion: A meta-label to reference the 'current' label

Kroc, 2017-07-04:
The purpose of this is to be able to reference (especially with calculations) the current (or parent, in the case of sub-labels) label without having to repeat the label's name.

Here's an example:
(I use `__label__` for the intended feature, but the syntax could be anything else)

```asm
bigloop:
    ...
    jp __label__
```
With sub-labels, this would allow jumping back to the parent label

```asm
bigloop:
    ...
@x:
    ...
@y:
    ...
    jp __label__
```

This gives the code more flexibility to be moved around / renamed.

Whilst "nice to have", it becomes more critical with calculation of offsets from the beginning of a label, i.e. a data table.

```asm
dataTable:
    .DB 1, 2, 3, 4, 5
@a
    .DB 6, 7, 8, 9, 10
    .DW @a - __label__
```
(this occurs in music data to specify loop points)

You might also consider `__sublabel__`, __section__` , but the exact choice of syntax for this feature will need some discussion

vhelin, 2021-03-19:
It's been a few years (sorry this took so long) and nobody objected to your suggestion, so I went and added support for __label__.

vhelin, 2021-04-29:
No bug reports after the addition of __label__ so I assume it's working. :)

closed 2021-04-29 by vhelin