vhelin/wla-dx #151: Compiler warnings

Drenn1, 2017-07-11:
I'm getting about 5 kazillion warnings when building with gcc 7.1.1. Some of them are a bit concerning. This is just from wla-gb by itself... https://pastebin.com/ECFFVhMw

I guess these are new warnings that weren't in older gcc versions? Some of them may be caused by the name length increase, but reverting that still leaves a lot of warnings.

A particular highlight:

```
pass_1.c:1825:6: note: ‘sprintf’ output 3 bytes into a destination of size 2
      sprintf(tmp_a, "%c%c", label[o-1], label[o]);
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

vhelin, 2022-08-29:
I don't think I ever saw these warnings. Do you still get them?

vhelin, 2023-02-28:
GCC aside I'm building WLA DX for Amiga using an old SAS/C compiler from the 90's. Old compilers, and different compilers in general can give you warnings that other compilers don't give, thus the more different compilers you use to compile a project the better. I've already fixed quite a few warnings that SAS/C gave me, but modern GCCs did not...

vhelin, 2023-02-28:
It's been years since this issue was posted. I'm curious about the current situation. Anyone here care to report what's the warning list today with gcc 7.1.1? I bet 7.0 would be as good...

Stewmath, 2023-03-01:
This is an old issue, it's entirely possible that the warnings I mentioned in this issue were fixed. I don't think you need to worry too much about gcc 7.1.1 specifically, that was just the latest version at the time.

With the current version of gcc, I only get one warning. I guess it's worth looking into.

```
/home/matthew/hdd/3rdprograms/wla-dx/parse.c: In function 'get_next_token':
/home/matthew/hdd/3rdprograms/wla-dx/parse.c:1445:35: warning: writing 1 byte into a region of size 0 [-Wstringop-overflow=]
 1445 |     g_current_directive[g_ss - 1] = 0;
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~
/home/matthew/hdd/3rdprograms/wla-dx/parse.c:28:32: note: at offset [2046, 2147483646] into destination object 'g_current_directive' of size 256
   28 | extern char *g_buffer, *g_tmp, g_current_directive[256], *g_label_stack[256];
```

(Side-note: I have changed my username, I was formerly Drenn)

vhelin, 2023-03-02:
I think I know where that warning originates from: we "recently" upped MAX_NAME_LENGTH from 255 to 511+ and g_current_directive is only 256 in size so the index into g_current_directive (bounded by MAX_NAME_LENGTH) could in theory flow out of that array... I'll fix that asap.

vhelin, 2023-03-02:
In phase_1.c g_current_directive had size MAX_NAME_LENGTH + 1 so there was a size mismatch which perhaps was the reason for that warning? Let's see, I'll push a commit soon...

vhelin, 2023-03-02:
@Stewmath Do you still get that warning with the latest sources?

vhelin, 2023-03-02:
Looking at the Linux build log I can only see warnings about ignored fscanf() return values. Fixing them (if less or equal to zero then print error and return FAILED?) should be quite easy, but would take time as there is quite a few of them.

vhelin, 2023-07-17:
Just tried to find these warnings or the code that produced them according to the log, but could not find tmp_a, xyz or emsg buffers any more in the code. If you get similar warnings with the latest code, please create an issue here.

closed 2023-07-17 by vhelin