vhelin/wla-dx #156: Line joining

maxim-zhao, 2017-08-04:
The inverse of #141, some syntax to act as a line break without actually being one. Some assemblers use a slash for this, for example [in TASM](http://jgmalcolm.com/z80/advanced/asse#back):

> `_arg1_ / _arg2_` - This treats the commands of _arg1_ and _arg2_ as if they were on separate lines.
> ```
> ld b,6 / ld c,4
> call _clrLCD / xor a
> 
> ; the above is the same as the below
> 
> ld b,6
> ld c,4
> call _clrLCD
> xor a
> ```

nicklausw, 2017-08-05:
So how does WLA figure out that `ld b,6 / ld c,4` is two separate statements and doesn't just try and divide 6 by `ld c,4`?

The only way I've seen line splitting implemented is in bass, which uses a semicolon to split and // for comments. For obvious reasons that's out of the picture, so some other symbol would have to suffice. The colon's uses are already fairly ambiguous, so that probably wouldn't work.

maxim-zhao, 2017-08-07:
I agree that it is quite a pain to overload anything for this purpose, but I suspect with a real lexer/parser it may be possible to disambiguate in the case where both sides of the chosen character are well-formed statements. I guess TASM doesn't use `/` to mean anything else (i.e. you can't do division). Is `\` free? Maybe it can do double-duty for this and #141...

nicklausw, 2017-08-09:
If we let \ be both the split and join symbol, there should be a clear rule as to how its function is decided.

Maybe just let it join lines if \ is the last symbol on the line, not counting spaces, and if not then make a new statement (I feel like someone's already suggested that exact thing, but might be thinking about the cc65 repo).

We'd need a line queue for the parser, but other than that this doesn't sound difficult to implement.

vhelin, 2023-08-26:
I forgot to mention this issue in the commit, but if you use the latest sources " \ " between instructions on the same line should work.

closed 2023-08-26 by vhelin