vhelin/wla-dx #159: Generating makefile dependencies and object files at the same time

Drenn1, 2017-08-07:
This boils down to a simple question: why must "test mode" be enabled when generating makefile dependencies? (As far as I can tell, all it does is disable the generation of the final output file.)

```
    else if (!strcmp(flags[count], "-M")) {
      makefile_rules = YES;
      test_mode = ON;
      verbose_mode = OFF;
      quiet = YES;
      continue;
    }
```

Rather than compiling my project twice - once to update dependencies, and once to generate the object file - I could kill two birds with one stone, and do both at once, if "test_mode" was not on.

I suppose the original concern was that the program's output shouldn't be mixed with the makefile output, therefore, makefile output should be a distinct operation. Which is also why quiet mode had to be enabled... If everything else printed to stderr, though, that wouldn't be a problem. This is already mostly the case, though there are a few exceptions.

So, my proposal is to make everything print to stderr, except makefile generation. Then, disable the overrides for "test_mode", "verbose_mode", and "quiet" when generating makefiles. Thoughts?

vhelin, 2022-10-11:
I hope the latest sources fix this issue. Also I didn't touch verbose levels/mode so that's left for the user to handle...

maxim-zhao, 2022-11-10:
The `-o` or `-l` parameter is needed at makefile generation time to specify the target (the very first text output). Thus its existence is insufficient to trigger also generating the file.

For me, the generated .o is basically nonsense because I'm using the `-M` mode to trigger the creation of many inputs to that .o, and thus it has to be rebuilt anyway.

Thus, please can we have a new flag to enable request? It comes out OK for me anyway because the .o timestamp ends up staler than its own inputs, but it seems like wasted effort.

See also #539 wherein I stepped back some of the changes mentioned above already because they were enabling messages to stdout during makefile generation.

vhelin, 2022-11-10:
> Thus, please can we have a new flag to enable request?

I don't think I understand what you mean here. :)

vhelin, 2022-11-10:
Do you want to use -M and -o/-l at the same time, but not create an object or a library file?

maxim-zhao, 2022-11-11:
Yes - I specify `-M` to emit makefile rules, in the form
```make
<output file>: <input 1>\
  <input 2> \
  <input 3> # etc
``` 
But the output filename depends on whether we're building a library or object file, and the filename will default if not specified. So I have to pass the `-o` parameter to tell it what name to use - regardless of whether I want it to produce that object file yet. In fact, in the case where the makefile generation mode had to "fake" missing inputs, the object file is nonsense. So it seems like it ought to be optional to produce it.

vhelin, 2022-11-11:
If you use -t (test mode) with -M and -o/-l, does that work? Looking at the code when test mode is turned on no object or library file is written...

vhelin, 2022-11-11:
It seems that -M turns test mode on, but -o/-l turns it off, but if you add -t it should override all and turn test mode on permanently for that run.

maxim-zhao, 2022-11-11:
That does the trick - although it's a change of behaviour compared to what happened before the changes from this issue. However, it works for me.

closed 2023-05-04 by vhelin