vhelin/wla-dx #160: Arithmetic across macros segmentation fault

lf94, 2017-09-08:
I had what's similar to this, segment fault wlalink:

```
.ramsection ""
wAlpha dsb 20
.ends

.section ""
.macro MyMacro
ld hl, (\1+1)
.endm

MyMacro (wAlpha+1)
.ends
```

The error is "A computation stack has been lost!"

vhelin, 2018-04-17:
Can you give us a complete project zip etc? I tried, but the previous code compiled just fine...

vhelin, 2019-05-23:
> The error is "A computation stack has been lost!"

I cannot find such an error in WLA or WLALINK... Could you shed some more light on this issue?

Neui, 2019-05-23:
The closest message I've found is [`"COMPUTE_STACK: A computation stack has gone missing. This is a fatal internal error. Please send the WLA DX author a bug report."`](https://github.com/vhelin/wla-dx/blob/master/wlalink/write.c#L1585)

vhelin, 2021-03-14:
I tried the example again and it assembled just fine, so I'll close this issue. If you have a better example that exhibits the bug the please reopen the issue.

closed 2021-03-14 by vhelin