vhelin/wla-dx #162: Documentation generation fails (doc-stext only)

Ramsis-SNES, 2017-11-01:
```
Scanning dependencies of target doc-stext
[100%] Generating simple text documentation

Exception occurred:
  File "/home/user/dev/wla-dx-master/doc/sphinx/singletext.py", line 68, in write
    mastertree, darkgreen)
TypeError: inline_all_toctrees() takes exactly 6 arguments (5 given)
The full traceback has been saved in /tmp/sphinx-err-HbhXZM.log, if you want to report the issue to the developers.
Please also report this if it was a user error, so that a better error message can be provided next time.
A bug report can be filed in the tracker at <https://github.com/sphinx-doc/sphinx/issues>. Thanks!
doc/CMakeFiles/doc-stext.dir/build.make:82: die Regel für Ziel „binaries/doc/wla-dx.txt“ scheiterte
make[2]: *** [binaries/doc/wla-dx.txt] Fehler 1
CMakeFiles/Makefile2:1166: die Regel für Ziel „doc/CMakeFiles/doc-stext.dir/all“ scheiterte
make[1]: *** [doc/CMakeFiles/doc-stext.dir/all] Fehler 2
Makefile:127: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2
```
I'm running Linux Mint 18.2 and have Sphinx v1.3.6-2ubuntu1.1 installed. What could be the issue here?

Thanks!

vhelin, 2018-04-17:
Well, I'm running Cygwin (latest, v2.891, 17-Apr-2018) under Windows 10, and cannot generate the docs. Help! :) I have python2-sphinx 1.6.5-1 installed.

/bin/sh: /usr/bin/sphinx-build: No such file or directory
make[2]: *** [doc/CMakeFiles/doc-stext.dir/build.make:82: binaries/doc/wla-dx.txt] Error 127
make[1]: *** [CMakeFiles/Makefile2:1238: doc/CMakeFiles/doc-stext.dir/all] Error 2
make: *** [Makefile:128: all] Error 2

Ramsis-SNES, 2018-04-17:
It's been working for me for quite some time now (including today's codebase). My workstation currently runs Kubuntu 17.10 with Sphinx 1.5.6-2 though, in case that matters. ;-)

vhelin, 2021-05-03:
It has happened to me twice that an update to Cygwin/Python has broken Sphinx/docs generation, but after some times / more updates I've gotten the docs to compile once again. Personally I cannot offer any support with docs generation as I have no idea how it works and I've never used Sphinx myself, other people integrated Sphinx to WLA DX, but if you run into such troubles again in the future posting an issue here doesn't cost much. :)

closed 2021-05-03 by vhelin