vhelin/wla-dx #165: Add checksums to test cases.

Drenn1, 2018-01-06:
Just a thought I had; just because a test builds successfully, doesn't mean it's actually working properly, right? It would be easier to quickly verify that nothing was broken if the makefiles for the tests added a verification step with the md5sum command. Or perhaps we could make a copy of the output rom and run a diff on it.

This could get annoying when working on stuff that's expected to change the output, though, like certain linker features.

vhelin, 2018-04-17:
You are correct in that if a test/example project builds it doesn't mean that it works. And you are also correct in that such a checksum would mean that if the linker makes different but working outputs later the checksum would change even if the output would work 100%...

vhelin, 2018-04-17:
I don't know how to fix this. Perhaps we need to make the tests/examples such that they are run through an emulator and make them produce some output that can be checksummed / verified to be working. The binaries themselves can be very different, but still produce working outputs...

Drenn1, 2018-04-21:
My 2 cents right now: it's probably worth knowing when a program is assembled differently, even if it's expected. Then you'll know that you probably need to test the ones that changed, and update their checksums if they look good.

Automated testing with an emulator sounds very much like overengineering, especially as every platform would need a different one...

vhelin, 2018-04-21:
Drenn1, you are right, I think. It doesn't happen very often that the checksums of examples change, so checking the changed ROMs / program files should not be a huge task. Sounds good!

vhelin, 2019-01-12:
Recently I've been putting some small parts (few lines) of the code in the examples directory between some binary markers like "BEGIN>>>" and "<<<END". After I assemble the sources it's pretty easy to compare the resuting binary in a hex editor with the sources/what it should produce. If we don't add checksums to resulting ROM files (pretty rigid), perhaps such a small scale partial ROM testing could be made?

vhelin, 2021-03-19:
As of today I've added quite a few tests where blocks of bytes (produced by assembling source code) are between markers like "01>" and "<01" and then we have byte_tester.exe that tests that the bytes between the markers are what we are expecting. These tests are done when executing run_tests.sh. It's like checksumming (important) parts of a test.

Checksumming the whole test would lead to very rigid tests, for example relocating a FREE section would break such checksums. 

I think we can close this issue as we have a working solution to this problem.

closed 2021-03-19 by vhelin