vhelin/wla-dx #166: Getting bank number with ":CADDR" is broken

Drenn1, 2018-02-07:
Something broke after the commits relating to the ".BASE" directive (after 9841ae688c8274d150a7bff2a1c385d4f058d2df). I have a line:

```
ld a,:CADDR
```

and it seems to be giving me an undefined value - that is, it's different every time, whether or not the ".BASE" directive was used. Any idea what may be causing this, @SolraBizna?

Drenn1, 2018-02-07:
If I go to wlalink/write.c and hardcode the base number to "0" for the CADDR case, the problem is fixed. However I suspect it should use the value of the last ".BASE" directive instead. This is a weird case since it's a reference that doesn't have a corresponding label anywhere, so the value of ".BASE" for it doesn't seem to be stored anywhere.

Drenn1, 2018-02-07:
Here's a sample project. I wasn't getting the undefined behaviour at first, but that appeared after adding another bank reference below it.

[wla_base_bug.zip](https://github.com/vhelin/wla-dx/files/1704341/wla_base_bug.zip)

Also, I found a bonus bug for you. I noticed that "base" is still wrapped in a W65816 define under the "stack" struct in defines.h. This means arithmetic won't be handled properly. I added a line for that in the test project above.

SolraBizna, 2018-02-07:
Valgrind confirms it. When `fix_references` creates a fake label for `CADDR` to link against, it doesn't initialize `base`. I'll fix that and the stack issue now, and while I'm at it I'll look at every W65816 case in the whole codebase to see if they should be promoted to general cases.

SolraBizna, 2018-02-07:
Your test case was very helpful. Thanks.

#167 fixes all the remaining oddness I could find. As I mention in the description there, I couldn't find a better way to fix this particular bug than setting `base` to 0 when referencing `CADDR`. A more correct fix would require breaking a rather non-future-proof on-disk format. For now, if anyone wants to use a non-zero `.BASE` *and* wants a correct `:CADDR`, they can address their hate mail to me. :)

vhelin, 2018-05-07:
On-disk object/library file format can be broken, if you ask me, if it makes a bug fix or a nice feature possible; it's quite easy and fast to recompile all assembly sources, so there are no big reasons why not to change the file formats...

vhelin, 2019-05-28:
> If I go to wlalink/write.c and hardcode the base number to "0" for the CADDR case, the problem is fixed. However I suspect it should use the value of the last ".BASE" directive instead. This is a [...]

I found multiple instances where base was set to 0, but if the reference was inside a section, then we could just copy the section's .BASE to be reference's .BASE as well? It would also be possible to put the reference's .BASE at assembling time into object and library files, but that's just waste of space if the .BASE is constant inside a .SECTION... ?

I'll try to submit a fix to this issue later today, got still some work to do as :CADDR inside computations is also broken...

vhelin, 2019-05-28:
Hey, can you test the latest sources? :CADDR should work once again...

vhelin, 2019-06-02:
This is one of the bugs I want to fix before releasing WLA DX v9.8 final... So the faster anyone can test this and close, the faster we'll have v9.8 final out. :)

vhelin, 2019-06-04:
If anyone could test this (and possibly close) before 8th of June 2019, that'd be great as I'll be at my parents' cottage then and could compile final v9.8 binaries for Windows. If the v9.8 final relase goes beyond that I don't know when I'm going to visit them... They have a Windows 7 computer I'd use to make the exes. I only have a modern Windows 10 computer at home myself.

SolraBizna, 2019-06-08:
I can confirm that @Drenn1's test case works correctly after 1228268.

vhelin, 2019-06-20:
Cool, thanks! :) I'll close this one so we'll get final v9.8 out soon.

closed 2019-06-20 by vhelin