vhelin/wla-dx #168: WLA-6510 min object file 8K

oziphantom, 2018-03-14:
If you assemble with WLA-6510 it will make an 8K .o file, this means that you can't "link" anything else into said 8K area as the linker thinks it is used.

I'm waiting from my AA to approve WLA-6502 to check it as well.

oziphantom, 2018-03-14:
WLA-6502 does also do it

vhelin, 2018-04-17:
Can you provide a project that does this? Doesn't the output size depend on your .ROMBANKMAP?

oziphantom, 2018-04-17:
The 6502 file "viceTest.asm"

Drenn1, 2018-04-22:
I've often found that feature to unexpectedly get in the way of naming. I don't suppose it would be possible to change it to an explicit "SIZE" parameter or something? Though I'm not normally one to request compatibility-breaking changes...

vhelin, 2018-04-22:
I have to admit that the current way section is given a specific size is pretty confusing. Optional SIZE parameter would be much better even if we broke compability, I think. Perhaps we should start using SIZE and give a warning when the size is defined as currently supported, and later when we release the next version of WLA DX we'd remove the warning? Or do you guys think of a better way to handle this?

vhelin, 2018-04-22:
Ok, let's do it so that when we release the next version (9.8) of WLA DX we'll remove the warning (that needs to be done along with SIZE). Looking at how long the latest version (9.7) was in beta (or limbo...) users should have plenty of time to fix their projects. 😁

vhelin, 2019-06-20:
Ok, I've removed the warning, deprecation period is over. Release of final v9.8 is imminent.

closed 2019-06-20 by vhelin