vhelin/wla-dx #169: .ORG can not take an address label

oziphantom, 2018-03-14:
You can't give .ORG a memory location to place code at. 
So file 1
Some Code
PutThingHere

file 2.
.ORG PutThingHere

file 2 fails to assemble. 

complete example
```
.MEMORYMAP
DEFAULTSLOT 0
SLOTSIZE $10000
SLOT 0 $0000
.ENDME

.ROMBANKMAP
BANKSTOTAL 2
BANKSIZE $10000
BANKS 2
.ENDRO

.BANK 0 SLOT 0
.ORGA $1c00 ; start of 128 BASIC
.SECTION "CPU_8500" FORCE
.DB $00,$1c ; PRG header
; 10 sys 7184
.DB $00,$0C,$1C,$0A,$00,$9E,$20,$37,$31,$38,$34,$00,$00,$00,$00,$00

; ORG $1c10
sei			; disable interupts
lda #$BE
sta $FF00	; bank 2 RAM + IO
lda #195	; jmp
sta $ffee 
lda #<Z80Start
sta $ffef
lda #>Z80Start
sta $fff0
lda #$b0
sta $d505   ; this makes me go to the Z80
nop ; do nothing
inc $d020
-	jmp - 
Z80Start 
.ENDS
```

```.MEMORYMAP
DEFAULTSLOT 0
SLOTSIZE $10000
SLOT 0 $0000
.ENDME

.ROMBANKMAP
BANKSTOTAL 2
BANKSIZE $10000
BANKS 2
.ENDRO

.BANK 0 SLOT 0
.ORGA Z80Start
.SECTION "CPU_Z80" FORCE
ld a,0
ld bc,$d020
out (c),a
inc bc
out (c),a
ld hl,messagetxt
ld de,$0400
ld bc,20
ldir
- jp -
messagetxt .asc "hello world z80 here"
messagelen 
.ENDS```

vhelin, 2018-04-17:
Well, that's by the original design. .ORG and .ORGA needs to be solved first, very early, and much later after the labels are placed into memory it is just too late to handle .ORG and .ORGA... I'm not sure how to fix this. Might require a huge rewrite of WLA DX...

oziphantom, 2018-04-17:
well I presume the linker can relocate the code. To which you assemble the code for well anywhere, $0000 is convenient ;) then the Linker relocates it to the label.

Drenn1, 2018-04-17:
For this particular use case, you could use the "APPENDTO" feature for sections.

I can't think of any other use case for this. This would require the linker to identify which labels are used in an ORG, then place those sections first; we'd need to detect circular references.

In general, it's weird to ORG to an already defined label. In 99% of cases, that label will already have data defined after it.

oziphantom, 2018-04-17:
Looking at the docs APPENDTO is within an assembler, can I "APEPNDTO" across objects during a link? And will it assemble with it being told to APPENDTO a section not defined in the asm files it is assembling?

Drenn1, 2018-04-17:
I haven't tried the feature, but the object file format has data specifically for append sections, so yes that should work.

oziphantom, 2018-04-18:
So I added what I needed to a Linux VM for a test. Seems APPENDTO does do what is needed mostly. Just need to get rid of the 6502 file being 8K min ;)

vhelin, 2018-04-22:
I compiled and linked the example. Changed the line
  .ORGA Z80Start
to
  .ORGA 0
to make it compile. The output might not work in real life, though. :) Anyway...

If I linked this project as a program file, the resulting program file was 8500 bytes as program files contain only the data from byte 0 to the highest used byte in the ROM banks. Not sure if anyone uses this mode as this kind of program size calculation might not really work.

If I linked this project as a ROM file, the file size was 131072 as .ROMBANKMAP has two ROM banks each sized $10000 (64k). 

Next I changed the ROM banks to 256 bytes and adjusted the code a little, removed the "_8500" size from section called "CPU_8500", and the resulting ROM was 512 bytes. Linking it as a program file gave a file of 151 bytes...

I'm not sure where you get that 6502 files are min 8K... ???

oziphantom, 2018-04-22:
I changed the name and 376 bytes.

vhelin, 2018-04-22:
WLA has no min size for ROM files. The size of the ROM file is just what .ROMBANKMAP defines... But yeah, if you name a section "SECTION_1234" then the section's size will be 1234 bytes. That's a feature in WLA.

oziphantom, 2018-04-22:
so SECTION_LEVEL_DATA_1 and SECTION_LEVEL_DATA_2 are bad ideas then. So one should do 
SECTION_LEVEL_1_DATA and SECTION_LEVEL_2_DATA?
Also is something like SECTION_8500CPU safe?

vhelin, 2018-04-22:
I don't remember where that section size setting in section's name feature request came, but yeah, SECTION_LEVEL_DATA_1 is a bad name... :( SECTION_8500CPU is safe.

vhelin, 2018-04-22:
You can use SECTION_LEVEL_DATA_1, but then WLA will give a warning if the section is bigger than 1 bytes, and enlarges the section, so it works, but you'll get a warning...

vhelin, 2018-04-24:
This is something we cannot address at this moment. It would require a lot of work currently... Sorry. :(

closed 2018-04-24 by vhelin