vhelin/wla-dx #170: Add support for Motorola 68000? :feature_request

sverx, 2018-03-16:
As per the subject - do you think it's possible to add support for the Motorola 68000?
That would be handy for SEGA MegaDrive / Neo-Geo / Amiga / Atari ST development...
Thanks!

vhelin, 2018-04-17:
It would not be impossible, but all the currectly supported architectures are 8bit, so support for the 16/32bit 68000 would require quite a lot of more work than adding support for another 8bit CPU. I myself don't have the energy to do this, but if someone wants to help here I will give it my support.

oziphantom, 2018-04-17:
You could try VASM/VLINK - it is basically on par with WLA-DX but it has a lot of support for 68K based systems. Atari ST TOS, Jaguar, Amiga etc http://sun.hasenbraten.de/vasm/index.php

rdrpenguin04, 2018-04-20:
Objection, vhelin, the 65816 is not 8-bit.

vhelin, 2018-04-21:
Oops, my bad. Anyway, do you guys have any idea if there is a lot of demand for 68k support in WLA DX? For just a nice to have feature 68k support would require quite a lot of work..

sverx, 2018-05-03:
I don't know if there's a lot of demand - I would use that surely, as I'm going to try my teeth at 68000 asm coding for the MegaDrive...

rdrpenguin04, 2018-05-06:
I'd love to see 68K asm in WLA-DX.

vhelin, 2018-05-23:
There's one big obstacle in adding 68k support to WLA DX: 68k is big endian while all the currently supported CPUs are little endian... We'd need to touch a lot of places to add 68k support to WLA DX. :(

rdrpenguin04, 2019-03-10:
Maybe someday when I know a bit more about assemblers (and maybe finish one of my own) I can help with that. Another thing that may be an issue is that the instruction set for the 68000 is more packed than the other systems, often containing an opcode and two operands in one 32-bit word, which can be difficult to work around.

vhelin, 2019-09-06:
> There's one big obstacle in adding 68k support to WLA DX: 68k is big endian while all the currently supported CPUs are little endian... We'd need to touch a lot of places to add 68k support to WLA DX. :(

This obstacle has been removed when I added support for MC6800 CPU. :)

vhelin, 2019-09-06:
> Maybe someday when I know a bit more about assemblers (and maybe finish one of my own) I can help with that. Another thing that may be an issue is that the instruction set for the 68000 is more [...]

Support for 68000 might just require a more intelligent parser than what the current mnemonics parsers in WLA DX are... If you used the old parsers you'd need to add every permutation of the registers per a mnemonic on its own line in the opcodes struct and that would be a huge task and take a lot of time and space. 8-bit CPUs are very simple compared to 68000...

cr1901, 2019-09-06:
I can think of another snag for 68k, though I think this feature should be supported eventually anyway:

IIRC, internally WLA-DX works in units of 64kB banks max. This works for every supported CPU, and SuperFX (which I would still like to port one day). However, 68k can internally deal with up to a continuous 16MB unit of address space just fine, so the `ROMBANKMAP` and `MEMORYMAP` directives would need to accommodate.

vhelin, 2021-03-19:
Another obstacle to implementing 68000 (and 68020+?) support has been cleared: we can now define 32-bit data using .DD.

cr1901, 2021-03-19:
Is it possible to change `ROMBANKMAP` and `MEMORYMAP` to accommodate > 64kB banks max (without breaking everything)?

vhelin, 2021-03-20:
I tried this:

        .MEMORYMAP
           DEFAULTSLOT 0
           SLOTSIZE $200000
           SLOT 0 $0000
        .ENDME

        .ROMBANKMAP
           BANKSTOTAL 1
           BANKSIZE $200000
           BANKS 1
        .ENDRO

... and it assembled just fine. :)

vhelin, 2022-08-28:
As the release of v10.3 is near, I'm thinking that 68000 support should be added to v10.4. If you guys have links to webpages with descriptions of 68000's opcodes and instructions, please comment down below. I'm looking for such web pages that are easily understood and that contain all the information that one would need to implement the support for 68000.

So far I've found web pages with so "compressed" (well done, if you consider how one can compress information) information about 68000, that reading through them three times didn't make me any wiser...

theunixman, 2022-08-29:
Hey @vhelin , I've worked through 68k assembly by hand back in the day, I might be able to answer questions.

vhelin, 2022-08-29:
After learning how to program (BASIC) on a C64 when I was a kid I got an Amiga 500 (later 1200) and learned 680x0 assembly and coded some things including a GameBoy emulator. I spent around 6 years coding using just 680x0 asm. But that was ages ago, and the docs I find in the internet about 68000 opcodes are pretty obscure to me... :) Perhaps all this needs is just time. A lot of time...

sverx, 2022-08-29:
This seems a good document about the 68k assembly http://wpage.unina.it/rcanonic/didattica/ce1/docs/68000.pdf but unfortunately it doesn't show the instruction encoding.

This one instead has also the encoding https://web.njit.edu/~rosensta/classes/architecture/252software/code.pdf

sverx, 2022-08-29:
Also, but this is just my opinion, I would not rush this.

This is something that can be done while you're releasing v10.3, then v10.4, and just release it (as beta) when you think it's got to the point that it's (beta) ready.

vhelin, 2022-10-02:
@sverx Thanks for the infos! After looking at the documents you provided I think I have a hunch of how to implement 68000 support to WLA DX! I'm definitely going to try to do that in v10.4.

cr1901, 2022-12-06:
>Still a bit horrified about parsing all those complex addressing modes

That's how you're supposed to feel :D. Ditto for encoding the opcodes. Excellent work!

theunixman, 2022-12-06:
Wrote an emulator for this lovely architecture once, can confirm.

Once.

vhelin, 2022-12-06:
Switched from coffee to beer for this feature... :) I'm getting there, but the parser is pretty nasty. Then again, which parser in WLA DX isn't? I kinda hope I had initially tokenized the code in 1st pass, but yeah, not going to do that now, too big a change for me to implement and at this point it'd be way more work than I can do on my free time... I guess people want MC68000 support this year, not in 2024 LOL :)

vhelin, 2022-12-07:
Ok, I have a question: Is this correct?

[image]

I have the illusion that it'd be $D4 $7A $12 $34

cr1901, 2022-12-07:
This looks correct to me, since it's PC-relative, and PC-relative will calculate the effective address using the _PC value of the second word_ in the instruction.

`1000 + 2 + 0232` = `1234`

vhelin, 2022-12-07:
I'm adding support for both $1234(PC) and ($1234,PC). Or actually it's working already. :)

vhelin, 2022-12-07:
Next issue:

[image]

Why $FE and not $01? There's something I don't get here...

cr1901, 2022-12-07:
Let's start with the lower byte (I don't care about the high byte here):

`0x3B` => `xx_111_011b` => PC Indirect w/ 8-bit Displacement Addressing Mode.

EA = `(PC of second word in instruction) + (D2) + 8-bit sign-extended displacement` = `(0x10002 + D2 - 0x2)` = `10000 + D2`.

The displacement in the instruction should be `+0x02`, not negative.

vhelin, 2022-12-07:
I get that 20 here comes from d2 in the instruction, but base address for the offset is byte 20, not FE? Hmm, I think I need to try another assembler, I'm using EASy68K here...

cr1901, 2022-12-08:
From Motorola's Family Reference Manual, page 2-2:

[image]

_`SCALE` is `00` for 68000, so ignore it._

(My manual suggests `(d8,PC,Xn.SIZE)` for syntax; I'm going to go ahead and assume that in "real assemblers", `SIZE` can be inferred and defaults to sign-extended word)

`0x20FE`: Use `D2` as a register. Take `D2`'s lower 16-bits and sign-extend them. Sign-extend the displacement `0xFE` to 32-bits; this becomes `-2`.

vhelin, 2022-12-08:
I installed VASM and it gives here D6 3B 20 02 which validates what you are saying. So far I've used EASy68K for testing my output, I hope it hasn't been wrong elsewhere as well...

vhelin, 2022-12-08:
Fortunately EASy68K is on GitHub: https://github.com/ProfKelly/EASy68K/issues/3

vhelin, 2022-12-11:
Status update: We now have these working:

```

struct instruction g_instructions_table[] = {
  { "ABCD", B16(11000001, 00000000), 1, B8(00000000), MC68000_MODE_ALL },
  { "ABCD.B", B16(11000001, 00000000), 1, B8(00000000), MC68000_MODE_ALL },

  { "ADD.B", B16(11010000, 00000000), 2, B8(00000000), MC68000_MODE_ALL },
  { "ADD.W", B16(11010000, 00000000), 2, B8(00000001), MC68000_MODE_ALL },
  { "ADD.L", B16(11010000, 00000000), 2, B8(00000010), MC68000_MODE_ALL },

  { "ADDA.W", B16(11010000, 00000000), 2, B8(00000001), MC68000_MODE_A },
  { "ADDA.L", B16(11010000, 00000000), 2, B8(00000010), MC68000_MODE_A },

  { "ADDI.B", B16(00000110, 00000000), 2, B8(00000000), MC68000_MODE_I },
  { "ADDI.W", B16(00000110, 00000000), 2, B8(00000001), MC68000_MODE_I },
  { "ADDI.L", B16(00000110, 00000000), 2, B8(00000010), MC68000_MODE_I },

  { "ADDQ.B", B16(01010000, 00000000), 2, B8(00000000), MC68000_MODE_Q },
  { "ADDQ.W", B16(01010000, 00000000), 2, B8(00000001), MC68000_MODE_Q },
  { "ADDQ.L", B16(01010000, 00000000), 2, B8(00000010), MC68000_MODE_Q },

  { "ADDX.B", B16(11010001, 00000000), 1, B8(00000000), MC68000_MODE_ALL },
  { "ADDX.W", B16(11010001, 01000000), 1, B8(00000001), MC68000_MODE_ALL },
  { "ADDX.L", B16(11010001, 10000000), 1, B8(00000010), MC68000_MODE_ALL },

  /* 4A FC */
  { "ILLEGAL", B16(01001010, 11111100), 0, 0, MC68000_MODE_ALL },

  /* 4E 71 */
  { "NOP", B16(01001110, 01110001), 0, 0, MC68000_MODE_ALL },

  /* 4E 70 */
  { "RESET", B16(01001110, 01110000), 0, 0, MC68000_MODE_ALL },

  /* 4E 73 */
  { "RTE", B16(01001110, 01110011), 0, 0, MC68000_MODE_ALL },

  /* 4E 77 */
  { "RTR", B16(01001110, 01110111), 0, 0, MC68000_MODE_ALL },

  /* 4E 75 */
  { "RTS", B16(01001110, 01110101), 0, 0, MC68000_MODE_ALL },

  /* 4E 76 */
  { "TRAPV", B16(01001110, 01110110), 0, 0, MC68000_MODE_ALL },

  { "E", 0x100, 0xFF, 0, MC68000_MODE_ALL },
};

```

WLA can automatically turn ADD -> ADDA, ADDI and ADDQ depending on the case, and I should be able to reuse the code I made for ADD with e.g., SUB.

Unfortunately I have to dedicate my free time this month on other things, but during my Christmas holiday (25.-28.Dec.2022) I will be free to work on WLA DX so expect big progress on the MC68000 support then. I think I've managed to solve the biggest parsing issues so the rest should be smooth sailing. I hope. :)

vhelin, 2022-12-11:
It seems that these are the same

[image]

VASM also eats "ADDQ" without .W so does this mean that every time the user doesn't specify the size it should default to .W?

vhelin, 2023-01-01:
Ok, MC68000 support is now in. All feedback is welcome. Next I think I'll need to try to check how to create Megadrive ROMs etc, perhaps they need headers etc...

vhelin, 2023-01-01:
Would it be confusing if I made .DL to define 32-bit words instead of 24-bit words in WLA-68000? Then .DD would define 24-bit words...

Just thinking as .L means a 32-bit word when talking about MC68000 instructions...

sverx, 2023-01-01:
> Next I think I'll need to try to check how to create Megadrive ROMs etc, perhaps they need headers etc...

Here all the info you need about that -> https://plutiedev.com/rom-header

cr1901, 2023-01-01:
I personally would expect `.DL` to define 32-bit words for 68000; AIUI, you can't really load/store 24-bits at a time in a single 68k instruction (which idly makes me wonder: it must've been easy to accidentally destroy [fat pointers](https://en.wikipedia.org/wiki/Classic_Mac_OS_memory_management#32-bit_clean) in the original MacOS...).

vhelin, 2023-04-29:
> But is there any precedent for directives to change their meaning depending on arch?

I don't think there is...

Ok, yeah, let's keep .DL 24-bits and .DD 32-bits for the sake of consistency.

Also as I've gotten zero feedback how the MC68000 support works in WLA DX and my tests pass I assume it's working well and I'm closing this issue. Feel free to open new ones if you spot any bugs.

closed 2023-04-29 by vhelin