vhelin/wla-dx #172: wla-dx v9.7?

Drenn1, 2018-04-14:
There've been a fair few changes since v9.6, I think it would be helpful to have a new version so we can use the new featurtes without people needing to build the latest source.

Kroc, 2018-04-14:
Ditto; and please ensure the docs are up to date.

vhelin, 2018-04-17:
I published v9.7. When one makes changes to the codebase he should also
make sure the docs are up to date. Updating the docs later, after all the
small and big commits, is a pretty scary and uncertain task especially if
one or two years have passed since the latest release...

closed 2018-04-17 by vhelin

Drenn1, 2018-04-17:
Thanks Ville, you've been busy today I see. :)

vhelin, 2018-04-21:
Well I was a bit drunk and felt I had the energy to do something. I hope I didn't screw things up much. :) Forgot to update CHANGELOG for the release, but did that right after...