vhelin/wla-dx #175: Online documentation / documentation with binary releases

Kroc, 2018-04-26:
Hi there. I intend to use WLA DX for a disassembly project (6502) and there's no where I can just link people to to read the documentation. This makes it nearly impossible to get people up to speed with WLA DX syntax.

Additionally, proper release packages should be made with binaries and the compiled documentation in place.

vhelin, 2018-05-02:
Could someone generate such release packages? For me the documentation generation fails currently... And I can only build using Cygwin...

vhelin, 2018-05-12:
I managed to compile the latest sources with Visual Studio Express 2012 under Windows 7 yesterday, and will put the binaries online later today if I have the time, but I cannot build the documents... Could anyone here help and generate the documentation for this release package? Thanks!

vhelin, 2018-05-13:
I solved my problem with sphinx: I uninstalled the cygwin packages of sphix, deleted all traces of it from the disk, and then used pip to install sphinx from the Interner - now I can generate the docs again.

I also put the Windows7 / Win32 binaries of WLA DX v9.8a here: http://www.villehelin.com/wla-win32.html (contains the documents).

vhelin, 2018-05-13:
I also put the latest documentation (README.html) here: http://www.villehelin.com/wla.html

maxim-zhao, 2018-08-15:
The loss of README.txt is a bit frustrating for me. Can sphinx generate a plain text version of the documentation? I couldn't figure out how to run it myself from the source.

cr1901, 2018-08-15:
Perhaps we should set up CI (Travis should be fine for both Linux and Windows binaries) that generates the docs and binaries as output each commit?

Not exactly sure how to deploy with Travis though.

Neui, 2018-08-15:
I've included some CI support in an (currently stalled) PR of mine for [Travis](https://github.com/vhelin/wla-dx/pull/164/commits/d245bb8bfc18430c28690e5c6d4393565b6d037d), [AppVeyor](https://github.com/vhelin/wla-dx/pull/164/commits/5519b5c51f63d6bb0f4503f510d1bd8b1b835bea) and [GitLab CI](https://github.com/vhelin/wla-dx/pull/164/commits/2feab1c0ecd338e4c32affd63a520d9b1076ea97). Note that travis can upload to GitHub Releases (but then I'd say only use for releases and not dev builds). For others, you need Amazon S3.

(Althrough they might be adjusted to upload the documentation seperately)

Anyway, [here is a generated text version](https://pastebin.com/raw/2vnfbpWN). It's a bit older, but hopefully still useful.

lhsazevedo, 2021-01-13:
I guess this can be closed now.

vhelin, 2021-01-13:
The README included in the release zips (unsupported) are generated from the Sphinx data files, and are thus a bit crappier than the original manually updated README (text file), some parts of the formatting sucks etc., but they do the same thing. So I guess you are right!

closed 2021-01-13 by vhelin