vhelin/wla-dx #176: MAX_NAME_LENGTH inconsistencies

Drenn1, 2018-04-26:
Just a simple question: does MAX_NAME_LENGTH include the null terminator or not? It seems inconsistent. wlalink has "MAX_NAME_LENGTH+1" in defines.h, while the assembler has just "MAX_NAME_LENGTH". There are places where it checks if "len > MAX_NAME_LENGTH-1", but also "len > MAX_NAME_LENGTH".

I guess someone should decide which it is and go through the code, to make sure buffer overflows don't happen.

vhelin, 2018-05-01:
I remember that MAX_NAME_LENGTH doesn't include the null terminator, thus +1 checks in many places. I could be wrong, though, but this is what I think went in my mind when I put that MAX_NAME_LENGTH there... :) 

I can go through the code perhaps tomorrow?

vhelin, 2018-05-01:
Just took a look, and in many places MAX_NAME_LENGTH is assumed to include the null terminator, but not all... I'll fix this.

vhelin, 2018-05-01:
Made it so that MAX_NAME_LENGTH doesn't include the null terminator. I hope it's more consistent now. :)

closed 2018-05-06 by vhelin