vhelin/wla-dx #185: Feature request: Make .DSB accept defined values instead of just plain numbers

Ramsis-SNES, 2018-07-07:
Let's say I have self-modifying routines that get copied to a .STRUCT in WRAM. The usual way of defining the space needed is like so (example from my WIP SNES game):

```
.STRUCT ram_code_block
	DoDMA			dsb 49
	UpdatePPURegs		dsb 121
.ENDST
```
Currently, the only way to get the correct numbers is either taking them from the SYM file, or counting manually. 

Now, let's say the skeleton code for these routines in ROM (i.e., what gets copied to WRAM at some point) resides under the labels ``SRC_DoDMA`` and ``SRC_UpdatePPURegs``, respectively. It would be so much easier to just use ``_sizeof_`` definitions in the .STRUCT, e.g.

```
.STRUCT ram_code_block
	DoDMA			dsb _sizeof_SRC_DoDMA
	UpdatePPURegs		_sizeof_dsb UpdatePPURegs
.ENDST
```
Thanks in advance for considering this! :D

vhelin, 2018-09-13:
.define val49 49
.define val121 121

.STRUCT ram_code_block
	DoDMA			dsb val49
	UpdatePPURegs		dsb val121
.ENDST

This (using defined values) should already work, but if you want to use WLALINK's automatically generated _sizeof_*'s with .STRUCTs, then we'd need to somehow make WLA generate them instead. Can't say for sure without some checking how difficult this would be...

Ramsis-SNES, 2018-09-13:
> .define val49 49
> .define val121 121
> 
> .STRUCT ram_code_block
> DoDMA dsb val49
> UpdatePPURegs dsb val121
> .ENDST
> 
> This (using defined values) should already work

Yes, but even doing it this way, I still have to manually determine the actual values of `val49` and `val121`.

BTW I just checked, neither `.DEFINE valX _sizeof_XBegin` nor `.DEFINE valX XEnd-XBegin` seems to yield the correct results when using valX in an .ENUM or .STRUCT (error message: DIRECTIVE_ERROR: DS/DSB needs size.)

vhelin, 2021-07-21:
Getting back to this issue, DS/DSB really needs the size to be an immediate number (or a definition that holds the immediate number). Currently WLA cannot calculate sizes for code blocks in pass_1 where DS/DSB needs the size as pass_1 just outputs data into WLA's internal data stream and was it pass_3 the first place where these block sizes can be calculated as we parse through the data stream.

So either we need to make pass_1 to keep count of outputted labels and bytes (risky, data is output all over the place) or make pass_1 to be able to read that data stream it's outputting for later passes (perhaps better choice) and parse block sizes or _sizeof_s itself.

Making a simplified internal data stream parser (similar to pass_3) isn't a huge task, but it's a small part of the solution:

```
.STRUCT ram_code_block
    DoDMA                  dsb (SRC_DoDMA_END - SRC_DoDMA_START + 1)
    UpdatePPURegs     dsb (SRC_UpdatePPURegs_END - SRC_UpdatePPURegs_START + 1)
.ENDST
```

1. Find out in stack.c/_resolve_string() that SRC_DoDMA_END is not a definition, while we are trying to calculate the first calculation
2. We assume that it's a label that we've already parsed and outputted to WLA's internal data stream, so we...
3. ... parse through the internal data stream we have so far, and create a list of labels and their addresses
4. We try again to find SRC_DoDMA_END in this list, and we find it, and use its address
5. Next when we process SRC_DoDMA_START we see that it's also on this list and we can calculate (SRC_DoDMA_END - SRC_DoDMA_START + 1)
6. Calculation is done and we can return the result to DSB in this case
7. Free the list of labels we created in step 3

This has the side effect that we could try to use the addresses of the labels before WLALINK has placed the code sections (and the labels, giving the labels their real, final addresses), but perhaps this process could be done so that only subtractions A-B where both A and B are labels from the same section are allowed?

vhelin, 2021-07-21:
This issue is actually the same as https://github.com/vhelin/wla-dx/issues/15

Ramsis-SNES, 2021-07-23:
Thank you so much, I'm going to try this out ASAP! :-)

vhelin, 2021-07-23:
Just a reminder that A and B must be defined & parsed before you can issue that A-B.

vhelin, 2021-08-01:
As the commit https://github.com/vhelin/wla-dx/commit/3f5dd1cd020f484da308f55e9e711d9201d242a8 fixes this issue I'm pressing the close button. If you find the fix is buggy etc. please reopen this issue.

closed 2021-08-01 by vhelin