vhelin/wla-dx #186: Compiling for MSYS2 or MinGW-w64?

StarWolf3000, 2018-07-15:
I'm using MSYS2 as development environment (which is a rewrite of MSYS and MinGW, based on newer Cygwin), which provides the build system for 3 targets: MSYS2 itself (for MSYS2 own utilities and ports of *NIX utilities), MinGW-w64/i686 and MinGW-w64/x86_64.

While the MinGW-w64 targets share most of the packages, this is not the case with the MSYS2 target.

Which one of the 3 possible cmake commands is needed to generate the suitable makefiles, when I want to target MinGW-w64/i686?
```
cmake -G "Unix Makefiles" .  ; under Unix / OS X / Cygwin
cmake -G "MSYS Makefiles" .  ; under MSYS/MinGW
cmake -G "MinGW Makefiles" . ; under (plain) MinGW
```

vhelin, 2018-07-15:
I have to say that I have no idea. Have you tried out any of those three
cmake commands, will they produce working executables?

cr1901, 2018-07-15:
I use `MSYS Makefiles` for MinGW (or `Ninja` w/ `CC` and `CXX` set manually). This implies you're running `cmake` at the `bash` prompt.

Neui, 2018-07-15:
MSYS2 is a fork of Cygwin, so "Unix Makefiles" should work.

It seems the "MSYS Makefiles" version [seems to include more windows paths to search from](https://gitlab.kitware.com/cmake/cmake/blob/master/Source/cmGlobalMSYSMakefileGenerator.cxx) and [so seems to be designed for MSYS 1.0](https://gitlab.kitware.com/cmake/cmake/blob/master/Modules/CMakeMSYSFindMake.cmake).

StarWolf3000, 2018-07-15:
It seems to build just fine with both `Unix Makefiles` and `MSYS Makefiles`. It's currently running under GCC 7.x, so there are lots of warnings regarding `-wformat-overflow`, but the tools build just fine regardless of those warnings. The results are around 500 kb.
When running `run_unit_tests.sh`, `wla-65816` will halt at the `linker_test` at `wla-65816 -l test.lib test.s`, so I have to kill it from the processes to let the tests continue.

vhelin, 2018-12-16:
I just installed MSYS2, and compiled the latest WLA DX sources (cmake -G "Unix Makefiles"), ran run_unit_tests.sh, all worked just fine... linker_test/wla-65816 and test.s halts, but it asks for input. After I give it what it needs (a name for the program) it continues and finishes without errors...

Neui, 2018-12-17:
> Also is sphinx required or is it just for generating docs?

Just for generating docs. It'll auto-disable when it couldn't find sphinx at the initial configure time.
(Set `GEN_DOC` to `ON` to generate it, or `GEN_DOC` to `OFF` to not generate it)

> (I'm trying to find the corresponding MSYS2/MinGW packages for it)

Might be easier when you use `pip`. The package is usually called `python3-sphinx` and/or `python-sphinx`. In MSYS2, there is `mingw-w64-x86_64-python3-sphinx` (and other variations), however I didn't tested these.

StarWolf3000, 2018-12-17:
This is what happens at this point (the cursor is blinking, but no question is shown):
[image]

vhelin, 2018-12-17:
Here's what happens on my laptop (MSYS2, but Cygwin gives the same output):

[image]

StarWolf3000, 2018-12-17:
I've updated my comment, I'm not using the latest work-in-progress 9.8, but release 9.7, so no export from master branch.

vhelin, 2018-12-17:
I don't think that test has changed in many years, should also ask for the program name also in 9.7... Does your test.s look like this?

.DEFINE TESB 1
.EXPORT TESB

.SECTION "test"
test_00:        NOP
.ENDS

.PRINTT "what's the name of this program? "
.INPUT INFO

.SECTION "information"
.DB INFO
.ENDS

StarWolf3000, 2018-12-17:
Yes, with date from April 17 2018:
[image]

vhelin, 2018-12-17:
Here's what .PRINTT does in the end:

    if (quiet == NO)
      printf("%s", t);

Looking at your screenshot, wla-65816 is not running in quiet mode, so that printf() then is somehow directed somewhere else than your shell... ???

vhelin, 2018-12-17:
Bash buffered that printf() and outputted it after the input? Could also be that stdout is buffered... ? I'll add stdout flushing to 9.8a...

If you want to test it with your 9.7, then make sure PRINTT handling in pass_1.c ends to this:

    if (quiet == NO) {
      printf("%s", t);
      fflush(stdout);
    }

   return SUCCEEDED;

StarWolf3000, 2018-12-17:
I have added that line to pass_1.c (as line 6601), and after rebuilding all binaries and running the tests, I finally get:
```
wla-65816 -l test.lib test.s
what's the name of this program?
```
with the cursor blinking behind the question mark. So yeah, seems flushing is somehow required.

vhelin, 2018-12-17:
Great! I've never needed such flushing before, but it seems that you'll have to do it for the code to work 100% all the time. :)

closed 2018-12-17 by vhelin

cr1901, 2018-12-17:
>I've never needed such flushing before.

_Typically_, `stdout` is [line-buffered](https://en.cppreference.com/w/c/io/setvbuf). So including `\n` would've also done the trick. But tbh, I assume that's not what you wanted, and MSYS2's C stdlib buffering can interact badly with `bash` for some reason.