vhelin/wla-dx #187: Feature request: Add ExHiROM support

Ramsis-SNES, 2018-07-19:
Even though there were only two games released back in the day, the ExHiROM format is officially documented, and widely used in hacks and fan translations.

So, it would be very useful simply to put ``EXHIROM`` into a project's ``.SNESHEADER`` definition and thus be able to create ExHiROM games. :)

Ramsis-SNES, 2018-07-30:
PS: Actually, it's already possible ... with a few workarounds: https://github.com/Ramsis-SNES/furryrpg/commit/66ba4e6917c3dd1875e7720f9e52c5d967515605

Still, this would be a nifty new feature :D

vhelin, 2018-09-17:
All I need to do is set the byte $15 in the SNES header correctly, and in the case of ExHiROM do the rest as if the user had defined HiROM? Same with ExLoROM and LoROM?

Ramsis-SNES, 2018-09-17:
[Fullsnes](https://problemkaputt.de/fullsnes.htm#snescartridgeromheader) contains a lot of useful info on ExHiROM, which is kind of an "official" format as it was used by at least two commercial games (Tales of Phantasia, Daikaijyuu Monogatari 2, not counting special-chip-enhanced games).

For 40 Mbit and bigger games, it is important to put the internal ROM header and reset vectors into bank $40 (compare https://github.com/Ramsis-SNES/furryrpg/commit/66ba4e6917c3dd1875e7720f9e52c5d967515605). Commercial games mirror this information in bank $00, which is probably not necessary but obviously considered "good practice".

Another very important thing is ROM checksum calculation: With 40 Mbit games, the last 8 Mbit are read and added to the checksum four times, and with 48 Mbit games, the last 16 Mbit are read and added to the checksum twice (in both cases, 64 Mbit in total are read and added). When adding, skip checksum and checksum complement bytes in bank $40 and add $1FE as a dummy value, just like with normal LoROM/HiROM games.

Unfortunately, I have zero information/experience on ExLoROM, on the other hand ...

vhelin, 2018-09-18:
Does this mean that .SNESNATIVEVECTOR and .SNESEMUVECTOR datas are also mirrored from bank $40 to bank $00 in case of ExHiROM? :P

vhelin, 2018-09-20:
Ok, I've added support for .EXHIROM - I hope it works. Could you check it out?

The checksum should be computed as you specified, and after it's placed into $40ffdc - $40ffdf WLALINK will (in compute.c's finalize_snes_rom()) mirror the cartridge rom header from $40ffb0-$40ffff -> $ffb0-$ffff. I hope this will work.

Anyway, it'd be great if someone could test this feature. :)

vhelin, 2018-09-20:
Oops, forgot to update doc/asmdiv.rst so there is no documentation about .EXHIROM currently. I'll try to write the documentation tomorrow...

Ramsis-SNES, 2018-09-20:
Woooow, thank you so much Ville!! Please give me some time to try this and report back (shouldn't take much longer until tomorrow night though). :-)

Ramsis-SNES, 2018-09-20:
Just gave it a go, and it *almost* works (SNES startup code is missing in bank $40). Seems like we just have to put the whole upper 32K of HiROM bank $00 ($C0 for the SNES) to the upper 32K of ExHiROM bank $40 instead (i. e., HiROM mirroring rules for startup apply to ExHiROM too, except that you have to use the upper 32K of bank $40 instead of bank $00/$C0).

Thanks again @vhelin and please, please keep up the good work!! :D

vhelin, 2018-09-21:
> Just gave it a go, and it *almost* works (SNES startup code is missing in
> bank $40). Seems like we just have to put the whole upper 32K of HiROM bank
> $00 ($C0 for the SNES) to the upper 32K of [...]

No problem. :)

Does this mean that if .EXHIROM is defined I can pretend .HIROM is defined
(plus the extra ExHiROM bit in ROM header) until it's the time to calculate
checksum? Then I'll copy the upper 32KB of bank $00 to the upper part of
bank $40? And then I'll just calculate checksum ExHiROM style and put the
results into both bank $00 and $40?

Ramsis-SNES, 2018-09-21:
Sorry, my bad ... Forget the part about copying the whole upper half of bank $00, I was erroneously thanking about ExHiROM _expansion_ (the kind of thing tools like [Lunar Expand](https://fusoya.eludevisibility.org/le/index.html) can do).

Actually, after relocating my SNES startup code to bank $40, it works perfectly, i.e. the ROM gets correctly recognized by emulators! :D Only thing is that the checksum is bad.

Just had a quick look at your code, and I guess `int finalize_snes_rom(void)` needs a little correction:

`/* 2*255 is for the checksum and its complement bytes that we skipped earlier */`
`  res += 2*255;`

It's definitely not enough to add the dummy checksum and complement only twice. Depending on the ROM size, you'd have to add it at least three times (48 Mbit game --> bank $00 gets read once, and bank $40 gets read twice) or even five times (40 Mbit game --> bank $00 gets read once, and bank $40 four times).

Ramsis-SNES, 2018-09-21:
One more small issue: The lower nibble of the ROM makeup byte (byte 0x15 of the internal ROM header) should be 5, not 1 (as in HiROM games). ;) This is important as otherwise, emulators like higan and Geiger's debugger don't recognize the RTC support in my game.

> 

Bit7-6 Always 0   Bit5   Always 1 (maybe meant to be MSB of bit4, for "2" and "3" MHz)   Bit4   Speed (0=Slow, 1=Fast)              (Slow 200ns, Fast 120ns)   Bit3-0 Map Mode
--

vhelin, 2018-09-21:
> 
> 
> Sorry, my bad ... Forget the part about copying the whole upper half of bank $00, I was erroneously thinking about ExHiROM _expansion_ (the kind of thing tools like [Lunar Expand](https://fusoya.eludevisibility.org/le/index.html) can do).
> 
> Actually, after relocating my SNES startup code to bank $40, it works perfectly, i.e. the ROM gets correctly recognized by emulators! :D Only thing is that the checksum is bad.
> 
> Just had a quick look at your code, and I guess `int finalize_snes_rom(void)` needs a little correction:
> 
> `/* 2*255 is for the checksum and its complement bytes that we skipped earlier */`
> `res += 2*255;`
> 
> It's definitely not enough to add the dummy checksum and complement only twice. Depending on the ROM size, you'd have to add it at least three times (48 Mbit game --> bank $00 gets read once, and bank $40 gets read twice) or even five times (40 Mbit game --> bank $00 gets read once, and bank $40 four times).

I just committed a fix, and it's to wlalink/compute.c/compute_snes_exhirom_checksum() - is it working better? Can you check out the code?

vhelin, 2018-09-21:
> 
> 
> One more small issue: The lower nibble of the ROM makeup byte (byte 0x15 of the internal ROM header) should be 5, not 1 (as in HiROM games). ;) This is important as otherwise, emulators like higan and Geiger's debugger don't recognize the RTC support in my game.
> 
> (Cf. section "ROM Speed and Map Mode (FFD5h)" in [Fullsnes](https://problemkaputt.de/fullsnes.htm))

pass_2.c/write_snes_cartridge_information() of WLA should do this already, check out the first few lines of that function.

Ramsis-SNES, 2018-09-25:
> I just committed a fix, and it's to wlalink/compute.c/compute_snes_exhirom_checksum() - is it working better? Can you check out the code?

Yes, of course. Unfortunately, Geiger still gives me a bad checksum error.

> pass_2.c/write_snes_cartridge_information() of WLA should do this already, check out the first few lines of that function.

Whatever that function does (I'm hopeless at understanding C/C++ stuff), WLA still puts 31h instead of 35h to 40FFD5h, which is wrong. :D

vhelin, 2018-09-27:
> Yes, of course. Unfortunately, Geiger still gives me a bad checksum error.
>
>
> Whatever that function does (I'm hopeless at understanding C/C++ stuff), WLA still puts 31h instead of 35h to 40FFD5h, [...]

How about now, I just submitted a fix? I realized I need to mirror the ROM header first from bank $40 -> $00 and then calculate the checksum, otherwise the checksum might be different...

vhelin, 2018-09-28:
> 
> 
> Thanks! I just tried it with the current codebase, and finally, Geiger is happy with the checksum! :D
> 
> Still not sure what the issue in `write_snes_cartridge_information` is ... From what I seem to understand, this is what the function does if EXHIROM is defined: The value `inz` (initially 20h for LoROM) is ORed with (1 SHL 2) and additionally ORed with 1, so that should give us 25h. Lastly, 10h is added for FASTROM, so that should yield 35h, which is the value I need. Yet, the D5h in both headers (bank $00 and $40) is still 31h in my freshly assembled ROM. Why? Is `inz` overwritten somewhere else, perhaps?

Ouch, it was sloppy coding from me. I used || instead of | for that OR (4 or 1)... If you pull the latest sources it should work now?

Ramsis-SNES, 2018-09-28:
Yay, I just checked and now it works perfectly! Thank you so much once again, Ville! Keep up the great work! :D

closed 2018-09-28 by Ramsis-SNES

vhelin, 2018-09-28:
> 
> 
> Yay, I just checked and now it works perfectly! Thank you so much once again, Ville! Keep up the great work! :D

Haha, I could have been sharper with the coding, but at least eventually it now works. :)