vhelin/wla-dx #189: String length analysis

maxim-zhao, 2018-07-25:
I'd like both of these to work:
````
.db "Hello world".length ; emits 11
.macro pascalstring args s
.db s.length, s
.endm
  pascalstring "Hello world"
````
Else I'm having to count chars manually.

vhelin, 2018-09-28:
Hey, has anyone tested this feature? Works in my test cases, but the more people test these new features the better... :)

maxim-zhao, 2018-10-03:
I tried it, but my specific use case didn't work, presumably due to the order things get evaluated...
```
.macro CentrePaddedText args s
  .dsb (26-s.length)/2, ' '
  .db s, 0
.endm
  CentrePaddedText "CONVERTED IN THE UK"
```
The intention is to left-pad the string with spaces such that it appears centred in a 26-character wide rendering. The error is:
```
INPUT_ERROR: .DSB needs size.
ERROR: Couldn't parse ".dsb".
```
It assembles OK if I don't have arithmetic in the first arg to `.dsb`, but of course that's not the length I wanted.

vhelin, 2018-10-03:
> I tried it, but my specific use case didn't work, presumably due to the order things get evaluated...
>
> ```
> .macro CentrePaddedText args s
>   .dsb (26-s.length)/2, ' '
>   .db s, 0
> .endm [...]

Would

.macro CentrePaddedText args s
  .dsb (26-s)/2, ' '
  .db s, 0
.endm
  CentrePaddedText "CONVERTED IN THE UK".length

work? I bet in your example WLA first sees (26-s.length)/2 as a computation and launches the stack calculator that then internally thinks that s.length is a unresolved variable name and lets WLALINK to find it, and that makes .DSB to fail as it need an immediate value. I'll check it out perhaps later today!

maxim-zhao, 2018-10-03:
My goal is to avoid typing the string twice, so I need to emit the padding and then the string.

maxim-zhao, 2018-10-04:
Yes, [it works great for me](https://github.com/maxim-zhao/battlemaniacs-restoration/commit/bca96bf6c6ac695597f5eb551485f6876e0998b5) 👍

closed 2018-10-04 by maxim-zhao