vhelin/wla-dx #191: Add address-to-line mapping and source file list to Symbols file

CypherSignal, 2018-07-27:
In an effort to try and improve the debugging experience of homebrew ROM development, I have been working on some changes to wla-dx to add some additional information to the .sym file output by the linker: an listing of the address-to-line mapping, plus a list of source-files that were included during obj compilation. The majority of the changes that I have so far rely on the existing list cmds that are generated in the obj files, so it's mostly outputting data already collected. There are some additional things saved and calculated such as source-file checksum at time of obj compilation, or the base of a given section's bank.

For example, the following is added to the .sym file for a [simple ROM from Neviksti's starter kit](http://wiki.nesdev.com/w/index.php/Nesdev#SNES), after the `[labels]` and `[definitions]` sections : 
```
[source-files]
0001 a22631df show.asm 
0002 cd810adf show.inc 
0003 07116f72 LoadGraphics.asm 
0004 3117b7df InitSNES.asm 
0005 c18e5c0e 2input.asm 

[rom-file checksum]
a3afd3d7

[addr-to-line-mapping]
00:8383 0002:00000059
00:8384 0002:0000005c
00:8386 0002:0000005d
00:8387 0002:0000005e
00:8388 0002:00000060
00:838a 0002:00000061
00:838d 0002:00000063
00:838e 0002:00000064
00:838f 0002:00000065
00:8311 0003:00000049
00:8312 0003:0000004a
00:8313 0003:0000004b
00:8314 0003:0000004c
00:8315 0003:0000004d
00:8316 0003:0000004f
00:8318 0003:00000052
00:831b 0003:00000053
00:831e 0003:00000054
00:8321 0003:00000055
00:8324 0003:00000057
00:8326 0003:00000058
...
```

...et cetera. A full gist of the full .sym file can be seen here: https://gist.github.com/CypherSignal/fb23d610f2f835caf6bbc7ba426be9ef

When acknowledged by a corresponding emulator that loads up the addr2line map and referenced source files, this can be used to display the source assembly of a disassembled ROM, or perform other operations using the source file as a reference, such as add breakpoints by referring to a line in a source file, as opposed to solely relying on addresses. For example, with some changes to bsnes's disassembler view, the following listing can be provided, as opposed to just a list of translated opcodes

[image]

So far I have only been testing against wla-65816, but none of the work is platform-specific, aside from the "base bank" which I don't think is relevant for other platforms. There is at least one related issue found so far that this change highlights (listings for macros appear to be muddled, with wrong file and/or line numbers) and there may be some broader testing necessary to make sure all of the necessary data is available and useful before committing to it, but I wanted to file this enhancement for discussion before submitting a PR.

vhelin, 2018-07-28:
Looks really good! Keep up the good work. :)

vhelin, 2018-07-28:
> Does this break compatibility with any emulators, ie. bgb and no&gmb? Do
> they know to ignore the "addr-to-line-mapping" section and others?

If you have those emulators installed, just add a couple of such lines (no
need to be correct) to the sym file and test. :)

Drenn1, 2018-07-29:
Turns out I've been using "-S" instead of "-s" this whole time. It appears to work better, otherwise bgb doesn't show me label names in the code window - it treats them like constants, I suppose.

For some reason, adding the lines in the "source-files" section seems to break the sym file in bgb, it doesn't even recognize the values as constants. This isn't a huge deal for me since I'm sure I could use awk or something to fix it.

Do .sym files follow a pseudo-standard among emulators? If so, it might be worth asking bgb's author about fixing the problem.

CypherSignal, 2018-07-29:
> For some reason, adding the lines in the "source-files" section seems to break the sym file in bgb, it doesn't even recognize the values as constants.

When you add them to the wla-style symbols file? If so, that's a bit of a bother and disappointment...I would hope that most emulators that are loading this format in would be sensible enough to properly skip over sections they don't know how to handle. 

Either way, the new `[source-files]` and `[addr-to-line-mapping]` sections will be excluded entirely (i.e. even the section headers won't exist) if no _list_ information is generated from the obj compilation. Ideally, I'd still like to leave the `[rom-file checksum]` section in regardless, just because that's generally useful as a mechanism to verify if the symbols and rom match, but if there are many misbehaving emulators that come up, I'd be fine excluding the `[rom-file checksum]` section in the absence of list information as well. 

This does make me think that, besides making sure the output of the files makes sense for the targets that are not wla-65816, it'd be worth a quick check through some popular emulators that support wla-style symbols to make sure they don't break with this extra data, or other previously-undefined sections.

>Do .sym files follow a pseudo-standard among emulators? 

Not that I know of. I'm only looking at the wla-style format of...

- lines starting with `;` are skipped
- lines of `\[\S+\]`/`[%s]` such as `[labels]` represent a new section
- lines following the section header are the data for that section. If you're acknowledging the section, utilize that section's formatting (e.g. `[0-9a-f]{2}:[0-9a-f]{4} \S+`/`%2x:%4x %s` for labels). Read lines that match until a new section header line is encountered.

Drenn1, 2018-07-29:
The rom-file checksum, at least, didn't seem to cause any problems with bgb.

maxim-zhao, 2018-08-03:
Just to mention that Meka supports the symbol files too, but as far as I remember it will ignore the new sections.

CypherSignal, 2018-10-12:
As indicated above, I've filed PR https://github.com/vhelin/wla-dx/pull/202 for this issue. Given the things noted above, the new information is off-by-default, to prevent emulators improperly following the Symbol file format from messing up too much.

Sorry this took a long time to wrap up. There was one feature in my proof-of-concept that I had to get back to finishing up before I could say that, at least in the short term, there should not be any extra information that needs to be included in the symbols file as part of a richer debugging experience.

closed 2018-10-12 by vhelin

vhelin, 2018-10-12:
Good work, thank you for it! I had to make some tweaks to the code, and updated CHANGELOG and FILE_FORMATS. I hope everything still works. :)