vhelin/wla-dx #196: wlalink discard option is buggy and causes use-after-free memory errors

Gekkio, 2018-08-26:
I recently started using the `-d` option with wlalink and it caused a lot of random errors in my previously working ROMs. For example, sometimes it complains about duplicate sizeof labels: 

```
TRY_PUT_LABEL: Definition "_sizeof__print_results_halt_1" was defined more than once.
```

After debugging this for a while, I noticed that the `discard_drop_labels` function can cause an use-after-free, because it frees labels even if they are global labels.

Here's an example scenario:

1. `insert_label_into_maps` puts some label into both the global label map *and* the section label map / section namespace label map
2. The section is unreferenced and gets discarded later on
3. `discard_drop_labels` frees the label, which is still in the global label map
4. Something uses the label from the global label map. I don't know what it would be, but it's clear that it happens

Somehow `discard_drop_labels` should determine if freeing the label is ok or not. I couldn't figure out how to do it properly, so I'm using a local workaround that just leaks memory instead:

https://github.com/Gekkio/wla-dx/commit/30f1d7b26ac7464830ef21416b2422ec9dae012e

vhelin, 2019-05-11:
Gekkio, could you test the latest sources with your tools? I'm running CygWin and can't get Valgrind to it... The problem should now be fixed.

Anyway, I made it so that unreferenced labels are now marked as dead, and thus not processed any longer instead of free()'ing the unreferenced labels right away and getting all the trouble with hashmaps...

Gekkio, 2019-05-11:
Running `wlalink` with valgrind didn't show errors any more, so this seems to be ok now! Thanks! :+1:

closed 2019-05-11 by Gekkio