vhelin/wla-dx #205: Feature request: Add a custom prefix to included labels

lajohnston, 2018-10-12:
Hi there,

I think a nice feature would be to allow optional prefixes to labels in a file, providing some namespacing to prevent labels clashing while aiding code organisation. For example, if file foo.asm defines 'bar' and 'baz', we can specify the prefix as 'foo.' and thereafter reference these labels as foo.bar and foo.baz.

```
.include "./foo.asm" prefix "foo."
foo.bar
foo.baz
````

This would be particulary useful when including third party libs, where it's not desirable to edit the labels yourself.

vhelin, 2020-05-15:
There is a small problem with this: let's say I have an instruction in include.s:

   CALL JumpPoint

Now, if JumpPoint was in main.s that includes include.s and prefixes all labels inside include.s with "foo.", then this call would become

   CALL foo.JumpPoint

and assembling would fail... How to solve this issue?

vhelin, 2020-05-15:
Or can we assume that every time we use label prefixes, the included file is completely self contained?

lajohnston, 2020-05-15:
My thinking is the prefix would only apply to labels defined within the file, so `JumpPoint` wouldn't be prefixed as it's a reference and not defined anywhere within `include.s`

If a `JumpPoint` label was defined somewhere in `include.s` though, then that would be prefixed to `foo.JumpPoint` and the reference would also be updated to `CALL foo.JumpPoint`

vhelin, 2020-05-15:
Ok, then this needs to be done differently from what I was initially thinking. Not a blocker, though, I guess...

Kroc, 2020-05-16:
This is "namespaces" by another name, typical syntax in other languages is:

```
; within the file, to define its namespace externally
namespace "foo"

; changing the prefix using an include:
include "foo.wla" as "bar"
```

Other assemblers use "scopes" or "zones" as namespaces.

vhelin, 2020-05-16:
Ok, now you can do:

.include "./foo.asm" namespace "foo"

is this ok?

lajohnston, 2020-05-18:
yes that looks great, thanks! I haven't had a chance to try it out yet. Does this apply to macro names as well?

vhelin, 2020-05-18:
It doesn't apply to macro names, just labels and label references... Making it affect .MACROs and macro calls might feel a bit weird, I think, but I can see why the namespace should affect them - to keep them separated from the rest of the code.

I have to admit that I don't really know. Should it affect .MACROs and macro calls as well?

lajohnston, 2020-05-18:
If it's possible I think it probably should apply to MACROs. If including a file with a namespace it would ideally prefix everything to ensure nothing gets polluted to the global namespace. I suppose that also leads to the question of whether .DEFINEs should be prefixed to prevent pollution, though I think .MACROs certainly.

Thanks for looking into this! :)

Kroc, 2020-05-23:
How does this affect section names? Using a namespace should prevent section names colliding, so that "modules" of code can use simpler, more generic section names without fear of name-collision, but at the same time this breaks the ability for a "module" to append to a section that exists externally;

For example, lets say you're coding a Pokemon game and you use a module for each Pokemon: `.INCLUDE "clefairy.wla" NAMESPACE "clefairy"`, affording you psuedo-polymorphism (e.g. `clefairy.hp`)

The problem now is that the `clefairy` module cannot append itself to numerous tables located in the main code, avoiding the need to maintain multiple separate tables that *must* match the same order that the Pokemon are INCLUDEed! -- I am always trying to drive all my disassembly projects into using include-order to build several internal tables.

Ergo, there needs to be a way for a namespace'd file to refer to sections outside of its namespace (if present).

I'm going to recommend some kind of "global prefix" for names, such as `*`, e.g. `*:pokemon_names` or whatever suits your fancy.

lajohnston, 2020-05-25:
SECTIONS seem to have an existing namespace feature but I haven't used it before (https://wla-dx.readthedocs.io/en/latest/asmdiv.html?highlight=section#section-init-force). If the file namespace prefixes the section names it might be sensible for SECTION namespaces to override it, so you could then specify the section's namespace as "" to opt it out of the file namespacing.

Kroc, 2020-05-29:
There appears to be a bug:

```
; main.wla:
.INC "test.wla" NAMESPACE "test"

; test.wla:
.SECTION "test"
do:
@loop: jr @loop
.ENDS
```

Produces: `FIX_REFERENCES: Reference to an unknown label "do@loop"`

Kroc, 2020-05-30:
Thank you, that was holding me back!

vhelin, 2020-05-30:
> How does this affect section names? Using a namespace should prevent section names colliding, so that "modules" of code can use simpler, more generic section names without fear of name-collision, but at the same time this breaks the ability for a "module" to append to a section that exists externally;
> 
> For example, lets say you're coding a Pokemon game and you use a module for each Pokemon: `.INCLUDE "clefairy.wla" NAMESPACE "clefairy"`, affording you psuedo-polymorphism (e.g. `.BYTE clefairy.hp ; the pokemon's health points`)
> 
> The problem now is that the `clefairy` module cannot append itself to numerous tables located in the main code, avoiding the need to maintain multiple separate tables that _must_ match the same order that the Pokemon are INCLUDEed! -- I am always trying to drive all my disassembly projects into using include-order to build several internal tables.
> 
> Ergo, there needs to be a way for a namespace'd file to refer to sections outside of its namespace (if present).
> 
> I'm going to recommend some kind of "global prefix" for names, such as `*`, e.g. `*:pokemon_names` or whatever suits your fancy.

Good point! I think I might be able to get this to work. Let's see tomorrow...

Kroc, 2020-05-31:
One thing I forgot to think about is how `APPENDSTO` should work; for example, let's say that a namespaced file wants to append to a section that is outside its own namespace; but there happens to exist a same-named section within the namespace:

```
; NOTE: this file is namespaced "bob"

.SECTION "table"
    ; this is section "bob.table"
.ENDS

.SECTION "table_data1" APPENDSTO "table"
    ; this section is "bob.table_data1" and appends to "bob.table"
.ENDS

.SECTION "table_data2" APPENDSTO "*:table"
    ; this section is "bob.table_data2" but appends to an un-namespaced section "table"
    ; the exact syntax-sugar for the global identifier is the only new syntax needed to be defined
.ENDS

.SECTION "table_data3" NAMESPACE "" APPENDSTO "table"
    ; IMO this is an un-namespaced section named "table_data3"
    ; but is appending to "bob.table" as that is the current namespace for this file
.ENDS

.SECTION "table" NAMESPACE "alice"
    ; this section is "alice.table", it is not considered part of the "bob" namespace
.ENDS

.SECTION "table" NAMESPACE ""
    ; this is a global section "table", without a namespace, that happens to be defined
    ; within a namespaced file -- useful for modules that wish to add global code routines
.ENDS
```

vhelin, 2020-05-31:
I messed up the Git commit message, so no reference in this thread, but in the latest sources this logic should now be implemented.

vhelin, 2021-03-22:
Any news about this issue? Is it now fixed, can we close this issue, or does it need more work?

lajohnston, 2021-03-23:
> Any news about this issue? Is it now fixed, can we close this issue, or does it need more work?

This feature seems to be working fine, thanks! If any issues come up with it I suppose we can always post here or create a new issue

closed 2021-03-23 by lajohnston

vhelin, 2022-07-15:
Can you create a small testcase that shows this behavior? I have this in main.s

```
.include "namespaced.i" namespace "namespaced"
```

... and then in namespaced.i I have this:

```
      .macro callMeSub
      .db $22
      .endm

      .macro callMe
      callMeSub
      .db $11
      .endm

      ...

      callMe                                 ; @BT 22 11
```

... and this works just fine.

vhelin, 2022-07-15:
Ok, if I call callMe from main.s then I get 

```
namespaced.i:12: ERROR: Unknown symbol "callMeSub".
```

this seems to be the issue you are having?

If I go and change callMeSub call in namespaced.i to namespaced.callMeSub then it works, but it's not how things should work. :) I'll see if I can fix this somehow...

lajohnston, 2022-07-15:
ok, thanks! Yes that looks like the same issue I'm having

vhelin, 2022-07-15:
Does the latest commit fix the issues you are having, or is there something left?

lajohnston, 2022-07-15:
Great, that was quick! Yes that fixes the issue, thank you. I'm not aware of any other issues :)

lajohnston, 2023-05-04:
Sorry @vhelin but I think there might be another issue very similar to this, this time when a macro refers to a label defined within the same namespaced file. I'm using v10.4 and the following fails:

fileA.asm
```
.section "someSection" free
    someLabel:
        ret
.ends

.macro "someMacro"
    call someLabel
.endm
```

fileB.asm
````
.include "fileA.asm" namespace foo

.section "main" free
    call foo.someLabel   ; this works
    foo.someMacro        ; FIX_REFERENCES: Reference to an unknown label "someLabel".
.ends
````

I can create a separate issue for this if needed. I can work around it for now though so there's no rush :)

vhelin, 2023-05-05:
Thanks for the example code, with it I can produce the error! I'll try to fix this issue this weekend...

lajohnston, 2023-05-06:
great, thanks! I'm starting to use this feature more in projects now and it really helps to clean up the code :)

A couple of other things I think might also benefit from namespacing:

1. `fsize` labels don't currently get namespaced, for example:

fileA:
```
    palette:
        .incbin "palette.bin" fsize paletteSize
```

when including fileA with namespace `foo`, `foo.palette` exists but `paletteSize` remains unchanged

2. What do you think about prefixing `.define`s in namespaced files? That would help ensure constants in one file don't clash with constants in another

Kroc, 2023-05-06:
Unprefixed constants are kind of important, but a local version would be useful. For example `.local CONST_NAME 3`

vhelin, 2023-05-06:
1. Some strings are parsed using functions that don't add namespace to the string...
2. ... .DEFINE is one of those...

I think it would make sense to add namespace to all strings in a namespaced file as that's why you want to achieve when putting them inside a namespace... ?

vhelin, 2023-05-06:
> Unprefixed constants are kind of important, but a local version would be useful. For example `.local CONST_NAME 3`

Hmm what's the harm in adding the namespace to a .DEFINE that's only used inside that namespaced file?

lajohnston, 2023-05-06:
> I think it would make sense to add namespace to all strings in a namespaced file as that's what you want to achieve when putting them inside a namespace... ?

yes I think basically everything that would normally get defined and accessible on the global namespace should instead be placed within the file namespace, but there might be some use-cases I've not run up against where this isn't desired (like with section names and APPENDTO etc.). If they're the exception rather than the rule they could perhaps have specific opt-outs

> Hmm what's the harm in adding the namespace to a .DEFINE that's only used inside that namespaced file?

There might be a use-case for having something like a `.DEFINEGLOBAL` keyword to opt-out of the file namespacing. I think I'd prefer to define these global constants in a non-namespaced file though and have .DEFINEd constants namespaced as the default for namespaced files

vhelin, 2023-05-06:
> Sorry @vhelin but I think there might be another issue very similar to this, this time when a macro refers to a label defined within the same namespaced file. I'm using v10.4 and the following [...]

Just pushed a commit that should fix at least this issue

vhelin, 2023-05-06:
> yes I think basically everything that would normally get defined and accessible on the global namespace should instead be placed within the file namespace, but there might be some use-cases I've not [...]

Currently, if I understood one of my tests correctly, .SECTION names also get namespace added to them. I think I'll try to make everything namespaced inside a namespaced file and let's then add opt-outs...

lajohnston, 2023-05-06:
> Just pushed a commit that should fix at least this issue

Ok great I believe that fixes the macro issue, but I think something else might be broken either with this commit or with something else between 10.4 to this latest commit.

```
.macro "myMacro" args myArg
    .print "myArg equals ", myArg, "\n"

    .if myArg < 0 || myArg > 15
        .fail "invalid arg"
    .endif

    ld a, myArg + 1
.endm
```

The above is imported with namespace `foo`. It prints with the correct value being passed in but the if statement now fails with `COMPUTE_STACK: Comparison between a string and a number doesn't work` and `ERROR: Couldn't parse ".if".` 

Commenting that if statement out, the `ld a, myArg + 1` now errors with `PARSE_STACK: Unresolved reference to "foo.myArg"`. This also worked fine in 10.4

Kroc, 2023-05-06:
> Currently, if I understood one of my tests correctly, .SECTION names also get namespace added to them. I think I'll try to make everything namespaced inside a namespaced file and let's then add opt-outs...

If defines are prefixed from one version to the next this is going to break quite some code; any namespaced module may want to define a global constant -- I have a menu module that defines `MENU_WIDTH` for example; that doesn't belong in a separate file, it belongs with the menu definitions.

Defines should be global by default as that's current behaviour, and a new thing should create local defines -- if you do it the other way then you both break existing code and require users to change their code to achieve the same thing without gaining any benefit from the new feature without *further* work. If you keep defines global, then no work needs to be done and users can incrementally improve their code.

Making defines local by default is very much a case of "throwing the baby out with the bath water"!

vhelin, 2023-05-06:
Hmm didn't think of that. Yeah, keeping existing code working is a top priority. Maybe I'll add a switch that can be used to add namespace to everything inside a namespaced file...

vhelin, 2023-05-06:
Ok, I added -a flag to the assembler that namespaces everything in a namespaced file. Also this should now work

```
.macro "myMacro" args myArg
    .print "myArg equals ", myArg, "\n"

    .if myArg < 0 || myArg > 15
        .fail "invalid arg"
    .endif

    ld a, myArg + 1
.endm
```

lajohnston, 2023-05-06:
> Also this should now work

Yes, that's working now. Thanks for the quick turn around on this! :)

The `-a` flag seems to also be working for most things like fsize. There is one problem though: when using `.DEFINE` within the file it's no longer possible to refer to it by its local unnamespaced name within the same file. You have to add the namespace prefix to it when referring to it, but the file itself won't know what the namespace is so isn't able to.

Also, the trouble with having this as a command line option is that it splits the 'language' into two, so a library that requires `-a` won't now be able to be assembled alongside code that doesn't. I wonder if something like you did with the macro `isolate` feature might work better as an opt-in, so:

`.include "fileA.asm" namespace "foo" isolated    ; opt-in to fully namespaced file`

That way you can opt-in on a per-file basis and everything should still work together. Or maybe add a .ISOLATED flag that can be added to the top of file so the file itself opts in to it. You could also introduce more granular options in future if needed like you did with `isolatelocal` etc, such as `isolatedefines`

vhelin, 2023-05-06:
> The -a flag seems to also be working for most things like fsize. There is one problem though: when using .DEFINE within the file it's no longer possible to refer to it by its local unnamespaced name [...]

Oof, I'll try to fix it asap, but it's getting late here...

That ISOLATED thing sounds good, perhaps I'll replace -a with it...

vhelin, 2023-05-06:
> The -a flag seems to also be working for most things like fsize. There is one problem though: when using .DEFINE within the file it's no longer possible to refer to it by its local unnamespaced name [...]

I think I got this to work now properly.

I'll look into implementing that ISOLATED tomorrow...

Kroc, 2023-05-07:
This is adding a lot of complexity when the issue at hand could be solved with a method of creating local defines, either by an alternative keyword or some addition to the existing define syntax. If you add a whole alternate assembler mode and then combine that with in-code overrides you get a combinatorial explosion of potential bugs

lajohnston, 2023-05-07:
I think the `-a` assembler mode will be scrapped in favour of an opt-in directive.

If the main breaking change is namespacing the .defines then yes I think an alternative opt-in for that would be the cleanest solution like the `.local` you suggested.

There's also stuff like fsize that would need its own opt-in, and I'm not sure what else. If there are lots of little things like that it might be cleaner to have one `isolate` directive for all of them, otherwise specific ones might be the way to go

Kroc, 2023-05-07:
Perhaps a `LOCAL` keyword (it could be `ISOLATED` but I don't personally like that word) can be appended to `FSIZE` to achieve that, i.e. `FSIZE LOCAL label` or `FSIZE label LOCAL`.

Defines could likewise be `.LOCAL name data` or `.DEFINE name data LOCAL`

vhelin, 2023-05-07:
Adding .LOCAL for creating local definitions would require changes in the file formats and the linker as well so it wouldn't be simple either. And looking at the amount of code ISOLATED (in .INCLUDE) took to implement, it's not that much really. I think it's better to have just one keyword for .INCLUDE more than adding LOCALs to multiple places...

vhelin, 2023-05-07:
Anyway, I could add .LOCAL later if ISOLATED in .INCLUDE doesn't work well enough... Also just pushed a commit that removes -a and adds ISOLATED to .INCLUDE.

lajohnston, 2023-05-07:
ok thanks! I'm happy with either approach but if the `ISOLATED` directive is the less complex to develop then perhaps that's the way to go.

Either way, I've tested with the latest commit and with `ISOLATED` for the `.INCLUDE` everything so far works as I expect and the additional issues I raised seem to all be fixed :)

1. fsize is now namespaced, but the label can be referred to by its local name within the file
2. .defines are now namespaced, but the value can be referred to by its local name within the file
3. Macros can call labels within the file
4. Macro arguments are working

I'll continue to actively use this feature and try out different things. Hopefully we can iron things out before the 10.5 release :)

Kroc, 2023-05-07:
I think this is a great solution for now using what tools already exist. I need to write down my thoughts somewhere for an overhaul of the entire scope system. A clean system of label hierarchy is needed like modern languages -- it's just not possible to tell what is and what isn't global/local, a constant, variable, label name, macro and such from the call site alone and I believe a future version of WLA-DX could make that a lot cleaner and clearer

lajohnston, 2023-05-07:
ok, strange bug. It seems when a namespaced section refers to a namespaced define from another file, the define reference gets double namespaced so isn't accessible:

foo.asm
```
.define VALUE 123
```

bar.asm
```
.section "barSection" free
    barLabel:
        ld a, foo.VALUE ; FIX_REFERENCES: Reference to an unknown label "bar.foo.VALUE"
        ret
.ends
```

main.asm
```
.include "foo.asm" namespace "foo" isolated
.include "bar.asm" namespace "bar" isolated

.section "main" free
    call bar.barLabel
.ends
```

It does seem to work in macros though, as I can `.print foo.VALUE` in a macro in bar.asm and it prints the correct value

vhelin, 2023-05-07:
With ISOLATED the namespace is added to everything inside the namespaced file. I could make it so that if a label already contains a dot then no namespace is added to it.

Also need to check out .PRINT, perhaps it's using a wrong function to get the printed value, a function that doesn't add the namespace...

vhelin, 2023-05-07:
@lajohnston If you try out the latest sources, do they fix this issue? At least my improved ISOLATED test worked...

lajohnston, 2023-05-07:
thanks, yes latest commit fixes that one :) Unfortunately now it's got past that issue it's snagged on something else, this time un-named labels. The following doesn't work anymore:

```
-:
djnz -    ; FIX_REFERENCES: Reference to an unknown label "foo.-".
```

> I  could make it so that if a label already contains a dot then no namespace is added to it.

I think basically everything that gets defined within the namespaced file and is globally accessible should be namespaced, but external references should be left unchanged. It also looks like un-named labels should be excluded as well, as they're not accessible outside the file. On a side note I've just tested child labels and these seem to work fine.

vhelin, 2023-05-07:
Good testing there! I think I need to make another exception to namespace prefixes: don't add them to unnamed labels... Another fix coming soon... Yeah, child labels have already an exception there, those I remembered. :)

About this: "... but external references should be left unchanged." - the assembler doesn't really know what's an external reference when it assembles the source code so we cannot add such an exception. Unless we mark external/global labels with a prefix that the namespace adder code can then detect... Currently '.' would work as such a prefix.

vhelin, 2023-05-07:
Ok, the fix is in, and I also remembered to make it so that namespacing also ignores local labels

lajohnston, 2023-05-07:
Thanks, that's all working now! :)

lajohnston, 2023-05-19:
Hi @vhelin, sorry it looks like structs don't currently get namespaced with `isolated`. Is that something you think could be added?

fileA.asm
```
.struct "Instance"
   ...
.endst
```

fileB.asm
```
.include "fileA.asm" namespace "foo" isolated

.ramsection "ram" slot 2
   foo: instanceof foo.Instance ; No STRUCT named "foo.Instance" available.
.ends
```

vhelin, 2023-05-19:
Absolutely! I'll try to give it a look tomorrow or on Sunday.

Kroc, 2023-05-20:
This appears to have broken something for me. I get this error;

```
Pass 1...
src/piles.wla:342: DIRECTIVE_ERROR: Reference 
to an unidentified structure "piles.PileROM". 
src/piles.wla:342: ERROR: Couldn't parse ".DSTRUCT".
```

for the given code:

```asm
.DEF    __pool_addr     storage

; for the purposes of providing pre-defined piles in ROM,
; this struct is used to store (most) fields linearly
;
.STRUCT PileROM
        type            BYTE    ; type of pile, i.e. deck, hand, foundation &c.
        pool            WORD    ; address of storage for card pile
        capacity        BYTE    ; max.number of cards allowed in pile, 1-based!
        row             BYTE    ; row position on the tilemap (not pixels!)
        col             BYTE    ; column position on the tilemap (not pixels!)
        left            BYTE    ; index of pile to the left of this one
        right           BYTE    ; index of pile to the right of this one
        up              BYTE    ; index of pile above this one
        down            BYTE    ; index of pile below this one
.ENDST

; use this macro before defining a set of piles in ROM
; in order to reset the starting pool address
;
.MACRO  Begin
;>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
.REDEF  __pool_addr     storage
;<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
.ENDM

; this macro simplifies writing a pre-defined Pile of cards into ROM
;
.MACRO  Define
;>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

.DSTRUCT \1 INSTANCEOF PileROM VALUES
        type:           .BYTE \2
        pool:           .WORD __pool_addr
        capacity:       .BYTE \3
        row:            .BYTE \4        ; screen row (Y) of pile
        col:            .BYTE \5        ; screen column (X) of pile
        left:           .BYTE \6        ; index of pile to the left
        right:          .BYTE \7        ; index of pile to the right
        up:             .BYTE \8        ; index of the pile above
        down:           .BYTE \9        ; index of the pile below
.ENDST

; move to the next available address in the pool;
; we include an extra byte for a zero terminator
.REDEF  __pool_addr     __pool_addr+\3+1

;<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
.ENDM
```

This file, "piles.wla", is namespaced by the file including it, therefore the namespace's name shouldn't be used within the file because it would change. This raises the question of how the macro is supposed to work when referring to PileROM when invoked from a file with a different namespace.

Regardless, for the purposes of testing, changing the `.DSTRUCT` to ```.DSTRUCT \1 INSTANCEOF piles.PileROM VALUES``` still gives the exact same error message!

vhelin, 2023-05-20:
Ouch, I'll start debugging right away!

vhelin, 2023-05-20:
Oh wow, never expected anyone to .DSTRUCT inside a .MACRO. :) Oh well...

Kroc, 2023-05-20:
Sorry, but it's kind of important because the order of the properties can, and have changed and whilst not in this particular example, I do have other macros where the input arguments have to be combined and modified and I don't want that logic to be duplicated hundreds of time throughout the codebase

vhelin, 2023-05-20:
I don't think I understand what you are saying, but I'll do my best to get this (your .DSTRUCT inside a .MACRO) to work (again)...

[69 more replies]