vhelin/wla-dx #207: wla-65816 discard input for assembling the file

hearues-zueke-github, 2018-10-28:
I have tried the following in the console (using Linux) and got the following output:
```
u@pc:my_first_snes_rom$ wla-65816 -vo first_snes_rom.asm first_snes_rom.obj

WLA 65816 Macro Assembler v9.8a
Written by Ville Helin in 1998-2008 - In GitHub since 2014: https://github.com/vhelin/wla-dx
$VER: WLA-65816 9.8a (19.10.2018)

USAGE: wla-65816 [OPTIONS] <ASM FILE>

Options:
-i  Add list file information
-M  Output makefile rules
-q  Quiet
-t  Test compile
-v  Verbose messages
-x  Extra compile time definitions
-I [DIR]  Include directory
-D [DEF]  Declare definition

Output types:
-o [FILE]  Output object file
-l [FILE]  Output library file
```

I have also tried out the version `v9.7b` which is also giving me the same message. But the version
`v9.5` is working for some reason.

Also I have compiled the version `v9.8a` from a fresh repo too. Am I using something wrong here?

vhelin, 2018-10-28:
The options are now separate:

  wla-65816 -v -o first_snes_rom.obj first_snes_rom.asm

... should work these days.

closed 2018-10-30 by vhelin

cr1901, 2018-10-30:
I could see this being an issue a number of old users of WLA have when using new versions. I would propose a FAQ, but nobody ever reads a FAQ :D.

vhelin, 2018-11-01:
It might be better then to make the executables print an example of how to
use the flags when they spot an error in the args?

hoit, 2019-04-26:
I don't know if will get a response because this issue is closed...
i try to upgrade wla dx (9.4 i think) version used by PVSneslib (C snes lib) to the last one 9.8b. It use also tcc816 and is no more compatible with the latest version : https://github.com/optixx/snes-sdk

When i compile a hello world, i get this error :

wla-65816 -v -o hello_world.obj hello_world.asm
Pass 1...
hello_world.asm:53: DIRECTIVE_ERROR: ROM banks == 8, selected bank 127.
hello_world.asm:53: ERROR: Couldn't parse "ram.data".
"make": *** [hello_world.obj] Error 1

asm file generated is like that, it seems to be the same that we get it in old version of wla dx :

> .include "hdr.asm"
> .accu 16
> .index 16
> .16bit
> .define __main_locals 0
> .section ".text_0x0" superfree
> main:
> .ifgr __main_locals 0
> tsa
> sec
> sbc #__main_locals
> tas
> .endif
> jsr.l consoleInit
> pea.w (0 * 256 + 1)
> sep #$20
> rep #$20
> jsr.l setMode
> pla
> sep #$20
> lda #1
> pha
> rep #$20
> jsr.l bgSetDisable
> tsa
> clc
> adc #1
> tas
> sep #$20
> lda #2
> pha
> rep #$20
> jsr.l bgSetDisable
> tsa
> clc
> adc #1
> tas
> jsr.l setScreenOn
> __local_0:
> jsr.l WaitForVBlank
> bra __local_0
> lda.w #0
> sta.b tcc__r0
> __local_1:
> .ifgr __main_locals 0
> tsa
> clc
> adc #__main_locals
> tas
> .endif
> rtl
> .ends
> .ramsection "ram.data" bank $7f slot 3
> __local_dummy.data dsb 1
> .ends
> .section ".data" superfree
> __local_dummy.data: .db 0
> .ends
> .section ".rodata" superfree
> __local_dummy.rodata: .db 0
> .ends
> 
> 
> 
> .ramsection ".bss" bank $7e slot 2
> __local_dummybss dsb 1
> .ends
> 

Do you have any idea how to solve it ?

vhelin, 2019-04-28:
This might not be the right thread for this question, but it seems that you are putting a .ramsection on line 53 to bank $7f (127):

.ramsection "ram.data" bank $7f slot 3

... here's the error again:

hello_world.asm:53: DIRECTIVE_ERROR: ROM banks == 8, selected bank 127.

I don't see anywhere .ROMBANKMAP that defines the number of ROM banks, so I suspect it's inside hdr.asm, included on the first line. Can you check there?

The funny thing in this error is that the assembler assumes that there are as many RAM banks as there are ROM banks, which is clearly a "feature" if not a bug. The linker seems to assume on the other hand that there are 256 RAM banks for 256 slots...

If you cannot change the number of ROM banks, or the bank the RAMSECTION uses, I'll try to code a fix for this asap...

vhelin, 2019-04-28:
If you try now the latests WLA-DX sources from GitHub, does it work? I made it so that the assembler also thinks there are 256 RAM banks available per SLOT, and doesn't use the number of ROM banks to determine the number of RAM banks (uhh...).

hoit, 2019-04-28:
Thanks a lot for this quick fixing :)

You are right, ROMBANKS are defined in header :

```
.ROMBANKSIZE $8000              ; Every ROM bank is 32 KBytes in size
.ROMBANKS 8                     ; 2 Mbits - Tell WLA we want to use 8 ROM Banks
```

I confirm that this step works correctly now but i still have difficulties with new parameters for wlalink.

The previous one was :

> wlalink -dsnov  hello_world.obj data.obj hdr.obj hello_world.obj /e//pvsneslib/lib/crt0_snes.obj /e//pvsneslib/lib/libc.obj /e//pvsneslib/lib/libm.obj /e//pvsneslib/lib/libtcc.obj /e/snes-examples/hello_world/hello_world.sfc

and i try differents things without success :

wlalink -d -v -S  hello_world.obj data.obj hdr.obj hello_world.obj -L /e//pvsneslib/lib/crt0_snes.obj /e//pvsneslib/lib/libc.obj /e//pvsneslib/lib/libm.obj /e//pvsneslib/lib/libtcc.obj /e/snes-examples/hello_world/hello_world.sfc

I tried with -L parameter for files in /e/pvsneslib/lib folder but still the same.
I an checking on differents forum if i find something.. :)

Sorry to abuse but i have an other question : PVSneslib tool provide library build with its own header. When in a new project you try to change important parameters like SRAMSIZE, rom speed to FASTROM) in header of your project, you get this error  : CHECK_HEADERS: The object files are from different projects.

I read in wla documentation that we can use .BACKGROUND directive to "patch" the rom :

"Currently WLA can also be used as a patch tool. Just include the original
ROM image into the project with .BACKGROUND and insert e.g., OVERWRITE .SECTIONs
to patch the desired areas. Output the data into a new ROM image and there
you have it. 100% readable (asm coded) patches are reality!"

My new header should become :

```
.BACKGROUND "hello_world.sfc"

OVERWRITE .SNESHEADER
  ID "SNES"                     ; 1-4 letter string, just leave it as "SNES"

  NAME "TEST OVERWRITE PARAM "  ; Program Title - can't be over 21 bytes,
  ;    "123456789012345678901"  ; use spaces for unused bytes of the name.

  SLOWROM
  LOROM

  CARTRIDGETYPE $00             ; $00 = ROM only $02 = ROM+SRAM, see WLA documentation for others
  ROMSIZE $08                   ; $08 = 2 Mbits,  see WLA doc for more..
  SRAMSIZE $00                  ; $00 = No Sram, $01 = 16 kbits, see WLA doc for more..
  COUNTRY $01                   ; $01 = U.S.  $00 = Japan, that's all I know
  LICENSEECODE $00              ; Just use $00
  VERSION $00                   ; $00 = 1.00, $01 = 1.01, etc.
.ENDSNES
```

I build the .sfc file, then i try to do a news one with wla (sorry, it is done with the old version !) :
- the new header
E:\devkitsnes\bin\wla-65816 -io newhdr.asm newhdr.obj

- the new .sfc file but using newhdr.obj :
E:\devkitsnes\bin\wlalink -dsnov hello_world.obj data.obj newhdr.obj E:\pvsneslib\lib\crt0_snes.obj E:\pvsneslib\lib\libc.obj E:\pvsneslib\lib\libm.obj E:\pvsneslib\lib\libtcc.obj E:\snes-examples\hello_world/hello_world1.sfc

but it gives me : 

E:\pvsneslib\lib\crt0_snes.obj:crt0_snes.asm: INSERT_SECTIONS: No room for section "Snes_Init" (143 bytes) in ROM bank 0.banks[s->bank] 32768, s->address 0x2

I opened a thread here without success : https://forums.nesdev.com/viewtopic.php?f=12&t=18589

Is it possible to do it ? 

For clarity reason, i propose to open a new "issue"..

vhelin, 2019-04-28:
> Thanks a lot for this quick fixing :)
>
> You are right, ROMBANKS are defined in header :
>
> ```
> .ROMBANKSIZE $8000              ; Every ROM bank is 32 KBytes in size
> .ROMBANKS 8                     ; 2 [...]

Ok, good, one problem solved! That's a start! :)

> The previous one was :
>
>
> and i try differents things without success :
>
> wlalink -d -v -S hello_world.obj data.obj hdr.obj hello_world.obj -L /e//pvsneslib/lib/crt0_snes.obj [...]

Oh no, a lot has changed since the ancient times (of v9.4 etc)... These days you'll need to create a linkfile, which is an ASCII file that has all the object files listed that you want to link. Check some of the projects in the examples directory for reference.

Today you'll need to use WLALINK as follows:

   wlalink -d -v -S linkfile /e/snes-examples/hello_world/hello_world.sfc

... and inside this file called "linkfile" you'll have something as follows:

...
[objects]
hello_world.obj
data.obj
hdr.obj
hello_world.obj
[libraries]
bank 0 slot 0 /e//pvsneslib/lib/crt0_snes.obj
bank 0 slot 0 /e//pvsneslib/lib/libc.obj
bank 0 slot 0 /e//pvsneslib/lib/libm.obj
bank 0 slot 0 /e//pvsneslib/lib/libtcc.obj
...

Please fix the BANKs and SLOTs, and the paths... We need a linkfile these days, because the command line command for linking would just be enormous for bigger projects without such a linkfile...

I'll reply to your other question a bit later...

vhelin, 2019-04-28:
> Sorry to abuse but i have an other question : PVSneslib tool provide library build with its own header. When in a new project you try to change important parameters like SRAMSIZE, rom speed to [...]

Yes, in every library or normal object file linked together needs to have the same .ROMBANKMAPs etc, all the special linking related bits need to be the same. .MEMORYMAP, .ROMBANKMAP, ...

> I read in wla documentation that we can use .BACKGROUND directive to "patch" the rom :
>
> "Currently WLA can also be used as a patch tool. Just include the original
> ROM image into the project with [...]

... I don't think OVERWRITE .SNESHEADER will work. Just use .SNESHEADER to define the SNES header you'll need and will match the ROM you are about to .BACKGROUND.

> I build the .sfc file, then i try to do a news one with wla (sorry, it is done with the old version !) :
>
>     * the new header
>       E:\devkitsnes\bin\wla-65816 -io newhdr.asm newhdr.obj
>
>     * the [...]

When you .BACKGROUND a ROM image, and create SECTIONs to overwrite some parts, you'll need to use OVERWRITE SECTIONs:

.SECTION "zardoz" OVERWRITE
.DBCOS 0.2, 10, 3.2, 120, 1.3
.ENDS

If you don't create OVERWRITE SECTIONs, you'll need to .UNBACKGROUND some room for other types of SECTIONs....

I hope this helps...
> 
> I opened a thread here without success : https://forums.nesdev.com/viewtopic.php?f=12&t=18589
> 
> Is it possible to do it ?
> 
> For clarity reason, i propose to open a new "issue"..

A new issue would be better than add stuff into this thread about completely something else... :D

hoit, 2019-04-28:
Thanks again for all your help. I will try all things during the week and will open a new issue for "rom patching" for others questions