vhelin/wla-dx #211: cmake -G "Unix Makefiles" fails under Linux Mint 17

sverx, 2019-01-15:
This is probably me being a newbie developer under Linux - I can't install WLA-DX on a Linux Mint 17 box.
cmake -G "Unix Makefiles" fails, saying that 'The C compiler "/usr/bin/cc" is not able to compile a simple test program'.
Which are the requisites? I've got:
cmake 3.5.1
gcc 5.4.0
Thanks!

vhelin, 2019-01-15:
I'm not sure at all, but have you tried to install g++ in addition to gcc?

Neui, 2019-01-15:
This is because by default CMake searches for an C and an C++ compiler when there is no explicit language set in `CMakeLists.txt`. This is actually fixed by my PR #184 but it's currently stalled.

sverx, 2019-01-16:
> This is because by default CMake searches for an C and an C++ compiler when there is no explicit language set in `CMakeLists.txt`.

makes sense...

Anyway build succeeded - thanks! :)

vhelin, 2019-01-18:
I added a note to INSTALL saying that one needs gcc, g++, make and cmake to be installed if one wants to compile WLA DX.

closed 2019-01-18 by vhelin