vhelin/wla-dx #212: Accept shadow register names

Kroc, 2019-02-05:
Can we please accept the shadow registers in Z80 code? i.e. `AF'`, `B'`, `C'`, `BC'`, `D'`, `E'`, `DE'`, `H'`, `L'` & `HL'`. This allows us to mark-up complex source code where registers are swapped back and forth and it make it clearer to readers when `HL` is not the same value as `HL'`

Kroc, 2019-02-05:
> You mean create aliases for all opcodes that use registers?
Yes.
> Is this done in some other assembler already?
Not that I know of, though there are many assemblers -- but shadow registers are part of a Z80 chip, so their inclusion seems natural. The reason they are typically not included is because the Z80 instructions `ex AF, AF'` and `exx` _swap_ the regular and shadow sets of registers; none of the opcodes work directly with the shadow registers; so `ld BC', $00` would not load the shadow register, it would only load the normal `BC` register. The inclusion of shadow-registers would only be to indicate in code when a swap has occurred and that the reader should understand that `BC` !- `BC'` in value. (i.e. a different 'variable' is being used).

I'm just in the process of trying to convert some code to WLADX and have all the shadow registers marked up for readability and I'd like to not have to strip this useful metadata. I'm even looking at WLA's source code right now, but I can't work out how to implement this fix myself (there's _so_ few comments in the code)

vhelin, 2019-02-05:
> You mean create aliases for all opcodes that use registers?
> Yes.
> Is this done in some other assembler already?
> Not that I know of, though there are many assemblers -- but shadow
> registers are part [...]

The easiest, and also the most laborous, I think, way to do this is to edit
opcodes_z80.c, and duplicate all opcodes that use registers, and edit the
first line to use shadow registers so that the parser tries to parse the
shadow register version before the original opcode.

Kroc, 2019-02-05:
Okay, working on this now. Thanks for the pointers. If this works, I can give you the fixed-up file.

Kroc, 2019-02-05:
Fixed file: https://gist.github.com/Kroc/d858300dda2443b09ee2d0942f223b9f Thank you very much for the directions!

vhelin, 2019-02-05:
Thanks! I'll add this to the GitHub repo!

closed 2019-02-05 by vhelin