vhelin/wla-dx #213: Z80 `RST` instruction cannot accept labels

Kroc, 2019-02-05:
It appears that WLA-Z80 is coded to only accept the `RST` instruction with the fixed values `$00`, `$08`, `$10`, `$18`, `$20`, `$28`, `$30` & `$38`. Even if labels are created that use these addresses, these will not be recognised:

```asm
interrupt:
.org $08
    rti

.org $1000
    rst interrupt
```

vhelin, 2019-02-05:
The problem with WLA is that labels get their addresses (and values) at the
linking phase in WLALINK. The assembler itself needs the label's value
already in the assembling phase to be able to select the correct opcode...
We would either be able to solve some (fixed sections') labels' addresses
already in the assembling phase, or delay optcode selection to linking
phase.

vhelin, 2019-02-05:
Just to make things clear, Z80 has 8 RTS opcodes with fixed addresses:

  { "RST *", 0xc7, 8, 0, 0x00 },
  { "RST *", 0xcf, 8, 0, 0x08 },
  { "RST *", 0xd7, 8, 0, 0x10 },
  { "RST *", 0xdf, 8, 0, 0x18 },
  { "RST *", 0xe7, 8, 0, 0x20 },
  { "RST *", 0xef, 8, 0, 0x28 },
  { "RST *", 0xf7, 8, 0, 0x30 },
  { "RST *", 0xff, 8, 0, 0x38 },
  { "RST", 0xc7, 0, 0, 0 },         // alias of "RST 0"

If there was a RTS with a variable address then all would work, but in this
case WLA has to decide which of these opcodes it chooses, depending on the
address value that follows RTS...

Kroc, 2019-02-19:
I can live without fixing this; if you think it's architecturally possible, one method might be to expand `RST` to some kind of macro / conditional that resolves at link-time. I don't know yet what the linker is capable of as I'm still looking into that.

vhelin, 2019-09-06:
The same problem exists in GB-Z80, 8008 and the upcoming support for 8080. I think the quickest way of fixing this would be to add support for opcode selection to WLALINK and write such cases to object files for the WLALINK to process. WLA would first emit a filler (opcode) byte to reserve space and and a command for WLALINK to do the processing, and WLALINK would later overwrite that filler with the correct opcode.

This is pretty doable, I think... If I have the time I might implement this later this month, let's see.

closed 2020-04-12 by vhelin