vhelin/wla-dx #215: Documentation doesn't define colon operator for macros

bferguson3, 2019-02-14:
This might be common knowledge for heavily experienced assembly devs, but I haven't been able to find information on it's usage.

From example code, it SEEMS that - 
`#:\1` will reference the _highest order_ byte (OR, memory page?) of a passed label parameter into a macro, and
`#\1` will reference the _lowest order_ byte (or 16-bit value) of a passed label parameter,

but this is ambiguous and the full usage remains obscured.

Could you please add the usage of the colon parameter to the .macro definition in the official docs? 
Thanks!

Neui, 2019-02-15:
Note that it is probably the bank of an label as specified in [3.3 Labels](https://wla-dx.readthedocs.io/en/latest/asmsyntax.html#labels). Maybe just put it in the example code with an reference since it's not strictly macro related in my opinion.

vhelin, 2019-02-17:
Well, I added :\4 to the examples in .MACRO's documents. I hope that helps.
And yes, it's the bank number of label \4.

vhelin, 2019-02-17:
Please reopen this if the latest fix wasn't enough. :)

closed 2019-02-17 by vhelin