vhelin/wla-dx #219: Better error info for "MEM_INSERT" error

Kroc, 2019-02-20:
I'm getting a "MEM_INSERT: Origin ($4000) overflows from bank (0)" error, but it doesn't give any useful information like the line-number to help me track down the fault. I'm converting some code over to WLA-DX and something is producing more bytes than it should be, but I've got no help to go on.

It appears that you can't just print the current assembly point when you want (`.PRINTV`). Some assemblers represent the current PC with`*` or `$`.

vhelin, 2019-02-21:
Ouch, I thought you meant the overflow errors from WLALINK and I added more information to those... I'll check if the WLA's error message could be enhanced.

Kroc, 2019-02-21:
Yeah, I can't even get to the linking phase to be able to list the symbol addresses! I need to check where my labels stop lining up. Some kind of "debug" output of WLA that would print out the current value of symbols as it gets to them would help.

vhelin, 2019-02-21:
Yes, it seems that I can give at least the file name and line number when mem_insert warnings/errors occur in the compiler plus a vague or an exact description of what's going on there. I don't think I can do this today, but I'll try to finish it tomorrow.

vhelin, 2019-02-21:
BTW, Have you tried compiling WLA in debug mode?

Kroc, 2019-02-21:
Can't say I know how to do that; it was quite difficult as it was to compile it for Windows using Chocolately. I can compile it in Linux via WSL pretty easily.

Kroc, 2019-02-21:
I just compiled the current commit and now it won't recognise `im` & `bit` instructions (and probably more)!

vhelin, 2019-02-21:
It seems that something recent has broken the assembler... I'll look into it as well.

vhelin, 2019-02-21:
Ok, the Z80 support should now be fixed. The Z80 decoder was broken by a recent commit that compressed the optcode struct definition in defines.h, but swapped two fields then Z80 support was compiled.

Kroc, 2019-02-21:
The MEM_INSERT error is still occurring, without extra info, as it's happening during "internal pass 2".
Here's the code being compiled: https://github.com/Kroc/Sonic1-Z80-ASM/blob/redo/sonic_the_hedgehog.asm

If you checkout that branch (redo), there's a build.bat and Windows executables included to easily build that source file.

vhelin, 2019-02-22:
Ok, if you try now with the latest sources, is the MEM_INSERT -error/warning better/good enough?

Kroc, 2019-02-22:
Ok, that is giving the line number now, thank you! (it doesn't solve my root problem of why my symbols don't line up, but it is helpful -- I can roughly see how many bytes I'm past where I should be)

Kroc, 2019-02-22:
I have found the source of the overflow!
WLA-DX does not handle multiple `.BYTE`/`.WORD` directives on one line correctly!

```asm
_0e72:
        .WORD _1129     .BYTE $04, $01
        .WORD _113b     .BYTE $04, $00
```

This appears to generate more bytes than it seems!

vhelin, 2019-02-22:
I think WLA DX handles only one directive per line correctly. If it handles more then it's good luck and something I never thought of. :)

vhelin, 2019-02-22:
It seems that WLA thinks that .BYTE in your case is a label, and generates a reference to it?

Kroc, 2019-02-22:
Understandable, but quite a nuisance as I have many, many lines like this and formatting them on one line each is quite important. I can reimplement these as `.STRUCT`/`.DSTRUCT`s but the fact that I *have* to provide a name for each one is pretty rough given many of these instances have unknown purposes at this time.

Kroc, 2019-02-22:
`.STRUCT`/`.DSTRUCT` is untenable given the code I'm working with; it would be too wordy, too wide (can't split lines) and I'd have to name every single row. The code is also littered with tons of ad-hoc tables like this throughout.

I'm not asking you to do the impossible and re-architect, but even some kind of quick-n-dirty fix would be appreciated, such as a symbol to fake a new-line `\` is typical.

```asm
_f4e:                                                                   ;$0F4E
;===============================================================================
; TODO: these rows need to be appended by the level definitons

        .WORD _0f84,    .BYTE $00       ; Green Hill Act 1
        .WORD _0f93,    .BYTE $00       ; Green Hill Act 2
        .WORD _0fde,    .BYTE $01       ; Green Hill Act 3
        .WORD _0fa2,    .BYTE $00       ; Bridge Act 1
        .WORD _0fb1,    .BYTE $00       ; Bridge Act 2
        .WORD _107e,    .BYTE $02       ; Bridge Act 3
        .WORD _0fc0,    .BYTE $00       ; Jungle Act 1
        .WORD _0fcf,    .BYTE $00       ; Jungle Act 2
        .WORD _1088,    .BYTE $03       ; Jungle Act 3
        .WORD _100b,    .BYTE $00       ; Labyrinth Act 1
        .WORD _101a,    .BYTE $00       ; Labyrinth Act 2
        .WORD _1092,    .BYTE $00       ; Labyrinth Act 3
        .WORD _1029,    .BYTE $00       ; Scrap Brain Act 1
        .WORD _1038,    .BYTE $00       ; Scrap Brain Act 2
        .WORD _109c,    .BYTE $00       ; Scrap Brain Act 3
        .WORD _1047,    .BYTE $00       ; Sky Base Act 1
        .WORD _1056,    .BYTE $00       ; Sky Base Act 2
        .WORD _1056,    .BYTE $00       ; Sky Base Act 3
        ;

_0f84:                                                                  ;$0F84
;===============================================================================
        ; Green Hill Act 1
        .WORD _10bd,    .BYTE $50, $68, $1E
        .WORD _10ab,    .BYTE $50, $68, $1E
        .WORD _0f84,    .BYTE $00, $00, $00
        ;

_0f93:                                                                  ;$0F93
;===============================================================================
        ; Green Hill Act 2
        .WORD _10cf,    .BYTE $50, $60, $1E
        .WORD _10ab,    .BYTE $50, $60, $1E
        .WORD _0f93,    .BYTE $00, $00, $00
        ;
_0fa2:                                                                  ;$0FA2
;===============================================================================
        ; Bridge Act 1
        .WORD _10e1,    .BYTE $60, $60, $1E
        .WORD _10ab,    .BYTE $60, $60, $1E
        .WORD _0fa2,    .BYTE $00, $00, $00
        ;

_0fb1:                                                                  ;$0FB1
;===============================================================================
        ; Bridge Act 2
        .WORD _10f3,    .BYTE $80, $50, $1E
        .WORD _10ab,    .BYTE $80, $50, $1E
        .WORD _0fb1,    .BYTE $00, $00, $00
        ;

_0fc0:                                                                  ;$0FC0
;===============================================================================
        ; Jungle Act 1
        .WORD _1105,    .BYTE $70, $48, $1E
        .WORD _10ab,    .BYTE $70, $48, $1E
        .WORD _0fc0,    .BYTE $00, $00, $00
        ;

_0fcf:                                                                  ;$0FCF
;===============================================================================
[... 77 lines cut ...]
        .WORD _107e     .BYTE $00, $00, $00
        ;

_1088:                                                                  ;$1088
;===============================================================================
        ; Jungle Act 3
        .WORD _1183     .BYTE $78, $30, $08
        .WORD _1088     .BYTE $00, $00, $00
        ;

_1092:                                                                  ;$1092
;===============================================================================
        ; Labyrinth Act 3
        .WORD _1183     .BYTE $70, $60, $08
        .WORD _1092     .BYTE $00, $00, $00
        ;

_109c:                                                                  ;$109C
;===============================================================================
        ; Scrap Brain Act 3
        .WORD _1129     .BYTE $68, $40, $08
        .WORD _113b     .BYTE $68, $40, $08
        .WORD _109c     .BYTE $00, $00, $00
        ;
```

vhelin, 2019-02-22:
How did you end up with data like

  .WORD _0f84,    .BYTE $00

? Was it generated by some tool? Is this format common? Would it be worth it to add support for .DB/.DW/.BYTE/.WORD switching on one line?

Kroc, 2019-02-22:
Actually, this is a conversion from my assembler's syntax which uses an implicit record type rather than using directives; that is, the following code defines a record type of a word followed by three bytes, and then any expressions that follow are automatically sized according to the record fields:

```asm
word, byte, byte, byte
_10cf, $50, $60, $1E
_10ab, $50, $60, $1E
_0f93, $00, $00, $00
```

This makes ad-hoc and complex data-tables a lot easier to write. Converting these to WLA-DX meant inserting .WORD/.BYTE directives before the numbers.

And yes, it would be very helpful to somehow support formatting data on one line as this works best with data tables where the fields are a mix & match of words and bytes. I'm honestly surprised you've not come across this need ever before!

vhelin, 2019-02-22:
> Actually, this is a conversion from my assembler's syntax which uses an
> implicit record type rather than using directives; that is, the following
> code defines a record type of a word followed by three bytes, and then any
> expressions that follow are automatically sized according to the record
> fields:
>
> word, byte, byte, byte_10cf, $50, $60, $1E_10ab, $50, $60, $1E_0f93, $00, $00, $00
>
> This makes ad-hoc and complex data-tables a lot easier to write.
> Converting these to WLA-DX meant inserting .WORD/.BYTE directives before
> the numbers.
>
> And yes, it would be very helpful to somehow support formatting data on
> one line as this works best with data tables where the fields are a mix &
> match of words and bytes. I'm honestly surprised you've not come across
> this need ever before!

I'm thinking about adding support for something like

.TABLE word, byte, byte, byte
.TABLEDATA _10cf, $50, $60, $1E
.TABLEDATA _10ab, $50, $60, $1E
.TABLEDATA _0f93, $00, $00, $00

... or something like that. Too much beer today, though. What do you guys
think?

Kroc, 2019-02-23:
You'll think I'm crazy but aesthetics and indentation are incredibly important to me because I go to great lengths to ensure my code is well commented and explained; see for example this section of 6502 code: https://github.com/Kroc/elite-harmless/blob/master/src/math_3d.asm#L431-L604 Therefore I ask that you don't use keywords longer than 7 letters (including dot) otherwise it screws up all the neat formatting! D:

I would recommend using `.DATA` instead of `.TABLEDATA` since "data" still describes what the expressions are, but does not mandate the word size.

There are a lot of strange cases you need to think about with a table/data construct like that. Firstly, how do custom types work in this? This should be reasonably self-effacing I hope:

```
.TABLE word, byte, myType ; where myType is a word and a byte
.DATA 1, 2, 3, 4 ;myType.1 would be 3 & myType.2 would be 4
```

Now, what if you can't fit all the data elements on a single line?
(I've got a level header structure that's 38 bytes long)

```
.TABLE ds 16
.DATA $00, $01, $02, $03, $04, $05, $06, $07
.DATA $08, $09, $0A, $0B, $0C, $0D, $0E, $0F
```

And what does this do?

```
.TABLE ds 16
.DATA $00, $01, $02, $03, $04, $05, $06, $07
.BYTE $FF
.DATA $08, $09, $0A, $0B, $0C, $0D, $0E, $0F
```

Having the ability to combine word/byte directives on one line would at least solve the immediate limitation in the short-term while the more tricky details of tables are worked out -- which I'd be more than glad to help out with as I've 35'000 lines of code I'm working on!

vhelin, 2019-02-23:
> You'll think I'm crazy but aesthetics and indentation are incredibly
> important to me because I go to great lengths to ensure my code is well
> commented and explained; see for example this section of [...]

How about .ROW instead of .DATA? .DATA is perhaps too generic, but .ROW
connects logically to .TABLE...

There are a lot of strange cases you need to think about with a table/data
> construct like that. Firstly, how do custom types work in this? This should
> be reasonably self-effacing I hope:
>
> .TABLE word, byte, myType ; where myType is a word and a byte
> .DATA 1, 2, 3, 4 [...]
The parser needs to be pretty clever to handle things like these,
especially myType.

> And what does this do?
>
> .TABLE ds 16
> .DATA $00, $01, $02, $03, $04, $05, $06, $07
> .BYTE $FF
> .DATA $08, $09, $0A, $0B, $0C, $0D, $0E, $0F
>
> Having the ability to combine word/byte directives on one [...]

If we don't put .DATA's parsing inside .TABLE, but keep .DATA (or .ROW) a
separate directive, then support for this mixing would come for free.

Kroc, 2019-02-23:
.ROW would not work well if you then had to split one 'record' row across multiple lines (i.e. two `.ROW` directives for one row's data), this would be confusing to the reader.

`.ROW` could be used to delineate the beginning of a new row:

```
.TABLE DSB 16
.ROW
.DATA ...
.ROW
.DATA ...
```

This would open up space for row-specific properties, such as adding a label name for a row and generating an index-enum for the table. In a lot of cases, I have a lookup table and I must match the index numbers to the order of the pointers. Code could be reorganised easier if the index numbers could be enumerated automatically by the order of the rows in the table.

Kroc, 2019-02-23:
Example:

```
music:
.TABLE WORD
.ROW ID MUSIC_ID_GREENHILL
.DATA music_greenHill_data
.ROW ID MUSIC_ID_BRIDGE
.DATA music_bridge_data
...
```
Where `.ROW ID <symbol>` defines a symbol with the index number of the row -- or something like that, this is just off the top of my head.

vhelin, 2019-02-25:
> 
> 
> Example:
> 
> ```
> music:
> .TABLE WORD
> .ROW ID MUSIC_ID_GREENHILL
> .DATA music_greenHill_data
> .ROW ID MUSIC_ID_BRIDGE
> .DATA music_bridge_data
> ...
> ```
> 
> Where `.ROW ID <symbol>` defines a symbol with the index number of the row -- or something like that, this is just off the top of my head.

Would there really be some use for such symbols (defines?)?

Anyway, what if .ROW would require a full row of bytes/words, while .DATA could be used to supply just the next n bytes (a partial .ROW)? This way those who would want extra security could use .ROW and the rest could use .DATA?

vhelin, 2019-02-25:
Just to let everybody know, I've started to work on this. Hopefully I'll have something working by tomorrow.

Kroc, 2019-02-25:
> Would there really be some use for such symbols (defines?)?

All the time:

```asm
music_pointers:                                                         ;$C716
;===============================================================================
; TODO  in the original ROM the order needs to be preserved, and there are
;       unused entries, we'll need a way to specify this for an original build,
;       but use auto-numbering for new builds

@greenHill:                             ; index $00
        .DEFINE MUSIC_ID_GREENHILL      $00
        .EXPORT MUSIC_ID_GREENHILL
        .WORD   music_greenHill         ;=$47D0 [$C7D0]

@bridge:                                ; index $01
        .DEFINE MUSIC_ID_BRIDGE         $01
        .EXPORT MUSIC_ID_BRIDGE
        .WORD   music_bridge            ;=$574A [$D74A]

@jungle:                                ; index $02
        .DEFINE MUSIC_ID_JUNGLE         $02
        .EXPORT MUSIC_ID_JUNGLE
        .WORD   music_jungle            ;=$524A [$D24A]

@labyrinth:                             ; index $03
        .DEFINE MUSIC_ID_LABYRINTH      $03
        .EXPORT MUSIC_ID_LABYRINTH
        .WORD   music_labyrinth         ;=$760C [$F60C]

@scrapBrain:                            ; index $04
        .DEFINE MUSIC_ID_SCRAPBRAIN     $04
        .EXPORT MUSIC_ID_SCRAPBRAIN
        .WORD   music_scrapBrain        ;=$5B4F [$DB4F]

@skyBase:                               ; index $05
        .DEFINE MUSIC_ID_SKYBASE        $05
        .EXPORT MUSIC_ID_SKYBASE
        .WORD   music_skyBase           ;=$61A7 [$E1A7]

@titleScreen:                           ; index $06
        .DEFINE MUSIC_ID_TITLESCREEN    $06
        .EXPORT MUSIC_ID_TITLESCREEN
        .WORD   music_titleScreen       ;=$64C3 [$E4C3]

@mapScreen:                             ; index $07
        .DEFINE MUSIC_ID_MAPSCREEN      $07
        .EXPORT MUSIC_ID_MAPSCREEN
        .WORD   music_mapScreen         ;=$663C [$E63C]

@invincibility:                         ; index $08
        .DEFINE MUSIC_ID_INVINCIBILITY  $08
        .EXPORT MUSIC_ID_INVINCIBILITY
        .WORD   music_invincibility     ;=$6704 [$E704]

@actComplete:                           ; index $09
        .DEFINE MUSIC_ID_ACTCOMPLETE    $09
        .EXPORT MUSIC_ID_ACTCOMPLETE
        .WORD   music_actComplete       ;=$68B4 [$E8B4]

@death:                                 ; index $0A
        .DEFINE MUSIC_ID_DEATH          $0A
        .EXPORT MUSIC_ID_DEATH
        .WORD   music_death             ;=$6991 [$E991]

@boss1:                                 ; index $0B
        .DEFINE MUSIC_ID_BOSS1          $0B
        .EXPORT MUSIC_ID_BOSS1
        .WORD   music_boss              ;=$6AC0 [$EAC0]

@boss2:                                 ; index $0C
        .DEFINE MUSIC_ID_BOSS2          $0C
        .EXPORT MUSIC_ID_BOSS2
        .WORD   music_boss              ;=$6AC0 [$EAC0]

@boss3:                                 ; index $0D
        .DEFINE MUSIC_ID_BOSS3          $0D
        .EXPORT MUSIC_ID_BOSS3
        .WORD   music_boss              ;=$6AC0 [$EAC0]

@ending:                                ; index $0E
        .DEFINE MUSIC_ID_ENDING         $0E
        .EXPORT MUSIC_ID_ENDING
        .WORD   music_ending            ;=$6D54 [$ED54]

        ; an unused entry
@unused1:                               ; index $0F
        .DEFINE MUSIC_ID_UNUSED1        $0F
        .EXPORT MUSIC_ID_UNUSED1
        .WORD   music_greenHill         ;=$47D0 [$C7D0]

@specialStage:                          ; index $10
        .DEFINE MUSIC_ID_SPECIALSTAGE   $10
        .EXPORT MUSIC_ID_SPECIALSTAGE
        .WORD   music_specialStage      ;=$712C [$F12C]

        ; a couple of unused entries
@unused2:                               ; index $11
        .DEFINE MUSIC_ID_UNUSED2        $11
        .EXPORT MUSIC_ID_UNUSED2
        .WORD   music_greenHill         ;=$47D0 [$C7D0]
@unused3:                               ; index $12
        .DEFINE MUSIC_ID_UNUSED3        $12
        .EXPORT MUSIC_ID_UNUSED3
        .WORD   music_greenHill         ;=$47D0 [$C7D0]

@allEmeralds:                           ; index $13
        .DEFINE MUSIC_ID_ALLEMERALDS    $13
        .EXPORT MUSIC_ID_ALLEMERALDS
        .WORD   music_allEmeralds       ;=$798C [$F98C]

@emerald:                               ; index $14
        .DEFINE MUSIC_ID_EMERALD        $14
        .EXPORT MUSIC_ID_EMERALD
        .WORD   music_emerald           ;=$7A26 [$FA26]
```

vhelin, 2019-02-25:
> @greenHill:                             ; index $00
>       .DEFINE MUSIC_ID_GREENHILL      $00
>       .EXPORT MUSIC_ID_GREENHILL
>       .WORD   music_greenHill         ;=$47D0 [$C7D0]

But in this case ".ROW ID" would just be an alias for combined .DEFINE and .EXPORT (plus autoincrementing the index). Perhaps it would be better to create a separate directive for that?

.INDEX [definitionName] (optional [INDEX])

.INDEX LABEL_1 0
.INDEX LABEL_2 ; == 1
.INDEX LABEL_3 ; == 2

??? That could be used with .TABLE and .ROW/.DATA and elsewhere as well?

Kroc, 2019-02-25:
That would work too, but `.INDEX` is already a directive :)

vhelin, 2019-02-26:
My commit somehow closed this? Anyway, reopened...

I just committed a first version of .TABLE, .ROW and .DATA. It doesn't do all you requested, but next I'll add support for DSB, DSW and DSL to .TABLE... Also it seems that WLA doesn't support .DSL at all. I think I'll need to add that as well.

Kroc, 2019-02-26:
.LONG is also not supported in Structs / RAMSections. Probably a number of places

vhelin, 2019-02-26:
> .LONG is also not supported in Structs / RAMSections. Probably a number of places

Thanks, I'll fix this.

cr1901, 2019-02-26:
>My commit somehow closed this?

Github will automatically close issues if the commit message follows a specific format. The phrase "fix #219" triggers this behavior it seems :D.

vhelin, 2019-02-26:
Ok, if .TABLE now good enough? :)

Kroc, 2019-02-27:
.ROW will not parse if there's a label before-hand:

```
underwaterPalette:                                                      ;$024B
;===============================================================================
        .TABLE  DSB 16
@tile:  .ROW    $10 $14 $14 $18 $35 $34 $2C $39 $21 $20 $1E $09 $04 $1E $10 $3F
@sprite:.ROW    $00 $20 $35 $2E $29 $3A $00 $3F $14 $29 $3A $14 $3E $3A $19 $25
```

vhelin, 2019-02-28:
> .ROW will not parse if there's a label before-hand:
>
> ```
> underwaterPalette:                                                      ;$024B [...]

It seems that you'll need to have at least one white space between the label's name and .ROW.

@sprite: .ROW

  works, but

@sprite:.ROW

  does not.

This requires a fix to the parser... I can take a look at it today or tomorrow...

vhelin, 2019-02-28:
Actually it was a silly bug in .DATA/.ROW parsing. I forgot to initialize and increment a counter that counted the number of items given to .DATA/.ROW... Should work now.

Ramsis-SNES, 2019-02-28:
Just chiming in to say I appreciate what's happening lately, esp. on the 65816 side of things. Thanks and keep up the good work, Ville! :D

vhelin, 2019-02-28:
For some unknown reason I feel good coding old style ANSI C and making WLA-DX move forward. :) I don't know how long this feeling lasts, but it's nice to know it's making asm coding around the world a little bit easier/better. :)

vhelin, 2019-06-02:
Can this issue be closed?

Kroc, 2019-06-02:
But of course :)

commit fd83e079 (2019-02-26): Added .TABLE, .ROW and .DATA. First attempt to fix #219.
closed 2019-06-02 by Kroc