vhelin/wla-dx #220: RAMSECTIONs, dotted syntax to not advance the pointer

Kroc, 2019-02-22:
The documentation says 

> RAMSECTIONs accept only variable labels and variable sizes, and the syntax to define these is identical to .ENUM (all the syntax rules that apply to .ENUM apply also to .RAMSECTION). Additionally you can embed structures (.STRUCT) into a RAMSECTION. Here’s an example:

The `.ENUM` documentation says:

> You can also use a dotted version of the symbols, but it doesn’t advance the memory address.

However, the syntax to 'hold' the address does not work in `.RAMSECTION`s, I get an error message `ERROR: Couldn't parse ".DS"`

vhelin, 2019-02-27:
The list of dotted no-advance symbols was really small and old. How does it work now?

Kroc, 2019-02-27:
Looks like that's working -- many thanks!

closed 2019-02-27 by Kroc