vhelin/wla-dx #225: Add support for a running ID that's automatically exported

vhelin, 2019-02-28:
.DEFINE ID_1 0
.EXPORT ID_1
.DEFINE ID_2 1
.EXPORT ID_2

... This gets ugly really fast. Perhaps a better solution would be

.ENUMID ID_1 0
.ENUMID ID_2      ; this is 1
.ENUMID ID_3      ; this is 2...

?

Kroc, 2019-03-01:
I like that very much! "ENUMID" is a good directive name. Have a look at Golang's `iota`; it allows you to specify an expression automatically applied to each step.

Kroc, 2019-03-01:
Suggestion: For consistency with define's export attribute, you should require the same for enumid

vhelin, 2019-03-01:
Goland's iota sounds a bit like overkill; Good, but too complex here and for me to write support for such a system. :) I hope this simple .ENUMID I coded is enough...

maxim-zhao, 2019-03-21:
Isn't this the same as enum with the export modifier?

Kroc, 2019-03-21:
You can't start and stop an enum!

maxim-zhao, 2019-03-21:
Aha - I see the use now - a way to generate unique (incrementing) IDs spread through the code. (The examples were all on consecutive lines.) I can see a need to have multiple counters on the go though - so maybe a context parameter is needed?

I tend to work around this by defining the IDs in one place using an `.enum` and then just referencing them in other places.

Kroc, 2019-03-21:
This would be critical in a situation where the order of data determines the indices; e.g. imagine the music in a game. Each chunk of sound data will have an associated index number. If you wish to add songs somewhere in the middle then the index numbers need to be updated across the whole code-base (or the wrong music gets played by the wrong thing). This is trivial with `ENUMID`, i.e. imagine a list of enum IDs, each followed by an `.INCLUDE` of the data.

The ability to continue an enum ID at any arbitrary point would help with being able to group together the disparate code & data of a conceptually single object into one place. For example, in Sonic 1, an enemy object will consist of a block of behaviour code, but also a set of entries in numerous tables spread across the code base, where the index within each of these tables *must* match. If all of this data can be brought together into one file then it makes it very easy to simply add / remove / re-order the objects in the game without manual table-editing over many, many files.

vhelin, 2022-08-29:
I have to say that I don't remember this issue or what should be done here. Or did my commit solve this issue. I see @maxim-zhao pushed a commit into his branch that touched this issue.

Reading through the comments didn't make me any wiser, only more confused.

Can someone explain what we were trying to accomplish here? If not, I'll close the issue as I'm kinda having a feeling that I was overengineering things here...

maxim-zhao, 2022-08-30:
That commit isn't in my fork, the link above seems to be when I synced with the main vhelin/wla-dx and picked up a variant of https://github.com/vhelin/wla-dx/commit/c2d835501751c72374a1a95adf6078ce79669bb4 but it's no longer a commit in my fork for some reason.