vhelin/wla-dx #226: Fail to compile with clang on mac

tsupplis, 2019-03-03:


tsupplis, 2019-03-03:
adding new line at the end of the file resolves the problem

vhelin, 2019-03-03:
Can you tell us which file is causing such problems, and I'll fix it to
GitHub?

tsupplis, 2019-03-03:
wla-dx/decode_spc700.c

vhelin, 2019-03-03:
Thanks, I added a new line to the end of decode_spc700.c!

closed 2019-03-03 by vhelin

tsupplis, 2019-03-03:
problem fixed, thank you

cr1901, 2019-03-03:
@vhelin For future reference, the "newline at end of file" requirement comes from `-pedantic-errors`; for whatever historical reason, ANSI C requires files to be terminated with newlines.

vhelin, 2019-03-03:
> @vhelin For future reference, the "newline at end of file" requirement comes from `-pedantic-errors`; for whatever historical reason, ANSI C requires files to be terminated with newlines.

Then it's weird that GCC 7.4.0 on my computer didn't complain about it...

cr1901, 2019-03-03:
Hrm interesting... wonder why that it's not caught by gcc... I mean, wouldn't be the first time compilers don't do what I expect :P.

I've def ran into said error before in gcc, but not w/ decode_spc700, and Idk if I've used 7.4.0...

commit 24d72cbe (2019-03-03): Added an empty line to the end of the file to make it compile on OSX. Thanks tsupplis! Fixes #226.