vhelin/wla-dx #228: _sizeof labels for structs in ramsections are wrong

aexoden, 2019-04-16:
If I have a struct with two bytes named x and y, and place in them a ramsection like this:

```
.ramsection "Vars" slot 0
    points instanceof point 64
    otherVar db
.ends
```

_sizeof_points will be defined but set to zero. After trying to debug wlalink, it looks like it simply sets the size to the difference between the address of points and points.1.

I don't know if this was intended to work or if was just a side effect of adding _sizeof for ROM labels.

aexoden, 2019-04-20:
This workaround also does not help in cases like this:
```point_data .dsb 1024
point_data_lower dsb 512
point_data_upper dsb 512```

The _sizeof_point_data label will be set to 0, even though anyone referencing it would clearly be looking for 1024.

vhelin, 2019-05-11:
Hi, could you try again with the latest sources? I just committed some code that fixed _sizeof_ labels from RAMSECTIONs that were inside library files. Perhaps the same bug caused the behaviour you encountered?

aexoden, 2019-05-15:
Still seems to define incorrect values unless I apply my workaround.

vhelin, 2019-05-16:
Ok, now I see the problem. WLALINK doesn't know anything about structs, WLA just turns structs into a pile of labels, and WLALINK just calculates distances between consecutive labels when making _sizeof_ -calculations...

I'll try to do the same as your patch does, but without any side effects...

vhelin, 2019-05-16:
I think I got it. Now I'll check the number of dots in the label, when computing label distances, and label with n dots will skip the following labels with n+1 or more dots -> this makes WLALINK understand the label hierarchy when using structs, when computing _sizeof_'s... It's kind of hard to explain. :)

vhelin, 2019-05-16:
> This workaround also does not help in cases like this:
> 
> ```
> point_data .dsb 1024
> point_data_lower dsb 512
> point_data_upper dsb 512
> ```
> 
> The _sizeof_point_data label will be set to 0, even though anyone referencing it would clearly be looking for 1024.

This is actually another problem not solved by my latest fix. In this case WLALINK should need real metadata (.dsb 1024) from WLA... Currently no struct metadata is transported from WLA to WLALINK so fixing this would require quite a lot of work.

vhelin, 2019-05-16:
If you get _sizeof_[label]s for structs in ramsections to work with my latest fix, please close this issue, and open a new one for the problem with _sizeof_point_data ...

aexoden, 2019-05-17:
Everything for the original issue looks like it's working now. Thanks. I'll open another issue for the secondary issue.

closed 2019-05-17 by aexoden