vhelin/wla-dx #232: wlalink upgrade from 9.4 to latest version - linkfile questions

hoit, 2019-05-05:
Hi,

I am trying to upgrade wla to use the latest version. The first part using wla-65816 works correctly now (i began questions in a [previous thread](https://github.com/vhelin/wla-dx/issues/207) ) but i still have difficulties with linkfile :

```
[objects]
E:\snes-examples\hello_world\data.obj
E:\snes-examples\hello_world\hdr.obj
E:\snes-examples\hello_world\hello_world.obj
[libraries]
bank 0 slot 1 E:\pvsneslib\lib\crt0_snes.obj
bank 0 slot 0 E:\pvsneslib\lib\libc.obj
bank 0 slot 0 E:\pvsneslib\lib\libm.obj
bank 0 slot 0 E:\pvsneslib\lib\libtcc.obj

```
When i try with this file, i get this ouput :

E:\snes-examples\hello_world\data.obj: MEM_INSERT: Overwrite at $7fc8 (old $48 n
ew $44).
   ^ Writing fixed data block from "E:\pvsneslib\lib\crt0_snes.obj".
E:\snes-examples\hello_world\data.obj: MEM_INSERT: Overwrite at $7fca (old $4c n
ew $46).
   ^ Writing fixed data block from "E:\pvsneslib\lib\crt0_snes.obj".
E:\snes-examples\hello_world\data.obj: MEM_INSERT: Overwrite at $7fcb (old $4c n
ew $41).
   ^ Writing fixed data block from "E:\pvsneslib\lib\crt0_snes.obj".
E:\snes-examples\hello_world\data.obj: MEM_INSERT: Overwrite at $7fcc (old $4f n
ew $55).
   ^ Writing fixed data block from "E:\pvsneslib\lib\crt0_snes.obj".
E:\snes-examples\hello_world\data.obj: MEM_INSERT: Overwrite at $7fcd (old $20 n
ew $4c).
   ^ Writing fixed data block from "E:\pvsneslib\lib\crt0_snes.obj".
E:\snes-examples\hello_world\data.obj: MEM_INSERT: Overwrite at $7fce (old $57 n
ew $54).
   ^ Writing fixed data block from "E:\pvsneslib\lib\crt0_snes.obj".
E:\snes-examples\hello_world\data.obj: MEM_INSERT: Overwrite at $7fcf (old $4f n
ew $20).
   ^ Writing fixed data block from "E:\pvsneslib\lib\crt0_snes.obj".
E:\snes-examples\hello_world\data.obj: MEM_INSERT: Overwrite at $7fd0 (old $52 n
ew $43).
   ^ Writing fixed data block from "E:\pvsneslib\lib\crt0_snes.obj".
E:\snes-examples\hello_world\data.obj: MEM_INSERT: Overwrite at $7fd1 (old $4c n
ew $41).
   ^ Writing fixed data block from "E:\pvsneslib\lib\crt0_snes.obj".
E:\snes-examples\hello_world\data.obj: MEM_INSERT: Overwrite at $7fd2 (old $44 n
ew $52).
   ^ Writing fixed data block from "E:\pvsneslib\lib\crt0_snes.obj".
E:\snes-examples\hello_world\data.obj: MEM_INSERT: Overwrite at $7fd3 (old $20 n
ew $54).
   ^ Writing fixed data block from "E:\pvsneslib\lib\crt0_snes.obj".
E:\pvsneslib\lib\crt0_snes.obj: crt0_snes.asm:205: PARSE_STACK: Unresolved refer
ence to "__nmi_handler".

"Unresolved reference to "__nmi_handler" but it seems to be correct in crt0_snes.asm file.

In fact, what is a slot because i saw that crt0_snes.obj use few differents (line 3 and line 35). In this case, should i have only 1 slot by asm file ?

I also tried to use all file in [libraries] part as simple [objects] and the result is the same but seems to be more easy because i don't need to manage bank and slots :

```
[objects]
E:\snes-examples\hello_world\data.obj
E:\snes-examples\hello_world\hdr.obj
E:\snes-examples\hello_world\hello_world.obj
E:\pvsneslib\lib\crt0_snes.obj
E:\pvsneslib\lib\libtcc.obj
E:\pvsneslib\lib\libc.obj
E:\pvsneslib\lib\libm.obj
```
[... 280 lines cut ...]
    lda #$6000 ; 2nd byte + rts
    sta.b move_backwards_insn + 2

    pea $ffff - __endramsectionram.data
    pea :__endramsectionram.data
    pea __endramsectionram.data
    jsr.l __malloc_init
    pla
    pla
    pla

    stz.b tcc__r0
    stz.b tcc__r1

    jsr.l main

    ; write exit code to $fffd
    lda.b tcc__r0
    sep #$20
    sta $fffd
    rep #$20
    stp
.ends

```

vhelin, 2019-05-05:
Ok, multiple questions. I hope I don't miss any of them. :)

1. I don't know about what data is overwriting who, but could it be something from the VBlank as no section is mentioned in the overwrite warning?

2. ""Unresolved reference to "__nmi_handler" but it seems to be correct in crt0_snes.asm file."

"__nmi_handler" seems to be a local label. Remove the two underscores from the beginning of it and it might just work (from label and the references).

Here's a quote from (old) WLA's documentation:

"3.3. Labels

Labels are ordinary strings (which can also end to a ':'). Labels starting
with "_" are considered to be local labels and do not show outside sections
where they were defined, or outside object files, if they were not defined
inside a section."

4. "In fact, what is a slot because i saw that crt0_snes.obj use few differents (line 3 and line 35). In this case, should i have only 1 slot by asm file ?"

Slot is an area in the CPU's memory. If you have an architecture where you can switch ROM banks in address $0000-$3fff, then define $0000-$3fff as a slot. If you have RAM in $f000-$ffff then define that area also as a slot.

5. "I also tried to use all file in [libraries] part as simple [objects] and the result is the same but seems to be more easy because i don't need to manage bank and slots :"

Libraries are collections of code &/ data that you can directly reuse in multiple projects. Objects should be used for everything else. And nothing forbids you from compiling everything as objects. :) I could also be said that library support in WLA DX is not worth the effort as they bring very little to the table...

6. ".ORG 0                          ; .ORG 0 is really $8000, because the slot starts at $8000"

In this case you could write

.ORGA $8000

if that's what you'd like to do. Check out the documents for .ORGA...

Hope this helps!

hoit, 2019-05-06:
Yes thanks, it is more clear but more complicated that i thought..
Replace __nmi_handler by nmi_handler works but wlalink fail after on 

E:\pvsneslib\lib\crt0_snes.obj: crt0_snes.asm:251: PARSE_STACK: Unresolved reference to "__endsection.data".

It is not declared in this file but in tcc65816 (tiny c compiler) which use its own version of wla (the old 9.4 and never updated). (i think it will be the same for others __xxxxx like __endsection...)
In old sources, i found in write.c (from wla dx folder) part which generate labels __startramsection and __endramsection but it seems to be different in latest version. Is it linked to [ramsections] in linkfile ?

I hope that the first step for me is not to update tcc with the latest version of wla :(

vhelin, 2019-05-08:
You cannot link together objects / libraries compiled with WLA v9.4 and v9.8... The file formats have changed a lot between those versions. Version 9.4 is now 11 years old. :D 

If you have a label that starts with "__", then it is not visible outside the section / file where it was defined. Not any more with v9.8...

I'm sorry. I don't know all the details, but it is very, very possible that you'll need to make a version of tcc65816 that uses the latest WLA DX v9.8...

hoit, 2019-05-09:
Ok, i will try to update wla dx in tcc816 to see if it solve something but i need to analyse code and understand how it works.

vhelin, 2019-05-09:
Great! Let us know asap if you hit any problems, and also if you succeed in making the upgrade. :)

hoit, 2019-05-09:
In fact, i just build tcc816 (in snes-sdk) by replacing wla_dx folder by the new one with few adaptations in makefiles.
I am on Windows and it seems to be less hard than i thought. it is not clear and a lot of folders seems no more usefull or unused in snes-sdk but i let them for now.

It strange, because in snes-sdk\trunk\libs folder, crt0_snes.asm file is provided and has been build (i have crt0_snes.obj). This is the original file (with __nmi_handler) and i didn't got errors like in my 1st message (i guess that functions will be called only to compile a game).

So i replaced tcc exe by the new one in PVSneslib folder (which should use the latest version of wladx), crt0_snes.obj by the new in lib folders and try to build hello_world :

it fails because of "__nmi_handler". I removed "__" and i finally get :

E:\pvsneslib\lib\crt0_snes.obj: crt0_snes.asm:251: PARSE_STACK: Unresolved refer
ence to "__endsection.data".
make: *** [/e/snes-examples/hello_world/hello_world.sfc] Error 1

This issue directly comes from crt0_snes.asm. I think that upgrade of wladx on tcc816 will be usefull later but it is not the cause of my issue here.

I don't know how to update crt0_snes.asm (i don't know asm!) to remove all "__" (__endramsectionram, __endramsection, __startramsection).

Does anybody can help me to update it ?
In parallel, i am looking on forums :)

vhelin, 2019-05-09:
If you could zip/rar the required folders and put that on the net, or email me it ville dot helin at gmail dot com, I could try to take a look at it (latest on Saturday, I'm moving to a new home tomorrow morning)... Please also give me instructions of what to do! :)

I have Windows 10 and Cygwin installed...

hoit, 2019-05-10:
I sent it yesterday by email, i hope i won't be in your junk mail  :)

vhelin, 2019-05-10:
Thanks, I got it! I just finished moving into a new apartment, I'll start
looking at this later today...

- Ville

vhelin, 2019-05-10:
I got the sources from https://code.google.com/archive/p/snes-sdk/source/default/source and then replaced the wla_dx directory with the latest 9.8a... The I issued "make", and it compiled WLA DX ok, but then ends as follows:

...
make[1]: Entering directory '/cygdrive/v/snes-sdk/trunk/tcc-65816'
make[1]: '816-tcc' is up to date.
make[1]: Leaving directory '/cygdrive/v/snes-sdk/trunk/tcc-65816'
cd libs && make PREFIX=/usr/local/snes-sdk
make[1]: Entering directory '/cygdrive/v/snes-sdk/trunk/libs'
../tcc-65816/816-tcc  -I. -I../tcc-65816/include -o libm_c.pasm -c libm_c.c
make[1]: *** [Makefile:5: libm_c.pasm] Aborted (core dumped)
make[1]: *** Deleting file 'libm_c.pasm'
make[1]: Leaving directory '/cygdrive/v/snes-sdk/trunk/libs'
make: *** [Makefile:14: libs] Error 2
...

So in fact ../tcc-65816/816-tcc crashes.

...

ville@DESKTOP-EDMJKER /cygdrive/v/snes-sdk/trunk/libs
$ gdb
GNU gdb (GDB) (Cygwin 8.1.1-1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-cygwin".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) file  ../tcc-65816/816-tcc
Reading symbols from ../tcc-65816/816-tcc...done.
(gdb) run  -I. -I../tcc-65816/include -o libm_c.pasm -c libm_c.c
Starting program: /cygdrive/v/snes-sdk/trunk/tcc-65816/816-tcc -I. -I../tcc-65816/include -o libm_c.pasm -c libm_c.c
[New Thread 12136.0x1e78]
[New Thread 12136.0x4d44]
[New Thread 12136.0x39d0]
[New Thread 12136.0x3c98]

Thread 1 "816-tcc" received signal SIGABRT, Aborted.
0x0000000100401441 in tcc_free (ptr=<optimized out>) at tcc.c:965
965         free(ptr);
(gdb) bt
#0  0x0000000100401441 in tcc_free (ptr=<optimized out>) at tcc.c:965
#1  tok_str_free (str=<optimized out>) at tcc.c:2269
#2  free_defines (b=0xffffc2d0) at tcc.c:2475
#3  0x0000000000000000 in ?? ()
Backtrace stopped: previous frame inner to this frame (corrupt stack?)
(gdb) l
960     static inline void tcc_free(void *ptr)
961     {
962     #ifdef MEM_DEBUG
963         mem_cur_size -= malloc_usable_size(ptr);
964     #endif
965         free(ptr);
966     }
967
968     static void *tcc_malloc(unsigned long size)
969     {

...

I tried to compile it with -O0 and -DMEM_DEBUG, but got no more information (in fact got less for some reason)...

You mention many .rar files in your email. Where can I download them? They are not attached to the email...

vhelin, 2019-05-10:
If I comment the free() away (make tcc to leak memory a lot, I guess) 816-tcc doesn't crash and I can move on... :) More later.

vhelin, 2019-05-10:
Ok, got a little bit further:

ville@DESKTOP-EDMURKLS /cygdrive/v/snes-sdk/trunk/libs
$ make
../wla_dx/binaries/wla-65816 -i -o libm.obj libm.asm
../wla_dx/binaries/wla-65816 -i -o libtcc.obj libtcc.asm
../tcc-65816/816-tcc  -I. -I../tcc-65816/include -o libc_c.pasm -c libc_c.c
make: *** [Makefile:5: libc_c.pasm] Segmentation fault (core dumped)

ville@DESKTOP-EDMURKLS /cygdrive/v/snes-sdk/trunk/libs
$ gdb
GNU gdb (GDB) (Cygwin 8.1.1-1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-pc-cygwin".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) file ../tcc-65816/816-tcc.exe
Reading symbols from ../tcc-65816/816-tcc.exe...done.
(gdb) run  -I. -I../tcc-65816/include -o libc_c.pasm -c libc_c.c
Starting program: /cygdrive/v/snes-sdk/trunk/tcc-65816/816-tcc.exe -I. -I../tcc-65816/include -o libc_c.pasm -c libc_c.c
[New Thread 15612.0x2e24]
[New Thread 15612.0x1460]
[New Thread 15612.0x2820]
[New Thread 15612.0x1f08]
[New Thread 15612.0x43c8]
[New Thread 15612.0x3020]

Thread 1 "816-tcc" received signal SIGSEGV, Segmentation fault.
0x000000010040bcaa in sym_push (v=2228285, type=0xffffc3a0, r=0, c=0) at tcc.c:1715
1715            ts = table_ident[(v & ~(SYM_STRUCT)) - TOK_IDENT];
(gdb) bt
#0  0x000000010040bcaa in sym_push (v=2228285, type=0xffffc3a0, r=0, c=0) at tcc.c:1715
#1  0x0000000100429ad2 in decl (l=240) at tcc.c:9443
#2  0x000000010042a1eb in tcc_compile (s1=0x60004a8b0) at tcc.c:9574
#3  0x000000010042e083 in tcc_add_file_internal (s1=0x60004a8b0, filename=0xffffcc75 "libc_c.c", flags=1) at tcc.c:9916
#4  0x000000010042e14d in tcc_add_file (s=0x60004a8b0, filename=0xffffcc75 "libc_c.c") at tcc.c:9933
#5  0x000000010042fde4 in main (argc=7, argv=0xffffcbd0) at tcc.c:10619

...

To my eye tcc is pretty buggy. Have you tried to contact its author, Fabrice Bellard? I have no idea of tcc's code, so I'm not sure I can make this work...

hoit, 2019-05-10:
I uploaded files here :
trunk.rar :  http://s000.tinyupload.com/?file_id=02371891205245714524
lib.rar : http://s000.tinyupload.com/?file_id=31333585284359941073
hello_world.rar : http://s000.tinyupload.com/?file_id=24688188881016047065
devkitsnes.rar : http://s000.tinyupload.com/?file_id=84318916626047766667

tcc seems to be more easy to build with msys. It is available in trunk.rar as exe but i think you don't need it.
Fabrice Bellard doesn't maintain tcc anymore. The port for tcc65816 has been developed by an other person who never continue the project !
I think error comes only from **crt0_snes.asm** file. You can see it in hello_world.rar. But to build hello_world, dependencies are in lib.rar and devkitsnes.rar.

To try it, i think it will be more easy if you merge libs and hello_word files then you wla/wlalink them to see result.
I need to clean code to create a minimal code...

hoit, 2019-05-10:
ok, should be more clear now :)

Forget all previous .rar files, I merged all necessary dependencies to build the hello_world here :
http://s000.tinyupload.com/?file_id=76826166167718687994

There are a lot of asm files (which often comes from .c files from PVSneslib that are converted with tcc816).

One file is crt0_snes.asm and has been written in asm directly. It worked with wla 9.4 but failed during wlalink with wla 9.8.
The first error was :

crt0_snes.obj: crt0_snes.asm:205: PARSE_STACK: Unresolved refer
ence to "__nmi_handler".

it has been solved by renaming __nmi_handler to nmi_handler

The second error :
crt0_snes.asm:251: PARSE_STACK: Unresolved reference to "__endsection.data".

So i tried the same thing, remove "__" when variable start with (__endsection, __startsection, __endramsection and __startramsection)

but it produce 

crt0_snes.obj: crt0_snes.asm:251: PARSE_STACK: Unresolved reference to "endsecti
on.data".

I think there are new rules in wla 9.8 and the file need some modifications.

Initial crt0_snes.asm file is available here :
https://github.com/alekmaul/pvsneslib/blob/master/pvsneslib/source/crt0_snes.asm

vhelin, 2019-05-10:
Works here 100%:

ville@DESKTOP-EDMTVDH /cygdrive/v/hello_world
$ ../wla-dx-master-2/binaries/wla-65816.exe -v -o crt0_snes.o crt0_snes.asm
Pass 1...
Directive checks...
Internal pass 1...
Internal pass 2...
Free space at $0000-$0090.
Free space at $00b7-$7fb1.
Free space at $7fb6-$7fbf.
Free space at $7fdc-$3ffff.
Bank 00 has 32698 bytes (99.79%) free.
Bank 01 has 32768 bytes (100.0%) free.
Bank 02 has 32768 bytes (100.0%) free.
Bank 03 has 32768 bytes (100.0%) free.
Bank 04 has 32768 bytes (100.0%) free.
Bank 05 has 32768 bytes (100.0%) free.
Bank 06 has 32768 bytes (100.0%) free.
Bank 07 has 32768 bytes (100.0%) free.
262074 unused bytes of total 262144.
Total 262144 additional bytes (from headers and footers).
Total size 524288 bytes.

vhelin, 2019-05-10:
Or is your linking failing? I don't really know what you are doing, what commands you are running. "make" gives me this:

$ make
Makefile:12: *** "Please create an environment variable DEVKITSNES with path to its folder and restart application. (you can do it on windows with <setx DEVKITSNES "/c/snesdev/">)".  Stop.

... and I don't know how to define DEVKITSNES.

ville@DESKTOP-EDMTVDH /cygdrive/v
$ li
total 10304
dr-xr-x---+ 1 ville          ville               0 Jan 13  2017 '$RECYCLE.BIN'/
drwxrwx---+ 1 Administrators ville               0 May 11 00:21  ./
dr-xr-xr-x  1 ville          ville               0 May 11 00:40  ../
drwxrwx---+ 1 ville          ville               0 May 11 00:37  hello_world/
drwxrwx---+ 1 ville          ville               0 Nov 19  2017  Ideas/
drwxrwx---+ 1 ville          ville               0 Mar 18  2016  snes-sdk/
drwxr-x---  1 Unknown+User   Unknown+Group       0 Jan 19  2018 'System Volume Information'/
drwxrwx---+ 1 ville          ville               0 May  6 19:56  Unity/
drwxrwx---+ 1 ville          ville               0 May  6 20:13 'Unity Projects'/
drwxrwx---+ 1 ville          ville               0 Apr 28 20:38  wla-dx-master/
drwxrwx---+ 1 ville          ville               0 May 11 00:36  wla-dx-master-2/
-rwxrwx---+ 1 ville          ville         2764933 Apr 28 19:55  wla-dx-master-before-branch.rar*
-rwxrwx---+ 1 ville          ville         2557377 Mar 10 15:49  wla-dx-master-old-1.rar*
-rwxrwx---+ 1 ville          ville         2565660 Mar 10 19:44  wla-dx-master-old-2.rar*
-rwxrwx---+ 1 ville          ville         2612364 Mar 10 20:11  wla-dx-master-old-3.rar*

vhelin, 2019-05-10:
/cygdrive/v/wla-dx-master-2 has the latest WLA DX v9.8a
/cygdrive/v/hello_world has your hello_world

hoit, 2019-05-10:
It will be more easy if you use the rar i uploaded from my previous message ( this one : http://s000.tinyupload.com/?file_id=76826166167718687994 )

yes, **wla-65816.exe -v -o crt0_snes.o crt0_snes.asm** works correctly, it is the linking command which failed and gave :

crt0_snes.obj: crt0_snes.asm:251: PARSE_STACK: Unresolved reference to "endsecti
on.data".

because i already removed the "__" for __endsection, __startsection, __endramsection and __startramsection.
The previous version with the double "__" failed too.

The initial version of the file is available here :
https://github.com/alekmaul/pvsneslib/blob/master/pvsneslib/source/crt0_snes.asm

vhelin, 2019-05-11:
Ok, now I got it, how to test this thing... :)

It seems that v9.4's WLALINK generates (wlalink/write.c) labels for section start and end.

                                /* generate labels for section (group) start and end
                                   (not for text sections) */
                                if(strncmp(s->name,".text", 5)) { //totalsections > 1) {
                                  struct label* l = malloc(sizeof(struct label));
                                  sprintf(l->name, "__startsection%s", s->name);
                                  l->file_id = s->file_id;
                                  l->file_id_source = s->file_id_source;
                                  l->linenumber = 0;
                                  l->rom_address = s->address;
                                  l->section = s->id;
                                  l->section_status = s->status;
                                  l->bank = s->bank;
                                  l->base = s->base;
                                  l->slot = s->slot;
                                  l->status = LABEL_STATUS_LABEL;
                                  l->address = s->address;
                                  add_label(l);
                                  struct label* ll = malloc(sizeof(struct label));
                                  *ll = *l;
                                  sprintf(ll->name, "__endsection%s", s->name);
                                  ll->rom_address = sa[p+j-1]->address;
                                  ll->address = sa[p+j-1]->address + sa[p+j-1]->size;
                                  add_label(ll);
                                }

... but only for SUPERFREE sections? It is very possible that such a feature was added by the author of tcc or someone else as the code is not ANSI C89 and doesn't look like at least what I'd do...

vhelin, 2019-05-11:
I could add such a feature to WLA DX, but I need to think about it first. Perhaps the best place to add those labels is the assembler, not the linker? A new flag (-X) that enables the label generation?

vhelin, 2019-05-11:
Ok, if you now give -x flag to the assembler, it'll create SECTIONSTART_[section name] and SECTIONEND_[section name] labels for each section that should do the trick.

Let us know if there are more problems!

hoit, 2019-05-11:
I was not very clear with my explanations :)
Cool, thanks a lot, you are very reactiv.
I was no here today but i will try it quickly.

hoit, 2019-05-12:
Bad news :(
I tried to generate crt0_snes.obj with this command:
wla-65816 -x -v -o crt0_snes.obj crt0_snes.asm

then link all files :

wlalink.exe linkfile test.sfc i i already have unresolved reference :

E:\snes-examples\hello_world>wlalink.exe linkfile test.sfc
crt0_snes.obj: crt0_snes.asm:251: PARSE_STACK: Unresolved reference to "__endsecti
on.data".

or if i remove the double "__" (like for __nmi_handler) :

E:\snes-examples\hello_world>wlalink.exe linkfile test.sfc
crt0_snes.obj: crt0_snes.asm:251: PARSE_STACK: Unresolved reference to "endsecti
on.data".

In fact, i am not really sure where i need to appli -x parameter.
In crt0_snes.asm, the function which produce error is tcc__start and i saw its name only in hdr.asm ( under .SNESEMUVECTOR section).

I also tried to apply -x parameter to all files to see the result but seems to be the same with some others errors :

```
E:\snes-examples\hello_world>wlalink.exe linkfile test.sfc
libc.obj: libc_c.asm:436: TRY_PUT_LABEL: Label "SECTIONSTART_.text_0x0" was defi
ned more than once.
libc.obj: libc_c.asm:250: TRY_PUT_LABEL: Label "SECTIONEND_.text_0x0" was define
d more than once.
libc.obj: libc_c.asm:12433: TRY_PUT_LABEL: Label "SECTIONSTART_ram.data" was def
ined more than once.
libc.obj: libc_c.asm:12451: TRY_PUT_LABEL: Label "SECTIONEND_ram.data" was defin
ed more than once.
libc.obj: libc_c.asm:12451: TRY_PUT_LABEL: Label "SECTIONSTART_.data" was define
d more than once.
libc.obj: libc_c.asm:12451: TRY_PUT_LABEL: Label "SECTIONEND_.data" was defined
more than once.
libc.obj: libc_c.asm:12451: TRY_PUT_LABEL: Label "SECTIONSTART_.rodata" was defi
ned more than once.
libc.obj: libc_c.asm:12464: TRY_PUT_LABEL: Label "SECTIONEND_.rodata" was define
d more than once.
libc.obj: libc_c.asm:12464: TRY_PUT_LABEL: Label "SECTIONSTART_.bss" was defined
 more than once.
libc.obj: libc_c.asm:12497: TRY_PUT_LABEL: Label "SECTIONEND_.bss" was defined m
ore than once.
libm.obj: libm.asm:45: TRY_PUT_LABEL: Label "SECTIONSTART_.fp" was defined more
than once.
libm.obj: libm.asm:17: TRY_PUT_LABEL: Label "SECTIONEND_.fp" was defined more th
an once.
crt0_snes.obj: crt0_snes.asm:251: PARSE_STACK: Unresolved reference to "__endsec
tion.data".

```

Should i use it differently ?
Thanks

hoit, 2019-05-12:
Ok i was not concentrated and said bullshit... i need to adapt few files and adapt names to see if it work. I’ll try soon....

hoit, 2019-05-13:
I tested it correctly this time :)
Sections that doesn't work in crt0_snes.asm seems to come from libc.asm, it is this one that need to be build with -x parameter.
I replaced startsectionxxx and endsectionxxx by the new SECTIONSTART and SECTIONEND.
Then i got a lot of errors of Unresolved reference to __xxxxx which came from  libc_c.asm. For each, i just removed the "_" or "__" before the name like for nmi_handler. (i don't know what will be the final impact on the rom...)

Now, wlalink.exe linkfile test.sfc produce this error :

crt0_snes.obj: crt0_snes.asm:243: COMPUTE_PENDING_CALCULATIONS: Result (339/$153
) of a computation is out of 8-bit range.

crt0_snes.asm line 243 is :     sta.b nmi_handler + 2
and nmi_handler is declared like that : nmi_handler dsb 4

If you have any idea, i will be happy to get it otherwise, i will continue to investigate on it :)

vhelin, 2019-05-13:
Haha, sorry, I'm too drunk today to check it out. I'll try to look into it tomorrow. :)

vhelin, 2019-05-14:
Did you solve this already? I cannot check it at the moment, which it is, but either

- nmi_handler has to be closer to crt0_snes.asm line 243 (distance in memory must be less than 127-2)
or
- nmi_handler has to be in memory in an address that's less than 255-2

I'm betting on the first one...

hoit, 2019-05-14:
Not again solved and next days will be very busy. I will have more time sunday i hope.

It is really strange like restriction :)
I'll try it, thanks !

vhelin, 2019-05-16:
I checked out the 65816 instructions, and 

   sta.b nmi_handler + 2

is "store accumulator", and "nmi_handler + 2" must fit inside a byte (0-255), it seems... It's actually a requirement by sta(.b), not by WLA DX... :) There is also a 16-bit sta, sta.w, but using that instead of sta.b here, I don't know if it would work. Probably not. I'd need to see the code and spend time understanding it, before I could say yes or no...

hoit, 2019-05-17:
"nmi_handler has to be closer to crt0_snes.asm line 243 (distance in memory must be less than 127-2)"
=> I tried to move .ramsection ".registers" (which contain nmi_handler dsb 4) just before .section ".start" where tcc__start function has error but result is the same.

I tried your other id : use sta.w instead of sta.b and this line compile correctly now but the same error occurs few lines after. Still without understanding final impact, I changed all lines with this error to use ".w" to finally get :

libm.obj: libm.asm:319: COMPUTE_PENDING_CALCULATIONS: Result (443/$1bb) of a computation is out of 8-bit range.

on this line :

-         sty  tcc__E - 1,X

STY means : STore Y register
"When the m flag is 0, LDA, STA, and STZ are 16-bit operations, and when the m flag is 1, LDA, STA, and STZ are 8-bit operations."

but i'm now looking what is m flag and how to set it!

Really, with all errors since the 1st message, i am not sure to be in the right way : i think that 65816 codes (and 8/16/24 bits constraints) are the same for any version of wla. So i guess that this error "out of range" should already appear in oldest version used. Is it possible that is due to an internal change in wla or the architecture used to compile the project ?

To be continued... :)

vhelin, 2019-05-21:
> => I tried to move .ramsection ".registers" (which contain nmi_handler dsb 4) just before .section ".start" where tcc__start function has error but result is the same.
>
> I tried your other id : use [...]

I'm afraid that if you do that change without understanding the code itself you'll end up with non-working code, even if you manage to compile it without errors...

> occurs few lines after. Still without understanding final impact, I changed all lines with this error to use ".w" to finally get :
>
> **libm.obj: libm.asm:319: COMPUTE_PENDING_CALCULATIONS: Result [...]

I don't think WLA handles that, you'll need to manage the m-flag status in your code yourself... At least 16-bit STA is 0x8D and 8-bit STA is 0x85, and if the parameter is 16-bit WLA will use 0x8D.

> 
> Really, with all errors since the 1st message, i am not sure to be in the right way : i think that 65816 codes (and 8/16/24 bits constraints) are the same for any version of wla. So i guess that this error "out of 

They should be unless there has been some fixes or changes. Reading the CHANGELOG between v9.4 and the latest might bring light to this matter...

> range" should already appear in oldest version used. Is it possible that is due to an internal change in wla or the architecture used to compile the project ?

It is very possible that something has changed inside WLA or WLALINK and this is the cause of the errors. New features or bug fixes might all affect this.

> 
> To be continued... :)

I hope we can solve this issue...

If you could once again let me have the sources and makefiles and etc. I could once again try to help us go past at least some issues... :)

hoit, 2019-05-23:
Your help is really appreciable :) We will solve it, i am sure !
But like you said, if i change without understanding the code i will get non-working code (and i already dit it!).

I really need to learn asm and especially 65816 but it will ask me more time.

This is final sources :

http://s000.tinyupload.com/?file_id=77760737083266259446

it contains all necessaries file and i removed makefiles with dependencies. You just need to execute command available in command_list.txt (wla and wlalink).

The code break on wlalink command with this error :
crt0_snes.obj: crt0_snes.asm:243: COMPUTE_PENDING_CALCULATIONS: Result (339/$153) of a computation is out of 8-bit range.

If you want to check it, i will be happy but don't loose too much time, i think i need to restart from the beginning to avoid issues.

vhelin, 2019-05-26:
I think I found the culprit: RAMSECTION ".fp" is the biggest RAMSECTION that goes into slot 1. And bigger RAMSECTIONs are placed into SLOTs before smaller ones. ".fp" goes in first, before RAMSECTION ".registers" where nmi_handler is thus nmi_handler's absolute address is > 0xff and the code fails...

Perhaps the RAMSECTION sorting came as a new feature after 9.4 and before that it worked, because the files were given in correct order (".registers" was given in linkfile before ".fp"...).

So we need a way or prioritizing RAMSECTIONs... I could introduce a new parameter for sections, like "PRIORITY", that could be used when we sort the sections. Sounds like a plan. I'll try to do that tomorrow. :)

vhelin, 2019-05-27:
Ok, added PRIORITY to .SECTIONs and .RAMSECTIONs, but then we get another problem:

libc.obj: sprites.asm:160: COMPUTE_PENDING_CALCULATIONS: Result (-423460860/$e6c28004) of a computation is out of 24-bit range.

I'll check what this is all about next...

vhelin, 2019-05-27:
Just for reference, I changed linke 3 of crt0_snes.asm to

  .ramsection ".registers" bank 0 slot 1 priority 1

and we got to this new problem...

vhelin, 2019-05-27:
WLALINK's write.c:get_snes_pc_bank() messes up this reference, and I don't know what is going on... :P

vhelin, 2019-05-27:
Looking at SNES's memory architecture I think I'm starting to understand this... Depending on the BANK the SLOT mapping of SNES changes. This is kinda crazy.  At this moment, I'm not sure how to handle the two WRAM banks $7E and $7F...

hoit, 2019-05-27:
ok really good news for the solved issue :) but i don't know how i can help you for now.

I'm still learning asm but it is ... not common ! I already read better tutorials but i keep motivation!

vhelin, 2019-05-27:
Another fix: libm.asm line 7:

  .ramsection ".fp" bank 0 slot 1 priority 2

This needs high priority as well. And you'll need the modified WLALINK I just modded... I'll commit that later.

vhelin, 2019-05-27:
> ok really good news for the solved issue :) but i don't know how i can help you for now.
> 
> I'm still learning asm but it is ... not common ! I already read better tutorials but i keep motivation!

I don't think you can help me at this point, I just need to go through this until the very end and see why it's not linking... :P Hard work is all this takes. :D More or less. I think more work...

vhelin, 2019-05-27:
Ok, if you now try the latest sources and make those two "priority" changes from .RAMSECTIONs, you'll get

crt0_snes.obj: crt0_snes.asm:284: FIX_REFERENCES: Reference to an unknown label "__malloc_init".

And there seems to be a lot of labels with "__" or "___" prefix... To be removed, the prefix. Can you make any progress from this?

hoit, 2019-05-27:
yes, i will check it but i think i will not finish today. i keep you in touch in the next days

vhelin, 2019-05-27:
No hurry! Let's give this a loose hard deadline of 31.12.2029. :D Just contribute when you have time, like me. This doesn't seem to be a simple problem fixed in a day... Needs lots of work.

hoit, 2019-05-28:
It is hard to get help and i enjoy your own ! I don't want to leave hanging because i know that i will not find solution myself :) 

So i solved few issues on " Reference to an unknown label xxxxxx" then i get :

E:\snes-examples\hello_world>wlalink.exe linkfile test.sfc
libc.obj: lzsss.asm:164: FIX_REFERENCES: Too large distance (138 bytes from $e71c to $e7a7 "_ldv_next_flag") for a 8-bit reference.

I just moved this part :

```
_ldv_next_flag:
	dec	LZSS_CNT2
	bne	+
	jmp	_ldv_loop
```

before `_ldv_flag_clear:` and it seems to work.
Now, it is a first victory : wlalink command finish correctly but the rom paused automatically after few seconds.

Here is the link to the current version with .sfc file :
http://s000.tinyupload.com/?file_id=06459745816719328138

I need to read informations on no$sns debugger to understand it. And i would like to build other samples to see if it produce others errors.

vhelin, 2019-05-29:
Ouch! If you moved that code block (three code lines and the label) starting from _ldv_flag_clear somewhere else, I'm 100% sure the code will fail even if it compiles...

_ldv_copy_string_done:
        sty     LZSS_TEXTBUF_OFFS
        plb
        ply
        plx
_ldv_next_flag:
        dec     LZSS_CNT2
        bne     +
        jmp     _ldv_loop
+:
        jmp     _ldv_check_flags
_ldv_done:
        plb
        ply
        plx
        plp

That's because the execution will flow from "plx" (push x register) command one line before into _ldv_next_flag and further... :)

If you look at

  http://6502.org/tutorials/65c816opcodes.html

it says about "bra": "These instructions can branch -128 to 127 bytes from the address of the next instruction." - 138 bytes is clearly too much.

BUT

about "brl" it says: "BRL is like BRA, except that the branch displacement is 16 bits, instead of 8 bits, which means it can branch -32768 to 32767 bytes from the address of the next instruction (or -32765 to 32770 bytes from the address of the BRL instruction)."

===> You should be able to replace "bra" on line lzsss.asm:164 with "brl" and the code should assemble and also work. :) - No need to move anything around...

vhelin, 2019-05-29:
I wonder why the code originally worked as that jump should have failed with WLA DX v9.4 as well. Perhaps WLA DX was buggier at the time?

vhelin, 2019-05-29:
I think we are pretty close to the solution here: If you can undo that latest code block moving, and use "brl" instead of "bra" here at lzsss.asm:164, then it might work?

If not, could you start over, and do just renaming of those labels that begin with "\_" or "\_\_"? I can do the rest (the 65816 asm coding)? :)

I've never written a line of SNES/65816 code in my life, but I could start from here. It's just like every other ASM, but the instructions are partially different, and I have good documents here at my disposal...

vhelin, 2019-05-29:
I just tried test.sfc in SNES9X, but it just gave me a black screen... Something has gone wrong? What should it do? Print just "hello wolrd" on screen?

vhelin, 2019-05-29:
> 
> 
> I think we are pretty close to the solution here: If you can undo that latest code block moving, and use "brl" instead of "bra" here at lzsss.asm:164, then it might work?
> 
> If not, could you start over, and do just renaming of those labels that begin with "_" or "__" and add those pirorities to those two sections? I can do the rest (the 65816 asm coding)? :)
> 
> I've never written a line of SNES/65816 code in my life, but I could start from here. It's just like every other ASM, but the instructions are partially different, and I have good documents here at my disposal...

hoit, 2019-05-29:
I replaced bra by brl and generated .sfc file. Result seems to be the same : the game is paused automatically after few seconds but now when i go to no$sns it crash !

> Perhaps WLA DX was buggier at the time?

tcc816 was provided with its own wla dx version, maybe it has been changed in sources but i don't think.

> If not, could you start over, and do just renaming of those labels that begin with "_" or "__" and add priorities to those two sections? I can do the rest (the 65816 asm coding)? :)

Not sure to understand :
You mean renaming **all labels** in sources that contains "_" or "__" ?
and priorities for .ramsection ".registers"  and ".fp" ?
Because last sources uploaded contains modifications to build the code without error (with lzsss wrong modification) but original file is here :
https://github.com/alekmaul/pvsneslib/blob/master/pvsneslib/source/lzsss.asm

For now, test.sfc only create a black screen because i removed all dependencies to ease understanding of issues. I need to clean sources and add snes text source files too to print something. Lot of them are written in C, i have to adapt pvsneslib structure to give a correct sample

vhelin, 2019-05-29:
It is very possible that the author of tcc816 modified WLA DX v9.4 even more than what we know...

I was just suggesting, that if you have no underestanding of 65816 code, then go to the starting point and do the following things:
- rename labels and references that start with "\_" and "\_\_", like you've done now
- add those PRIORITY things to .ramsection ".registers" and ".fp"
- make this "bra" -> "brl" change

and let me have the zip of the project when it doesn't assemble and I'll try to fix the 65816 code as best as I can. Don't move 65815 code blocks around as it doesn't work. :)

...

Anyway, if you have a hunch of what to do next, feel free to do what you want. I wast just drinking beer here and suggesting something LOL. :D

hoit, 2019-05-29:
Thanks, it is more clear now, i am sometime lost in translation :)

Like you said before, use sta.w instead of sta.b will maybe not work and i did it in sources.
Ok i prepare a new and clean sample with only modifications that we are sure it works.

> Anyway, if you have a hunch of what to do next, feel free to do what you want. I wast just drinking beer here and suggesting something LOL.

It was a good suggestion ! I am following your advise before (by providing a new sample) to solve this point.
I will think about how to test more complex sample with text or pictures (already existing in pvsneslib) later.

hoit, 2019-05-30:
I identied all modifications we did in differents files.
For the next example that i will upload this afternoon, i will cancel all bold italic lines (xxx.b => xxx.w).

I think other lines should be ok !

Sorry for the format :) i will join this file uptodate with the sample.

file => line => old value => new value

crt0_snes.asm => 3 => .ramsection ".registers" bank 0 slot 1 => .ramsection ".registers" bank 0 slot 1 priority 1
**_crt0_snes.asm => 243 => sta.b nmi_handler + 2 => sta.w nmi_handler + 2_**
crt0_snes.asm => 247 => -   lda.l startsection.data,x => -   lda.l SECTIONSTART_.data,x
crt0_snes.asm => 248 => sta.l startramsectionram.data,x => sta.l SECTIONSTART_ram.data,x
crt0_snes.asm => 251 => cpx #(endsection.data-startsection.data) => cpx #(SECTIONEND_.data-SECTIONSTART_.data)
crt0_snes.asm => 260 => ldx #(((endramsection.bss-startramsection.bss) & $fffe) + 2) => ldx #(((SECTIONEND_.bss-SECTIONSTART_.bss) & $fffe) + 2)
**_crt0_snes.asm => 273 => sta.b move_insn + 2 => sta.w move_insn + 2_**
**_crt0_snes.asm => 279 => sta.b move_backwards_insn + 2 => sta.w move_backwards_insn + 2_**
crt0_snes.asm => 281 => pea $ffff - endramsectionram.data => pea $ffff - SECTIONEND_ram.data
crt0_snes.asm => 282 => pea :endramsectionram.data => pea :SECTIONEND_ram.data
crt0_snes.asm => 283 => pea endramsectionram.data => pea SECTIONEND_ram.data
crt0_snes.asm => 284 => jsr.l __malloc_init => jsr.l malloc_init
**_crt0_snes.asm => 297 => sta.l $fffd => sta $fffd_**

**_libc.asm => 19 => sta.b move_insn + 1 => sta.w move_insn + 1_**
**_libc.asm => 40 => sta.b move_insn + 1 => sta.w move_insn + 1_**
**_libc.asm => 77 => sta.b move_backwards_insn + 1 => sta.w move_backwards_insn + 1_**

libc_c.asm => 71 => .define ____tccs_consoleVblankMode7_locals 0 => .define __tccs_consoleVblankMode7_locals 0
libc_c.asm => 81 => __tccs___compact: => tccs___compact:
libc_c.asm => 332 => lda.w __tccs_msys + 0 => lda.w tccs_msys + 0
libc_c.asm => 346 => lda.w __tccs_msys + 0 => lda.w tccs_msys + 0
libc_c.asm => 348 => lda.w __tccs_msys + 0 + 2 => lda.w tccs_msys + 0 + 2
libc_c.asm => 373 => lda.w __tccs_msys + 4 => lda.w tccs_msys + 4
libc_c.asm => 375 => lda.w __tccs_msys + 4 + 2 => lda.w tccs_msys + 4 + 2
libc_c.asm => 378 => jsr.l __tccs___compact => jsr.l tccs___compact
libc_c.asm => 384 => sta.w __tccs_msys + 0 => sta.w tccs_msys + 0
libc_c.asm => 386 => sta.w __tccs_msys + 0 + 2 => sta.w tccs_msys + 0 + 2
libc_c.asm => 387 => lda.w __tccs_msys + 0 => lda.w tccs_msys + 0
libc_c.asm => 399 => lda.w __tccs_msys + 0 + 2 => lda.w tccs_msys + 0 + 2
libc_c.asm => 401 => lda.w __tccs_msys + 0 => lda.w tccs_msys + 0
libc_c.asm => 406 => lda.w __tccs_msys + 0 => lda.w tccs_msys + 0
libc_c.asm => 408 => lda.w __tccs_msys + 0 + 2 => lda.w tccs_msys + 0 + 2
libc_c.asm => 438 => sta.w __tccs_msys + 0 => sta.w tccs_msys + 0
libc_c.asm => 440 => sta.w __tccs_msys + 0 + 2 => sta.w tccs_msys + 0 + 2
libc_c.asm => 441 => lda.w __tccs_msys + 0 => lda.w tccs_msys + 0
libc_c.asm => 443 => lda.w __tccs_msys + 0 + 2 => lda.w tccs_msys + 0 + 2
libc_c.asm => 459 => sta.w __tccs_msys + 0 => sta.w tccs_msys + 0
libc_c.asm => 461 => sta.w __tccs_msys + 0 + 2 => sta.w tccs_msys + 0 + 2
libc_c.asm => 492 => __malloc_init: => malloc_init:
libc_c.asm => 513 => sta.w __tccs_msys + 4 => sta.w tccs_msys + 4
libc_c.asm => 515 => sta.w __tccs_msys + 4 + 2 => sta.w tccs_msys + 4 + 2
libc_c.asm => 517 => sta.w __tccs_msys + 0 => sta.w tccs_msys + 0
libc_c.asm => 519 => sta.w __tccs_msys + 0 + 2 => sta.w tccs_msys + 0 + 2
libc_c.asm => 520 => lda.w __tccs_msys + 0 => lda.w tccs_msys + 0
libc_c.asm => 522 => lda.w __tccs_msys + 0 + 2 => lda.w tccs_msys + 0 + 2
libc_c.asm => 524 => lda.w __tccs_msys + 4 => lda.w tccs_msys + 4
libc_c.asm => 526 => lda.w __tccs_msys + 4 + 2 => lda.w tccs_msys + 4 + 2
libc_c.asm => 564 => lda.w __tccs_msys + 4 => lda.w tccs_msys + 4
libc_c.asm => 566 => lda.w __tccs_msys + 4 + 2 => lda.w tccs_msys + 4 + 2
libc_c.asm => 569 => jsr.l __tccs___compact => jsr.l tccs___compact
libc_c.asm => 575 => sta.w __tccs_msys + 0 => sta.w tccs_msys + 0
libc_c.asm => 577 => sta.w __tccs_msys + 0 + 2 => sta.w tccs_msys + 0 + 2
libc_c.asm => 1969 => __tccs_skip_atoi: => tccs_skip_atoi:
libc_c.asm => 2068 => __tccs_number: => tccs_number:
libc_c.asm => 3605 => jsr.l __tccs_skip_atoi => jsr.l tccs_skip_atoi
libc_c.asm => 3723 => jsr.l __tccs_skip_atoi => jsr.l tccs_skip_atoi
libc_c.asm => 4227 => lda.w #:__tccs_L.2 => lda.w #:tccs_L.2
libc_c.asm => 4229 => lda.w #__tccs_L.2 + 0 => lda.w #tccs_L.2 + 0
libc_c.asm => 4503 => jsr.l __tccs_number => jsr.l tccs_number
libc_c.asm => 5162 => jsr.l __tccs_number => jsr.l tccs_number
[... 134 lines cut ...]
libc_c.asm => 12 492 => _m7sin dsb 1 => m7sin dsb 1
libc_c.asm => 12 493 => _m7cos dsb 1 => m7cos dsb 1

libm.asm => 7 => .ramsection ".fp" bank 0 slot 1 => .ramsection ".fp" bank 0 slot 1 priority 2
**_libm.asm => 83 => ror.b tcc__M1 + 1 => ror.w tcc__M1 + 1_**
**_libm.asm => 84 => ror.b tcc__M1 + 2 => ror.w tcc__M1 + 2_**
**_libm.asm => 116 => sta.b tcc__M1 + 2 => sta.w tcc__M1 + 2_**
**_libm.asm => 136 => sta.b tcc__M1 + 2 => sta.w tcc__M1 + 2_**
**_libm.asm => 155 => sta.b tcc__X1 + 2 => sta.w tcc__X1 + 2_**
**_libm.asm => 172 => lda.b tcc__M1 + 2 => lda.w tcc__M1 + 2_**
**_libm.asm => 216 => lda.b tcc__r9h + 1 => lda.w tcc__r9h + 1_**
**_libm.asm => 221 => lda.b tcc__r9 + 1 => lda.w tcc__r9 + 1_**
**_libm.asm => 222 => sta.b tcc__M1 + 2 => sta.w tcc__M1 + 2_**
[... cut ...]
**_libm.asm => 253 => stz.b tcc__M1 + 2	; value loaded is 16 bits, clear low byte of => stz.w tcc__M1 + 2	; value loaded is 16 bits, clear low byte of_**
**_libm.asm => 261 => stz.b tcc__M1 + 2 => stz.w tcc__M1 + 2_**
**_libm.asm => 267 => ror.b tcc__M1 + 1 => ror.w tcc__M1 + 1_**
**_libm.asm => 268 => ror.b tcc__M1 + 2 => ror.w tcc__M1 + 2_**

**_libtcc.asm => 332 => lda.b tcc__r10 + 2 => lda.w tcc__r10 + 2_**

lzsss.asm => 164 => bra	_ldv_next_flag => brl _ldv_next_flag

**_sprites.asm => 160 => ora.l oamMemory+3,x => ora.w oamMemory+3,x_**
**_sprites.sm => 236 => sta.l oamMemory+0,x => sta.w oamMemory+0,x_**

hoit, 2019-05-30:
wooow ! I do not understand anything...
I cancelled all modifications like i said in previous message (all lines in italic bold) and try to compile the project.
Now it build without errors. Maybe it is linked to new changes you did during this issue ?

But the most cool thing : the rom run correctly without pause automatically or crash no$sns. :) We are really near the end !!
Once again, big big thank you for all your time, advises and help !

Here is the current version if you want to see it :
http://s000.tinyupload.com/?file_id=83957380916398180004

So i think the next steps are to test most complex samples (with text, pictures, sound, ...) provided with PVSneslib (but need to adapt some things in PVSneslib makefile to do it).
Or maybe do you have other ideas to do before ?

vhelin, 2019-05-30:
I tried to run the test.sfc inside your latest zip, but I got just a black screen. Perhaps it's not supposed to show anything?

Anyway, very good news if you think it's working! :) 

Yes, more complex tests with text, pictures and sound would be really good! 👍 I have no other ideas. Happy hacking! :D 

It will be great if you can get more complex tests to work, too!

hoit, 2019-05-30:
Yes, it is not supposed to show anything. I know it stupid but i was to remove a maximum of dependencies.
I saw that libc_c.asm comes from a .c file build with tcc816 so i need to check also how to change this .c file to get the same result.
While checking tcc816 sources, i read :

```
/* yet another terrible workaround
   WLA does not have file-local symbols, only section-local and global.
   thus, everything that is file-local must be made global and given a
   unique name. not knowing how to choose one deterministically (filename?
   could be more than one file with the same name...), I chose to use a
   random number, saved to static_prefix. */
char* static_prefix = "__tccs_";
```

Do you confirm that now it is the case with __ used ? I think i don't need to change it.

I begin with others samples, i keep you informed of course !

vhelin, 2019-05-30:
If you have a label that begins with a \_ and it is inside a section, then the name doesn't show outside of that section. If it is not inside a section it doesn't show outside of the object file...

If a random number is added to that static_prefix, then it would be better to change it to

    char* static_prefix = "tccs_";

Perhaps this would work as well, I'm not 100% sure...

    char* static_prefix = "$tccs_";

But anyway, if your code is not inside a section, labels starting with "__tccs_" will not show outside of the object file...

Did I answer your question? I'm pretty sleepy at the moment... :)

hoit, 2019-05-30:
lol
You should become the owner of tcc816 to improve it !
Yes thanks, your answered it ! So the comment will no more usefull when upgrade will be finished.

For other tests, i have some difficulties for now related to makefiles are not user friendly :( i am trying to generate a linkfile dynamically with all .asm files available in project.

I losing my hairs but i am near the solution !
$(foreach dir,$(SOURCES), $(echo $(notdir $(wildcard $(dir)/*.asm)) >> linkfile;))

vhelin, 2019-05-31:
Haha, I know very little of compilers and even less about SNES's 65816 programming. :) You know tcc816 much better than I do already...

I hope you'll get those makefiles working! I can only write basic makefiles myself, very basic. Other people added the cmake support to WLA DX, not me...

hoit, 2019-06-08:
Hi !

Some news about this thread !
I found files to update (which are used to generate libc_c.asm) and build a new sample with text but the text doesn't appear to the screen for now. I need to do some tests to understand the problem.

But i have few questions :)

Like it was done previously in the lib, i generated .sym file with this command :
`wlalink -S linkfile /e/snes-examples/hello_world/hello_world.sfc`

From this [link](https://wla-dx.readthedocs.io/en/latest/symbols.html), it seems to be useful for emulators but i get "Symbolic Info File Corrupt, Line 4" when loading the rom in No$sns.

Really strange because the last compilation rule (after the command wlalink.exe) in makefile contains :

```
	@sed 's/://' <$(TARGET).sym >$(TARGET).tmp
	@mv $(TARGET).tmp $(TARGET).sym
```

Line 4 is : `[labels]`

how can i solve it ?
Is it really usefull for a game ?

And other question about linkfile, but i think it is not related to wla.
I tried to create a file like this (i am on windows) :

```
[objects]
data.obj
hdr.obj
hello_world.obj
/e/pvsneslib/lib/crt0_snes.obj
libc.obj
libm.obj
libtcc.obj

```

but i get **LOAD_FILE_DATA: Could not open file "/e/pvsneslib/lib/crt0_snes.obj".**

if i change the path to : E:\pvsneslib\lib\crt0_snes.obj
it works correctly.

All other path are using the first format, do you know why it doesn't work with link file ?

Thanks!

[15 more replies]

closed 2020-11-18 by hoit