vhelin/wla-dx #233: Symbol file can't be parsed by NO$SNES

JensRestemeier, 2019-05-05:
It looks like NO$GMB style symbol files can't be parsed by the current version of NO$SNES. 
The linker writes symbols as XX:XXXX, while NO$SNES expects XXXXXXXX. A small patch fixes this:

```
diff --git a/wlalink/write.c b/wlalink/write.c
index 999a6b7..1fc4594 100644
--- a/wlalink/write.c
+++ b/wlalink/write.c
@@ -903,35 +903,50 @@ int write_symbol_file(char *outname, unsigned char mode) {
   
   if (mode == SYMBOL_MODE_NOCA5H) {
     /* NO$GMB SYMBOL FILE */
-    fprintf(f, "; no$gmb symbolic information for \"%s\".\n", outname);
-
-    l = labels_first;
-    while (l != NULL) {
-      if (is_label_anonymous(l->name) == SUCCEEDED || l->status == LABEL_STATUS_SYMBOL || l->status == LABEL_STATUS_BREAKPOINT) {
-	l = l->next;
-	continue;
-      }
-      /* skip all dropped section labels */
-      if (l->section_status == ON) {
-	s = sec_first;
-	while (l->section != s->id)
-	  s = s->next;
-	if (s->alive == NO) {
-	  l = l->next;
-	  continue;
+    if (snes_mode == 0) {
+			fprintf(f, "; no$gmb symbolic information for \"%s\".\n", outname);
+		} else {
+			fprintf(f, "; no$snes symbolic information for \"%s\".\n", outname);
+		}
+
+		l = labels_first;
+		while (l != NULL) {
+			if (is_label_anonymous(l->name) == SUCCEEDED || l->status == LABEL_STATUS_SYMBOL || l->status == LABEL_STATUS_BREAKPOINT) {
+				l = l->next;
+				continue;
+			}
+			/* skip all dropped section labels */
+			if (l->section_status == ON) {
+				s = sec_first;
+				while (l->section != s->id)
+					s = s->next;
+				if (s->alive == NO) {
+					l = l->next;
+					continue;
+				}
+			}
+			if (snes_mode == 0)
+			{
+				if (l->status == LABEL_STATUS_LABEL) {
+					fprintf(f, "%.2x:%.4x %s\n", l->base + l->bank, (int)l->address, l->name);
+				}
+				else {
+					fprintf(f, "00:%.4x %s\n", (int)l->address, l->name);
+				}
+			}
+			else
+			{
+				if (l->status == LABEL_STATUS_LABEL) {
+					fprintf(f, "%.4x%.4x %s\n", get_snes_pc_bank(l) >> 16, (int)l->address, l->name);
+				}
+				else
+				{
+					fprintf(f, "0000%.4x %s\n", (int)l->address, l->name);
+				}
+			}
+			l = l->next;
+		}
 	}
-      }
-      if (l->status == LABEL_STATUS_LABEL) {
-	if (snes_mode == 0)
-	  fprintf(f, "%.2x:%.4x %s\n", l->base + l->bank, (int)l->address, l->name);
-	else
-	  fprintf(f, "%.2x:%.4x %s\n", get_snes_pc_bank(l)>>16, (int)l->address, l->name);
-      }
-      else
-	fprintf(f, "00:%.4x %s\n", (int)l->address, l->name);
-      l = l->next;
-    }
-  }
   else {
     /* WLA SYMBOL FILE */
     fprintf(f, "; wla symbolic information for \"%s\".\n", outname);

```

vhelin, 2019-05-05:
Thanks for the fix! I added it to the project, manually (as I don't know how to do it otherwise). :) I hope it works! If it does, please close this issue.

JensRestemeier, 2019-05-05:
Looks fine to me.

closed 2019-05-05 by JensRestemeier