vhelin/wla-dx #234: WLALINK should discard SUPERFREE sections

maxim-zhao, 2019-05-09:
It seems the `-d` option only causes unreferenced `FREE` and `SEMIFREE` sections to be discarded, as is written in the docs. I think unreferenced `SUPERFREE` sections should be discarded too.

vhelin, 2019-05-10:
Thanks for the report, unreference SUPERFREE sections should now be discarded if -d is used...

BTW, could someone who is familiar with hashmaps and labels in WLA DX update "wlalink/discard.c"'s label freeing to work properly (remove labels also from hashmaps)? I can also try to do that, but I'm not 100% sure of what to do...

vhelin, 2019-05-17:
Hey, have you had time to check out the latest sources?

maxim-zhao, 2019-05-18:
I just tried it - but a few other issues are getting in the way, I'll create some new issues shortly...

maxim-zhao, 2019-05-20:
Works On My Machine 👍

closed 2019-05-20 by maxim-zhao