vhelin/wla-dx #235: _sizeof labels should have knowledge of the data they're referencing

aexoden, 2019-05-17:
This workaround also does not help in cases like this:
```
point_data .dsb 1024
point_data_lower dsb 512
point_data_upper dsb 512
```

The _sizeof_point_data label will be set to 0, even though anyone referencing it would clearly be looking for 1024.

_Originally posted by @aexoden in https://github.com/vhelin/wla-dx/issues/228#issuecomment-485044927_

vhelin, 2019-05-19:
To be able to fix this issue, we'll need to make WLA write metadata to object files (in this case: { label: "point_data", size: 1024 bytes } ). Currently WLA just skips dotted .DSBs and others when parsing the code...

I might have time next week to do this, let's see...

vhelin, 2019-05-21:
How are the latest sources working?

aexoden, 2019-05-22:
Took a bit to find somewhere in my project where that feature is useful, but it seems to be working as expected, so I think I'll close this issue.

closed 2019-05-22 by aexoden