vhelin/wla-dx #236: SECTIONSTART labels cause warnings when section names have spaces

maxim-zhao, 2019-05-18:
````
.section "Interrupt handler hook" force
```
Gives warning:
````
WARNING: Could not create label "SECTIONSTART_Interrupt handler hook" as it contains spaces
````
Section labels are free text, spaces should be expected. I suggest you should normalise the string and also ensure uniqueness to avoid collisions.

vhelin, 2019-05-18:
I cannot ensure uniqueness as there might be multiple sections with the same name in separate .asm files, and the assembler only sees the file it is assembling. If we moved the label generation to WLALINK then it would be possible.

maxim-zhao, 2019-05-18:
If the intention is not to see all these labels then maybe you can throw a GUID in there?

vhelin, 2019-05-18:
That's a good point, and very doable. :) I'll think about it a bit more, how to do it...

maxim-zhao, 2019-05-22:
Side note: the extra labels are also causing some pain because they end up exported and then emulators often see three labels for the same address (SECTIONEND_previous_section, SECTIONSTART_this_section, and a real label). Does it make sense to filter these out of the symbol file, if they serve some internal purpose?

vhelin, 2019-05-22:
There is a reason why they are behind the -x flag. :)

vhelin, 2019-05-22:
Some people need them, thus they are behind the -x flag. Don't use -x if you don't need them. Or should I create a seprate flag for just these extra labels?

vhelin, 2019-05-22:
-X?

maxim-zhao, 2019-05-22:
The docs only say that -x gets you the filename, time and version definitions. I thought these extra ones were there for helping some other functionality?

I can live without -x, though.

vhelin, 2019-05-23:
I forgot to update the docs after adding this new feature. :P I'll fix that...

vhelin, 2019-05-23:
Ok, I made it so that if WLALINK finds duplicate SECTIONSTART_* and SECTIONEND_* labels, it'll make them unique:

main.o: SECTIONSTART_Main (has a section called "Main")
game.o: SECTIONSTART_Main (has a section called "Main")
setup.o: SECTIONSTART_Main  (has a section called "Main")

===>

main.o SECTIONSTART_Main
game.o: SECTIONSTART_Main_1
setup.o: SECTIONSTART_Main_2

I hope this solves this issue.

closed 2019-05-23 by maxim-zhao