vhelin/wla-dx #237: Z80 assembly does not work

maxim-zhao, 2019-05-18:
Any attempt to assemble anything I've written fails on the first opcode seen :( e.g. code at https://github.com/maxim-zhao/mappy-gg-translation
````
mappy-gg-translation.gg.asm:62: ERROR: Unknown symbol "call".
````

vhelin, 2019-05-18:
I just downloaded the latest master from GitHub, and all seems to work also on the z80 side. At least the included tests run through without errors...

maxim-zhao, 2019-05-18:
It may well be me at fault :) I built my fork after rebasing on latest, which has no code changes, but maybe it needs a clean. I'll check later when I get access to power

maxim-zhao, 2019-05-20:
Hmm... building Release-x64 in VS2017 gets me non-working programs via ninja. Building at the commandline with
````
cmake .
cmake . --build --config Release
````
...gets me something that works, via msbuild. I think incomprehensible CMake is at fault here, or perhaps just PEBKAC on my end.

closed 2019-05-20 by maxim-zhao

vhelin, 2019-05-22:
I have no idea what you just said, but I'll trust you. 👍