vhelin/wla-dx #238: Address arithmetic in 6502 throws out-of-bounds error in COMPUTE_PENDING_CALCULATIONS

beholdnec, 2019-05-18:
Here is a file I am trying to assemble: https://gist.github.com/beholdnec/50194596e34d6db0b913b727e113d8e5

The commands to assemble and link are:
```
wla-6502 -v test.s
wlalink -v linkfile test.bin
```

When I run these commands, wla-link returns the following message:
> test.o: test.s:64: COMPUTE_PENDING_CALCULATIONS: Result (65547/$1000b) of a computation is out of 8-bit range.

At line 64 of test.s, there is the following line:
`lda ballY+1 ; HERE IS THE PROBLEMATIC LINE`

ballY is a 16-bit value, and I am attempting to access the high byte. So if ballY is at $0A, then ballY+1 should access $0B. Instead, it seems to try to access $1000b. Is this a bug in the linker?

vhelin, 2019-05-19:
Your .MEMORYMAP is a bit wrong

  .memorymap
    slotsize $10000
    defaultslot 0
    slot 0 $10000   ; problem!
  .endme

Here you make slot 0 to start from $10000. Replace that line with

  slot 0 $0

and all should work. :)

beholdnec, 2019-05-19:
Thanks for the tip.

closed 2019-05-19 by beholdnec