vhelin/wla-dx #239: "Definition <x> is a string definition" if I use too many decimal places

maxim-zhao, 2019-05-21:
I'm going back to some old code which has this:
````
.define BASE_ACC 100 ; base acceleration - change in velocity per frame for button pressed in the appropriate direction
PlayerAccelerationTable:
.define ANGLE_ACC BASE_ACC*0.707106781 ; BASE_ACC * sin(45 degrees)
;   X              Y
.dw 0,             0          ; %0000 = 0
.dw 0,             -BASE_ACC  ; %0001 = U
.dw 0,             +BASE_ACC  ; %0010 = D
.dw 0,             0          ; %0011 = UD = 0
.dw -BASE_ACC,     0          ; %0100 = L
.dw -ANGLE_ACC,    -ANGLE_ACC ; %0101 = UL
.dw -ANGLE_ACC,    +ANGLE_ACC ; %0110 = DL
.dw -BASE_ACC,     0          ; %0111 = UDL = L
.dw BASE_ACC,      0          ; %1000 = R
.dw +ANGLE_ACC,    -ANGLE_ACC ; %1001 = UR
.dw +ANGLE_ACC,    +ANGLE_ACC ; %1010 = DR
.dw +BASE_ACC,     0          ; %1011 = UDR = R
.dw 0,             0          ; %1100 = LR = 0
.dw 0,             -BASE_ACC  ; %1101 = ULR = U
.dw 0,             +BASE_ACC  ; %1110 = DLR = D
.dw 0,             0          ; %1111 = UDLR = 0
````
This assembled fine in 2009 :) but today I get this:
````
ono.sms.asm:1954: STACK_CALCULATE: Definition "ANGLE_ACC" is a string definition.
ono.sms.asm:1954: STACK_CALCULATE: Definition "ANGLE_ACC" is a string definition.
ono.sms.asm:1955: STACK_CALCULATE: Definition "ANGLE_ACC" is a string definition.
ono.sms.asm:1955: STACK_CALCULATE: Definition "ANGLE_ACC" is a string definition.
ono.sms.asm:1958: STACK_CALCULATE: Definition "ANGLE_ACC" is a string definition.
ono.sms.asm:1958: STACK_CALCULATE: Definition "ANGLE_ACC" is a string definition.
ono.sms.asm:1959: STACK_CALCULATE: Definition "ANGLE_ACC" is a string definition.
ono.sms.asm:1959: STACK_CALCULATE: Definition "ANGLE_ACC" is a string definition.
ono.sms.asm:1954: INTERNAL_PASS_2: Reference to a string definition "ANGLE_ACC"?
````
If I drop the last decimal place of my float:
````
.define ANGLE_ACC BASE_ACC*0.70710678 ; BASE_ACC * sin(45 degrees)
````
Then it is fine! Is there some deliberate upper limit on the precision? Obviously it doesn't actually make any difference to my program...

vhelin, 2019-05-21:
Looks super weird. There should be no accuracy limit on the precision... I'll look into it.

vhelin, 2019-05-21:
Found it. The same parsing loop was used to parse integers and floats, and max digit count was set to 9. I'll make it so that if WLA spots a float it'll up that limit to something like 24... Fix coming soon.

maxim-zhao, 2019-05-21:
Seems fixed. Well... arguably it ought to be an error if I have a number that is too large, rather than silently converting it to a string.

closed 2019-05-21 by maxim-zhao

vhelin, 2019-05-21:
That's very true. I'll add such error messages later this week...

vhelin, 2019-05-22:
If you test the latest sources, number parsers should now give an explicit error message when integers/floats have too many digits.