vhelin/wla-dx #240: Add a SNESCHECKSUMOVERRIDE for use in the .SNESHEADER instruction

Icedude907, 2019-05-31:
A simple feature request. Allow us to override the sneschecksum that is automatically computed when we declare .SNESHEADER. This is because I don't think the checksum wla-dx is giving me the right checksum when using HiROM and FastROM, so I'd much rather fill that in by hand, rather than having to do the whole header by hand otherwise.
I would do this myself, but I can't seem to make heads or tails of the codebase... sorry about that.

vhelin, 2019-05-31:
I think better would be if we could make WLA DX calculate the correct checksum? :)

Can you give me pointers or a small test case that fails? Anyway, I'll try to find some checksum calculating tools etc. mean while and try to test WLA DX's calculations myself...

Icedude907, 2019-06-01:
Hey. Here is a demo of the incorrect checksum. It requires an EarthBound US ROM to work properly. Just copies the input to the output via the `.INCBIN` macro, but you'll have to copy your own input ROM to the `inout` directory and then run `make` from the root folder. Comparing the two ROMs, the EarthBound ROM has a checksum and complement of `0xB7BF4840`, while the one by wla-dx returns `0xAC5053AF` 

https://github.com/Icedude907/BrokenWLA-DXChecksum/tree/master/inout

Icedude907, 2019-06-01:
I looked at a program (located here) https://forums.nesdev.com/download/file.php?id=4813&sid=96e29e37d1fac51b1974642817d278e4
According to what the community knows, it seems to be that EarthBound had the faulty checksum, not wla-dx.
That's quite unusual, considering this was an officially licenced Nintendo product...

vhelin, 2019-06-01:
> Hey. Here is a demo of the incorrect checksum. It requires an EarthBound US ROM to work properly. Just copies the input to the output via the `.INCBIN` macro, but you'll have to copy your own input [...]

Thanks for this! I'm checking it out...

BTW you could just write

.BACKGROUND "./inout/EarthBoundUS.smc"

instead of

.BANK 0 SLOT 0
.ORG 0
.SECTION "Bank0"
    .INCBIN "./inout/EarthBoundUS.smc" READ $10000 - 78
.ENDS

.BANK 1 SLOT 0
.SECTION "Bank1"
    .INCBIN "./inout/EarthBoundUS.smc" SKIP $010000 READ $10000
.ENDS

...

vhelin, 2019-06-01:
For Street Fighter II ROM WLA DX calculates the checksum correctly... I'll test more ROMs.

vhelin, 2019-06-01:
Super Castlevania IV checksum also is correct if written by WLA DX, but Soldiers of Fortune (24Mbit) is not. Then again Soccer Shootout's checksum (also 24Mbit) is correctly calculated...

vhelin, 2019-06-01:
I can't really find a bug in WLA DX's checksum calculator. If you look at wlalink/compute.c and its compute_snes_checksum(), that's where can checksum calculations happen...

strager, 2019-09-08:
I encountered the same problem with Super Metroid (NTSC). I'll upload a fix soon. (My fix works for Earthbound as well.)

strager, 2019-09-08:
#256 should fix this.

vhelin, 2019-09-08:
@Icedude907 If you try out the latest sources, will they work better?

Icedude907, 2020-06-10:
Sorry for the huge gap in a response. 
Yes, my version of snes9x is reporting the checksum to be ok now.

closed 2020-06-10 by Icedude907