vhelin/wla-dx #241: Allow support for partially remapped strings

aexoden, 2019-06-14:
When working with strings, the encoded values often do not match ASCII encodings. .asciitable and .asc help with this somewhat, by allowing the user to define a mapping table.

However, sometimes there are control codes that make no sense to map to any particular character. For example, a text routine might have letters A-Z as $42 through $5B (vs. $41 through $5A in ASCII), with $40 and $41 interpreted as icons (with no real text equivalent). If you try the naive solution, and do something like this:
`
.asc $41, "Cure"
`
It won't work, because the $41 is interpreted as an "A" and is remapped to $42. Therefore, my request is to either a) modify .asc to not map bytes specified directly (but someone may have relied on the existing behavior) or b) add a new directive that provides this hybrid functionality instead (where characters inside a string are remapped, but bytes specified directly are not). Of course, it's entirely possible there is already existing functionality that can meet this need, but if so, I haven't been able to find it.

vhelin, 2019-06-14:
My question is: If .ASC does mapping and .DB does not, why not just write

.db $41
.asc "Cure"

in this case? :) It wouldn't be a huge task to make e.g., .ASC2 that maps only given strings and not bytes, but is there really a need for such a directive when .ASC and .DB can be used together to accomplish the same thing?

aexoden, 2019-06-14:
I agree, for simpler examples that works (though it's a little bit more clutter when you have a few hundred entries). I thought I had examples that needed multiple codes in one entry, but the only ones I can find also encode multiple letters with one byte (and that's clearly a much different problem).

vhelin, 2021-03-19:
Sorry that this issue was also ignored for so long... If you try out the latest sources and .ASCSTR, does it fix this issue?

aexoden, 2021-03-19:
Appears to work correctly, thanks.

closed 2021-03-19 by aexoden