vhelin/wla-dx #244: Addresses in ramsection are not correct.

aexoden, 2019-07-05:
It's possible I'm doing something wrong, but given that behavior has recently changed, I'm posting this. Given the following test program:

```
.memorymap
	slotsize $8000
	defaultslot 1
	slot 0 $0000
	slot 1 $8000
.endme

.lorom

.rombanksize $8000
.rombanks 1

.base $7F
.ramsection "RAM" bank 0 slot 0
	test_var1 db
	test_var2 db
.ends

main:
	LDA test_var2.l
	LDA test_var2.l + 1
```
It currently fails to link, giving this error:
```
test.o: test.asm:21: COMPUTE_PENDING_CALCULATIONS: Result (24969218/$17d0002) of a computation is out of 24-bit range.
```
In addition, if I eliminate the last line, even the address of the first calculation is wrong in the generated output. (Generated bytes were AF 01 00 7D, which is not the expected bank.) The errors only seem to happen if LoROM is enabled (which is unnecessary for this test program, but needed for my larger project.)

I also tried doing the ramsection like this:
```
.ramsection "RAM" bank $7F slot 0
	test_var1 db
	test_var2 db
.ends
```
But the generated code in that case is AF 01 00 FE, which is also the wrong bank. Again, disabling LoROM seems to fix it. (I don't think this second version worked last time I tried either, which is why I was using the first variant.)

vhelin, 2019-07-06:
Thanks! I'll try to look into this later today.

vhelin, 2019-07-07:
Do the latest sources work any better?

aexoden, 2019-07-07:
After testing the latest source, it seems slightly better, but there are still some weird issues (again, I could be doing things that aren't intended, so let me know if that's the case!).

If I try to specify the ramsection bank like below, the bank value seems to be completely ignored and defaults to zero.

```
.ramsection "RAM1" bank $7E slot 0
  test_var1 dsb $8000
.ramsection
```

If I resign myself to just using .base instead of specifying the bank directly, and use this test program:

```
.memorymap
	slotsize $8000
	defaultslot 1
	slot 0 $0000
	slot 1 $8000
.endme

.lorom

.rombanksize $8000
.rombanks 1

.base $7E
.ramsection "RAM1" bank 0 slot 0
	test_var1 dsb $8000
.ends

.base $7F
.ramsection "RAM2" bank 0 slot 0
	test_var2 dsb $8000
.ends

.bank 0 slot 1
.section "Bank"

main:
	LDA test_var1.l
	LDA test_var1.l + 1
	LDA test_var2.l
	LDA test_var2.l + 1

.ends
```
I get this error:

```
INSERT_SECTIONS: No room for RAMSECTION "RAM2" (32768 bytes) in slot 0.
```

vhelin, 2019-07-07:
In the latter case both RAM1 and RAM2 ramsections are in bank 0 slot 0, the slot is $8000 in size and the ramsections are both $8000 in size -> RAM1 takes up the whole RAM bank (0). Thus there is no room for RAM1. If you write

.base $7F
.ramsection "RAM2" bank 1 slot 0
	test_var2 dsb $8000
.ends

... then RAM2 should be placed inside RAM bank 1 (that has room for RAM2)... Assuming your target system has more than one RAM bank in slot 0.

The first one is a bug / issue. Oh crap I forgot how SNES's memory map works. Depending on the bank it either has ROM in that bank or RAM...

I think I need some time to get this to work (I hope I can get it to work, WLA DX doesn't really have a support for systems with changing memorymaps). 

But in this example, 

.base $7F
.ramsection "RAM2" bank 1 slot 0
    test_var2 db
.ends

    .dl test_var2

... the .dl should procude $00 $00 $80 ?

aexoden, 2019-07-07:
I've been looking at the code somewhat, and I can figure out how to make it work for either with .lorom or without (since get_snes_pc_bank assumes the rom_address is a ROM address and ends up multiplying it to get the RAM address, which doesn't make sense since it was a RAM address in the first place).

For SNES (at least with lorom, which is the only one I have experience with), except for banks $7E and $7F (I don't know about banks $80+ but I don't work with them at all), anything about $8000 is mapped to a ROM bank. $7E and $7F are entirely RAM. The first $2000 of $7E is mirrored in other banks ($00 to $3F I think). 

In any case, for the example you gave, I would expect bank 1 relative to a $7F base to produce 00 00 80, yes, though I'd more likely be doing that with $7E (and expecting 00 00 7F) so I can define variables in either bank of RAM.

vhelin, 2019-07-07:
If you can code a fix to this, please do. My head is spinning after looking at SNES's memory map. :) Anyway, I'll try to fix it, too, but if you are faster, that's just good! For some reason I cannot fully understand this problem...

vhelin, 2019-07-07:
I think I found a working solution to this...

vhelin, 2019-07-07:
Is my latest fix helping any? I with I had SNES programming experience so I could fully understand what I'm doing here, LOL. :D

aexoden, 2019-07-07:
Ok, with the latest commit everything appears to be working correctly, at least as far as I can tell for the moment. It looks like you implemented one of the fixes I was considering, but you actually knew how to do it without examining tons of code first.

vhelin, 2019-07-07:
No problem! I really hope this solves the issue as I'm still confused about SNES memory map and how all that works. :D Thank you for bringing the issue up! And if you find more bugs, please let us know so we can fix them.

Scryfox, 2019-07-30:
Hello! If you don't mind me jumping in on the issue, I seem to be having a similar issue that the ramsection variables are being addressed absolutely rather than using long addressing, and that there seems to be not other way the bank is changed despite the bank label. My memorymap may be off, I was just wondering how @aexoden managed to get their implementation working. Thanks!

vhelin, 2019-07-30:
By absolute addressing you mean 16-bit (.w) addresses? If long addressing is 24-bit (.l), why not try to add .l to the end of your memory address? That should try to force WLA to choose the right opcode with 24-bit address... Does this help?

Scryfox, 2019-07-30:
Sorry, i forgot that the hints affect the memory addresses, not the values. I'm still learning... However, I am noticing an odd behavior where the bank portion of the address is twice the value specified in the bank statement in the ramsection title. (e.g. If my ramsection title is .RAMSECTION "Controller Statuses" BANK $3F SLOT $0, then the bank that is used when I reference a variable is $7E)

vhelin, 2019-07-30:
Hmm, are you using the latest WLA DX? I think that bug appeared in v9.8, but was fixed in the latest v9.9 sources... ?

Scryfox, 2019-07-30:
Wow, didn't realize a new release was out... Works great now, thanks so much! Your work is very much appreciated, this project is great!

vhelin, 2019-07-30:
Well, with WLA DX people should really use the latest sources if possible, most of the time they are pretty stable. Good thing your problem was solved. :) And thank you.

closed 2019-09-06 by vhelin