vhelin/wla-dx #245: Defines with ":" broken

Drenn1, 2019-07-15:
I had some code like this:

```
.define w7Label $c000
.define :w7Label 7
```

To be clear, this is _jank_, and I was doing it because I hadn't gotten around to putting it in a ramsection. It could be argued that it shouldn't be allowed in the first place.

Anyway, the code broke after 9b1129d9927f4759702dd3f1354efbe07ab49f27.

I'm fine with changing my codebase to fix the issue, but I thought I should report the breakage in case it affects anyone else.

vhelin, 2019-07-15:
It is clearly a jank, but a good one! :) Thus I fixed it to work again. I hope this doesn't break anything else.

Thanks for teaching me a new usable word (jank). 👍

Drenn1, 2019-07-20:
Seems to be fixed, thanks.

closed 2019-07-20 by Drenn1