vhelin/wla-dx #247: write_bank_header_calculations() segfaults on inter-library .DEFINE arithmetic

a0r10n, 2019-07-23:
When using .DEFINES in one library file, and creating .DEFINES in another library file using the defines in the first, then attempting to use those .DEFINES causes a SEGFAULT. Attached is a sample project for wla-gb that exhibits this behavior.

Versions I have tested this in:
WLA-GB 9.8a (22.11.2018)
WLA-GB 9.9a (21.7.2019)

WLALINK 5.10a (12.10.2018)
WLALINK 5.11a (7.7.2019)

Built using: gcc (x86_64-win32-seh-rev1, Built by MinGW-W64 project) 7.2.0

Run on Windows 10 Home, version 1803, build 17134.885

[Example.zip](https://github.com/vhelin/wla-dx/files/3424341/Example.zip)

vhelin, 2019-07-24:
Thanks, I'll try to check this out this weekend!

a0r10n, 2019-07-24:
Using a debug build of wlalink indicates that the error is on line 1766, which is `while (sta->section != s->id)`. Specific cause of failure is that `s` is null.

a0r10n, 2019-07-24:
A workaround is to put the computed .DEFINES into a section.

vhelin, 2019-07-27:
Thanks, I spent some time trying to find a fix to this, but couldn't... I'll try more tomorrow! I also found out that CMake doesn't generate proper debug executables of WLALINK, so that should be fixed as well. :)

vhelin, 2019-07-28:
Ok, if you try the latest sources, does it work now?

I found out that WLALINK assumed that all pending computations given inside a library belonged to a section as you can only write code that's inside a section when making library files. I had forgotten that a definition could be issued there, too, that contained pending computations... So WLALINK forced the calculation to belong to a section that didn't exist and the code that was calculating the pending computations didn't find the section and crashed to a NULL pointer...

a0r10n, 2019-07-28:
Yes, the latest sources work as expected. Thank you for all the hard work you do to develop and maintain this project.

Since this fixed the issue, I see fit to close it.

closed 2019-07-28 by a0r10n