vhelin/wla-dx #249: Debug executables don't contain any linenumbers etc...

vhelin, 2019-07-27:
When I compile debug executables (under Cygwin, using -G "Unix Makefiles") using the instructions in INSTALL:

...

Using CMake:

  cmake -DWLA_DEBUG=ON -DWLALINK_DEBUG=ON -DGDB_DEBUGGING=ON -G "[your platform]"
  make

...

... and I use gdb to debug WLALINK, crash occurs due to a bug in WLALINK, which was expected, but I don't get any linenumbers or code lines there...

HACKY SOLUTION:

I go and edit the file "wlalink/CMakeFiles/wlalink.dir/flags.make" and add "-g -O0" to C_FLAGS -> after this I see linenumbers and code lines in gdb...

Could someone who knows about CMake enough fix this bug in the CMake files?

Neui, 2019-07-27:
Either add `-DCMAKE_BUILD_TYPE=Debug` or modify `CMakeCache.txt`, find `CMAKE_BUILD_TYPE` and set it to `Debug`. There is also `RelWithDebInfo` (Release with debug info). Also in `CMakeCache.txt`, you can set the compile flags (`CMAKE_C_FLAGS`). Then re-build (`make`).

vhelin, 2019-07-28:
Thanks, adding "-DCMAKE_BUILD_TYPE=Debug" to cmake command fixes the problem! I'll update INSTALL...

closed 2019-07-28 by vhelin