vhelin/wla-dx #250: Arithmetic on inter-library .DEFINEs fails

a0r10n, 2019-07-27:
Steps to recreate:

- .DEFINE a symbol in one file
- .EXPORT this symbol
- .DEFINE a symbol in a separate library file, performing arithmetic with the symbol defined previously. (Currently, this must be done within a section due to a separate bug in 9.9)

Using this new symbol alone works, as expected.
However, attempting to use this symbol in arithmetic, either for another .DEFINE or within code generates a link error: `PARSE_STACK: Unresolved reference to "Symbol"`.

Attached is a sample project that exhibits this bug
[Example.zip](https://github.com/vhelin/wla-dx/files/3438576/Example.zip)

vhelin, 2019-07-28:
A quick hack to get this to work is to export "Bad" in lib.asm in your Example.zip.

The "Bad+1" computation is saved for the linker and if "Bad" is not exported the linker cannot find it... So what we'll need to do is to either replace "Bad" in "Bad+1" with its definition, when assembling it, or export quietly "Bad" to the linker, but so that "Bad" isn't visible outside the file it was exported from...

I'll try to fix this soon. BTW, great bug reports! :) It's really great when someone includes a tiny project that can be used to show the bug with his bug reports.

vhelin, 2019-07-28:
I managed to fix this issue, but found another bug concerning pending computations and defines. I don't think I have the time to fix it today...

vhelin, 2019-07-29:
How about now, does it work any better with the latest sources?

a0r10n, 2019-07-29:
Yes. It works with the new sources.

closed 2019-07-29 by a0r10n